riscv-collab/riscv-gnu-toolchain #1105: fails to build newlib with march

eflaner, 2022-07-30:
Branch: gcc-12
configured with` ./configure --prefix="${BRISCV}" --target=riscv32-unknown-elf --with-arch=rv32imc_zba_zbb_zbc_zbs_zicsr_zifencei --with-abi=ilp32 --with-cmodel=medany`

BRISCV is the path to which binaries should get into.. 

performed `make -j4`

> In file included from ../../../../../newlib/libc/string/strstr.c:95:
> ../../../../../newlib/libc/string/str-two-way.h: In function 'two_way_long_needle':
> ../../../../../newlib/libc/string/str-two-way.h:410:1: error: unrecognizable insn:
>   410 | }
>       | ^
> (insn 425 424 426 56 (set (reg:SI 205)
>         (const_int 2048 [0x800])) "../../../../../newlib/libc/string/str-two-way.h":326:14 -1
>      (expr_list:REG_EQUAL (const_int 2048 [0x800])
>         (nil)))
> during RTL pass: vregs
> ../../../../../newlib/libc/string/str-two-way.h:410:1: internal compiler error: in extract_insn, at recog.cc:279 

finally ending up with 

> riscv32-unknown-elf-ar: ../stdlib/lib.a: No such file or directory
> riscv32-unknown-elf-ar: ../stdio/lib.a: No such file or directory
> riscv32-unknown-elf-ar: ../string/lib.a: No such file or directory

**looks like the issue of building newlib with this arch..** 

while i switched to couple of other branches of newlib .. i ended up getting 

> ../../../../../newlib/libc/stdio/setvbuf.c: In function 'setvbuf':
> ../../../../../newlib/libc/stdio/setvbuf.c:222:1: error: unrecognizable insn:
>   222 | }
>       | ^
> (insn 226 225 227 31 (set (reg:HI 131)
>         (const_int 2048 [0x800])) "../../../../../newlib/libc/stdio/setvbuf.c":188:17 -1
>      (expr_list:REG_EQUAL (const_int 2048 [0x800])
>         (nil)))
> during RTL pass: vregs
> ../../../../../newlib/libc/stdio/setvbuf.c:222:1: internal compiler error: in extract_insn, at recog.cc:2791

and 

> ../../../../../newlib/libc/stdio/fseeko.c: In function '_fseeko_r':
> ../../../../../newlib/libc/stdio/fseeko.c:337:1: error: unrecognizable insn:
>   337 | }
>       | ^
> (insn 179 178 180 27 (set (reg:HI 193)
>         (const_int 2048 [0x800])) "../../../../../newlib/libc/stdio/fseeko.c":193:15 -1
>      (expr_list:REG_EQUAL (const_int 2048 [0x800])
>         (nil)))
> during RTL pass: vregs
> ../../../../../newlib/libc/stdio/fseeko.c:337:1: internal compiler error: in extract_insn, at recog.cc:2791

Additional info: The assembler and gcc got built correctly (i guess) 

`riscv32-unknown-elf-gcc -v` gives below response (trimmed)
riscv32-unknown-elf
Configured with: /riscv_tools/riscv-gcc/configure --target=riscv32-unknown-elf --prefix=/tools/latest --disable-shared --disable-threads --disable-tls --enable-languages=c,c++ --with-system-zlib --with-newlib --with-sysroot=/tools/latest/riscv32-unknown-elf --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././riscv-gcc --disable-multilib --with-abi=ilp32 **--with-arch=rv32imc_zba_zbb_zbc_zbs_zicsr_zifencei** --with-tune=rocket --with-isa-spec=2.2 'CFLAGS_FOR_TARGET=-Os   -mcmodel=medany' 'CXXFLAGS_FOR_TARGET=-Os   -mcmodel=medany'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 12.1.1 20220723 (GCC)

TommyMurphyTM1234, 2022-08-11:
Have you tried without `-j4` parallel building?
Does that make any difference?

pz9115, 2022-08-11:
I checked it and found that cause by zbs, still track on this.

apinski-cavium, 2022-08-11:
I fixed the building of newlib upstream already:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=106532

apinski-cavium, 2022-08-12:
Note even after the fix in upstream PR 106532, this is still broken.  See https://gcc.gnu.org/bugzilla/show_bug.cgi?id=106586.

pz9115, 2022-09-06:
This has been fixed with Apinski's patch, thanks for his works:

https://gcc.gnu.org/git/?p=gcc.git;a=commit;h=2c721ea9473ad7615bb47b66509097bd254bb839

eflaner, 2022-09-22:
@pz9115 Should i use the upstream master to fix.. ? 

I've used the master branch after #1128  
The error remains same. 

To iterate here's the configuration option

> ./configure --prefix=/scratch1/software/ --with-arch=rv32imac_zba_zbb_zbc_zbs_zicsr_zifencei --with-abi=ilp32 --with-cmodel=medany

Here's the error

> In file included from /scratch1/sw_files/riscv-tools/newlib/newlib/libc/string/strstr.c:95:
> /scratch1/sw_files/riscv-tools/newlib/newlib/libc/string/str-two-way.h: In function 'two_way_long_needle':
> /scratch1/sw_files/riscv-tools/newlib/newlib/libc/string/str-two-way.h:410:1: error: unrecognizable insn:
>   410 | }
>       | ^
> (insn 253 252 254 56 (set (reg:SI 205)
>         (const_int 2048 [0x800])) "/scratch1/sw_files/riscv-tools/newlib/newlib/libc/string/str-two-way.h":326:14 -1
>      (expr_list:REG_EQUAL (const_int 2048 [0x800])
>         (nil)))
> during RTL pass: vregs
> /scratch1/sw_files/riscv-tools/newlib/newlib/libc/string/str-two-way.h:410:1: internal compiler error: in extract_insn, at recog.cc:2791
> 0x5c53ae _fatal_insn(char const*, rtx_def const*, char const*, int, char const*)
> 	/scratch1/sw_files/riscv-tools/gcc/gcc/rtl-error.cc:108
> 0x5c53d0 _fatal_insn_not_found(rtx_def const*, char const*, int, char const*)
> 	/scratch1/sw_files/riscv-tools/gcc/gcc/rtl-error.cc:116
> 0x5c480e extract_insn(rtx_insn*)
> 	/scratch1/sw_files/riscv-tools/gcc/gcc/recog.cc:2791
> 0x86984f instantiate_virtual_regs_in_insn
> 	/scratch1/sw_files/riscv-tools/gcc/gcc/function.cc:1611
> 0x86984f instantiate_virtual_regs
> 	/scratch1/sw_files/riscv-tools/gcc/gcc/function.cc:1985
> 0x86984f execute
> 	/scratch1/sw_files/riscv-tools/gcc/gcc/function.cc:2034

TommyMurphyTM1234, 2022-10-06:
Building using the upstream gcc sources completed successfully.
So, the issue is fixed upstream and will presumably be reflected in the riscv-gnu-toolchain repo when the gcc submodule here is next bumped?
@kito-cheng and/or @cmuellner can probably clarify if necessary.
```
cd ~/Downloads/riscv-tools
git clone git://gcc.gnu.org/git/gcc.git gcc-upstream
git clone https://github.com/riscv/riscv-gnu-toolchain issue-1105
cd issue-1105
./configure --prefix=`pwd`/installed-tools --with-arch=rv32imc_zba_zbb_zbc_zbs_zicsr_zifencei --with-abi=ilp32 --with-cmodel=medany --with-gcc-src=$HOME/Downloads/riscv-tools/gcc-upstream
date > build.log; make 2>&1 | tee -a build.log; date >> build.log
```
riscv-gnu-toolchain
Branch: master
Commit: c2857f9c03867f9fe9ee8e44b7cf57dd83020455

gcc
Branch: master
Commit: e99dcbb54e07b798c3353124f38336f96a826d43

Build log: [build.log.tar.gz](https://github.com/riscv-collab/riscv-gnu-toolchain/files/9722836/build.log.tar.gz)

closed 2022-10-06 by TommyMurphyTM1234

cmuellner, 2022-10-06:
To keep efforts low, we prefer to use release versions only (not release branches and certainly not upstream/master).
So fixes will trickle down to us once a minor release, including the backported patch, or a major release is published (and somebody does the bump of the submodule).

To get changes earlier, one can always override the sources (`--with-gcc-src=...`), just like @TommyMurphyTM1234 did.

TommyMurphyTM1234, 2022-10-06:
Thanks for that @cmuellner and for clarifying the update/sync cadence.

gojimmypi, 2022-12-01:
Hi. I built the latest (`29d02b7`) toolchain today, using the @Wren6991 [Wren6991/Hazard3 instructions](https://github.com/Wren6991/Hazard3#risc-v-toolchain), in particular:

```
./configure --prefix=/opt/riscv --with-arch=rv32ia --with-abi=ilp32 --with-multilib-generator="rv32i-ilp32--;rv32ia-ilp32--;rv32iac-ilp32--;rv32ic-ilp32--;rv32im-ilp32--;rv32ima-ilp32--;rv32imac-ilp32--;rv32imc-ilp32--"
```

The sample code in that repo seems to compile just fine. However when attempting something considerably more complex, I'm seeing a handful of _error: unrecognizable insn_ errors such as this:

```
./wolfssl/src/ssl.c: In function 'wolfSSL_connect':
./wolfssl/src/ssl.c:12975:5: error: unrecognizable insn:
12975 |     }
      |     ^
(insn 133 132 134 27 (set (reg:SI 156)
        (const_int 2048 [0x800])) "./wolfssl/src/ssl.c":12707:49 -1
     (expr_list:REG_EQUAL (const_int 2048 [0x800])
        (nil)))
during RTL pass: vregs
./wolfssl/src/ssl.c:12975:5: internal compiler error: in extract_insn, at recog.cc:2791
0x5caa90 _fatal_insn(char const*, rtx_def const*, char const*, int, char const*)
        /tmp/riscv-gnu-toolchain/gcc/gcc/rtl-error.cc:108
0x5caab2 _fatal_insn_not_found(rtx_def const*, char const*, int, char const*)
        /tmp/riscv-gnu-toolchain/gcc/gcc/rtl-error.cc:116
0x5c9f7c extract_insn(rtx_insn*)
        /tmp/riscv-gnu-toolchain/gcc/gcc/recog.cc:2791
0x867d1f instantiate_virtual_regs_in_insn
        /tmp/riscv-gnu-toolchain/gcc/gcc/function.cc:1611
0x867d1f instantiate_virtual_regs
        /tmp/riscv-gnu-toolchain/gcc/gcc/function.cc:1985
0x867d1f execute
        /tmp/riscv-gnu-toolchain/gcc/gcc/function.cc:2034
Please submit a full bug report, with preprocessed source (by using -freport-bug).
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.
```
In my `Makefile` I'm using the same flags as the [samples](https://github.com/Wren6991/Hazard3/blob/52e665fb45e336bcf87da1b9d5c473be92f99ab5/test/sim/hellow/Makefile#L3):

```
CCFLAGS = -march=rv32i_zicsr_zba_zbb_zbc_zbs -Os
AFLAGS = -march=rv32i_zicsr_zba_zbb_zbc_zbs
```
If I change the flags to
```
CCFLAGS = -march=rv32i -Os 
AFLAGS = -march=rv32i
```
The app will compile successfully, however I'll get an error like this during the `make` simulation:
```
*** Unhandled trap ***
handle_exception @ mepc = 0004ead4
                 mcause = 0000000b
s0: 0005e7c0
s1: 000ffed8
a0: 00000001
a1: 000ffe30
a2: 000ffed8
a3: 00000000
a4: 00000001
a5: 00002009
ra: 0004cd34
sp: 000ffe30
```
The problem is similar, but not exactly the same as this issue:
 (`-march=rv32im_zba_zbb_zbc_zbs` vs `-march=rv32i_zicsr_zba_zbb_zbc_zbs`)
but perhaps close enough. Not sure if I'm making a mistake, if this is a known issue, or if I should create a new issue.

There's a much similar, but much older issue from 2016: https://github.com/riscv-collab/riscv-gnu-toolchain/issues/199

Any suggestions will be greatly appreciated. Thank you.

apinski-cavium, 2022-12-01:
Can you file a different bug; maybe even upstream at https://gcc.gnu.org/bugzilla (with the preprocessed source)?
I am going to be testing RV32 with the bitmanip extensions again (and have a regression tester even).

cmuellner, 2022-12-01:
I agree, if it is not a build issue with the RISC-V toolchain, but a compilation issue with upstream master, then please report it to upstream's Bugzilla. Even better if you could reduce the source code to reproduce the issue.