riscv-collab/riscv-gnu-toolchain #1106: rvv-intrinsic  opt issue at -march=rv64gcv0p7

haolongzhangm, 2022-08-02:
#include "riscv_vector.h"                                                                                                                                                                                     
#include <cstdio>                                                                                                                                                                                             
                                                                                                                                                                                                              
int main() {                                                                                                                                                                                                  
                                                                                                                                                                                                              
  float src[16] = {1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16};                                                                                                                                    
                                                                                                                                                                                                              
  vfloat32m1_t t0, t1, t2, t3, ret0, ret1;                                                                                                                                                                    
                                                                                                                                                                                                              
  t0 = vle32_v_f32m1(src, 4);                                                                                                                                                                                 
  t1 = vle32_v_f32m1(src + 4, 4);
  t2 = vle32_v_f32m1(src + 8, 4);
  t3 = vle32_v_f32m1(src + 12, 4);

  ret0 = vfadd_vv_f32m1(t0, t1, 4);
  ret1 = vfadd_vv_f32m1(t2, t3, 4);
   
  float dst [8] = {0};

  vse32_v_f32m1(dst, ret0, 4);
  vse32_v_f32m1(dst + 4, ret1, 4);

  for (size_t i = 0; i < 8; i++) {
      printf("%f ", dst[i]);
  }

  printf("\n");

  return 0;
}

   10508:       0087f7d7                vsetvli a5,a5,e32,m1,d1
   1050c:       181c                    addi    a5,sp,48
   1050e:       0207f207                vle.v   v4,(a5)
   10512:       1004                    addi    s1,sp,32
   10514:       009c                    addi    a5,sp,64
   10516:       0207f087                vle.v   v1,(a5)
   1051a:       0204f107                vle.v   v2,(s1)
   1051e:       089c                    addi    a5,sp,80
   10520:       0207f187                vle.v   v3,(a5)
   10524:       02221157                vfadd.vv        v2,v2,v4
   10528:       021190d7                vfadd.vv        v1,v1,v3
   1052c:       e802                    sd      zero,16(sp)
   1052e:       ec02                    sd      zero,24(sp)
   10530:       e002                    sd      zero,0(sp)
   10532:       e402                    sd      zero,8(sp)
   10534:       081c                    addi    a5,sp,16
   10536:       840a                    mv      s0,sp
   10538:       6941                    lui     s2,0x10
   1053a:       02017127                vse.v   v2,(sp)
   1053e:       0207f0a7                vse.v   v1,(a5)
   10542:       00042787                flw     fa5,0(s0) # ffffffffffffd000 <__global_pointer$+0xfffffffffffea800>
   10546:       67090513                addi    a0,s2,1648 # 10670 <__libc_csu_fini+0x4>
   1054a:       420787d3                fcvt.d.s        fa5,fa5
   1054e:       0411                    addi    s0,s0,4
   10550:       e20785d3                fmv.x.d a1,fa5
   10554:       f6dff0ef                jal     ra,104c0 <printf@plt>
   10558:       fe8495e3                bne     s1,s0,10542 <main+0x72>
   1055c:       4529                    li      a0,10
   1055e:       f53ff0ef                jal     ra,104b0 <putchar@plt>
   10562:       70e6                    ld      ra,120(sp)
   10564:       7446                    ld      s0,112(sp)
   10566:       74a6                    ld      s1,104(sp)
   10568:       7906                    ld      s2,96(sp)
   1056a:       4501                    li      a0,0
   1056c:       6109                    addi    sp

use vcreate_
int main() {

[... 102 lines cut ...]
   10616:       c2102ff3                csrr    t6,vtype
   1061a:       00807057                vsetvli zero,zero,e32,m1,d1
   1061e:       5e0081d7                vmv.v.v v3,v1
   10622:       81ff7057                vsetvl  zero,t5,t6
   10626:       32002057                vmv.x.s zero,v0
   1062a:       c2002f73                csrr    t5,vl
   1062e:       c2102ff3                csrr    t6,vtype
   10632:       00807057                vsetvli zero,zero,e32,m1,d1
   10636:       5e010257                vmv.v.v v4,v2
   1063a:       81ff7057                vsetvl  zero,t5,t6
   1063e:       32002057                vmv.x.s zero,v0
   10642:       c2002f73                csrr    t5,vl
   10646:       c2102ff3                csrr    t6,vtype
   1064a:       00807057                vsetvli zero,zero,e32,m1,d1
   1064e:       5e0180d7                vmv.v.v v1,v3
   10652:       81ff7057                vsetvl  zero,t5,t6
   10656:       32002057                vmv.x.s zero,v0
   1065a:       02017227                vse.v   v4,(sp)
   1065e:       0207f0a7                vse.v   v1,(a5)
   10662:       00042787                flw     fa5,0(s0) # ffffffffffffd000 <__global_pointer$+0xfffffffffffea800>
   10666:       79090513                addi    a0,s2,1936 # 10790 <__libc_csu_fini+0x4>
   1066a:       420787d3                fcvt.d.s        fa5,fa5
   1066e:       0411                    addi    s0,s0,4
   10670:       e20785d3                fmv.x.d a1,fa5
   10674:       e4dff0ef                jal     ra,104c0 <printf@plt>

haolongzhangm, 2022-08-02:
* riscv64-unknown-linux-gnu-g++ -march=rv64gcv0p7 -mabi=lp64d,  we cat find when use "vcreate_" gcc will create so much no need vector ,eg  v8 v9....,    and performance is so pooling 

* when build with -march=rv64gcv, do not find this issue when use "vcreate_"  , but now, there so so many board are v07 rvv, so  
need fix issue

zhongjuzhe, 2022-08-03:
RVV0.7 is no longer maintained. The latest actively maintained RVV GCC is the branch "rvv-next" which support the official RVV1.0 intrinsics and auto-vectorization. Same situation for LLVM.

haolongzhangm, 2022-08-03:
@zhongjuzhe but there are so many board just support v0.7 standard， if build with 1.0 , will crash caused by unkown instruct
how to fix this

what is more, so many software always have LTS version, which means need fix issue on this version

eg. now latest ubuntu is 22.04,  but community also fix issue find on 20.04,

what is more, linux kernel is the same logic

what   riscv-gnu-toolchain do not?   

just ask you use rvv-next or latest is not a good idea.

what is more, the requirement is not ask to modify rvv 0.7 doc-spec, is fix riscv-gnu-toolchain optimize issue

cmuellner, 2022-08-04:
In general non-upstream branches (incl. the one mentioned for RVV 0.7) are not supported/maintained.
The RVV 0.7 specification is not a ratified extension (but we all know it is implemented in available HW).
Given the support for the ratified RVV 1.0 extension is not upstream as of now, there is no chance of getting RVV 0.7 upstream. Therefore nobody is looking at this at the moment.

haolongzhangm, 2022-08-04:
no need release RVV 0.7 , just fix the optimize issue when user set -march=rv64gcv0p7 at  latest release (rvv-next?)

I think this is a optpass issue, not  RVV 0.7 issue

cmuellner, 2022-08-04:
As I said, nobody is working with these branches anymore as everyone is working on branches that target upstream.

haolongzhangm, 2022-08-05:
so "target upstream" can not do the fix?

kito-cheng, 2022-08-05:
Not sure which toolchain you are usign, but currently we are targeting rvv 1.0 only for upstream, which is the FSF GCC, is not upstream yet, but that review in progress.

haolongzhangm, 2022-08-05:
@kito-cheng  may u can just test the test code with "-march=rv64gcv0p7"  and "-march=rv64gcv"

kito-cheng, 2022-08-08:
@haolongzhangm could you post your gcc version info by `riscv64-unknown-elf-gcc --version`? that sounds like a T-head toolchain?

haolongzhangm, 2022-08-08:
@kito-cheng 

riscv64-unknown-linux-gnu-g++ (Xuantie-900 linux-5.10.4 glibc gcc Toolchain V2.2.6 B-20220516) 10.2.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

kito-cheng, 2022-08-09:
Hmmm, I think you might report to T-head guys rather than report here, @rjiejie

haolongzhangm, 2022-08-09:
@kito-cheng  so thanks for you suggest

@rjiejie need i create a new issue or not ? or just follow this one

rjiejie, 2022-08-15:
> @kito-cheng so thanks for you suggest
> 
> @rjiejie need i create a new issue or not ? or just follow this one

Follow this is ok also, and we do some optimization for it.
we will ship this optimization in version v2.6.1，
you can download toolchain from https://occ.t-head.cn/community/download?id=4073475960903634944 if it released :)

haolongzhangm, 2022-08-16:
@rjiejie   I test Xuantie-900-gcc-linux-5.10.4-glibc-x86_64-V2.6.0-20220715.tar.gz ,  issue more severe
* when i use old Xuantie-900-gcc-linux-5.10.4-glibc-x86_64-V2.2.6-20220516.tar.gz, when with -march=rv64gcv  have no issue , with -march=rv64gcv07   have issue
* when i test Xuantie-900-gcc-linux-5.10.4-glibc-x86_64-V2.6.0-20220715.tar.gz, both  -march=rv64gcv  and -march=rv64gcv07 have issue ................

u can test the example code at the head of this issue

haolongzhangm, 2022-08-16:
so any base test code for rvv api  performance test?

haolongzhangm, 2022-08-16:
what`s more, Xuantie-900-gcc-linux-5.10.4-glibc-x86_64-V2.6.0-20220715.tar.gz upgrade  libc depends to GLIBC_2.33 ? 

* I thinks upgrade depends libc version is so bad idea,  so many user boads or system, for example , fedora, ubuntu, do not upgrade system libs quickly, or no chance to upgrade.
* do not ask user build with --static , which always a so bad idea

malaterre, 2022-09-01:
> The latest actively maintained RVV GCC is the branch "rvv-next" which support the official RVV1.0 intrinsics and auto-vectorization.

For some reason I can find the old rvv-0.7x branch but none seems to match "rvv-next":

* https://github.com/riscv-collab/riscv-gcc/tree/rvv-0.7.x

Could someone please guide me on where is the latest branch (one in progress for merged in upstream GCC).

Thanks

zhongjuzhe, 2022-09-01:
> 

You should pull latest riscv gnu toolchain, there is no rvv-0.7x branch any more. 
The only RVV branch is called "rvv-next" in riscv-gnu-toolchain.
And you also should pull the latest "riscv-gcc-rvv-next" branch in riscv-gcc.

The rvv-next support full feature RVV1.0.  And it is going to merged in upstream GCC.
I am working on splitting rvv-next into small patches to upstream GCC so that reviewers can easily review that.
So far there are part of the rvv-next codes already merged in upstream GCC. But for full support, you still need
to use rvv-next.

TommyMurphyTM1234, 2022-09-27:
Questions relating to non-standard pre-ratification v0.0.7 vector extension used/implemented in some RISC-V cores (see here for example: https://github.com/riscv/riscv-v-spec/issues/667) and how to build the latest `rvv-next` branch, answered.

closed 2022-09-27 by TommyMurphyTM1234