riscv-collab/riscv-gnu-toolchain #1109: Missing .git directory in riscv-gnu-toolchain tarball causing make failed

wanlinwang, 2022-08-09:
Hi there,

Missing .git directory in riscv-gnu-toolchain tarball causing make failed. Could you include the .git directory when making the tarball?

SpriteOvO, 2022-08-09:
It is common practice to not include the `.git` directory in tarball. Would you mind sharing the detailed reproduction steps and error messages?

wanlinwang, 2022-08-09:
> It is common practice to not include the `.git` directory in tarball. Would you mind sharing the detailed reproduction steps and error messages?

[image]

wanlinwang, 2022-08-17:
> It looks like you're downloading the "Source code" item from the release page, which is automatically generated by GitHub at the time of release, and AFAIK modifying it is not possible.
> 
> 
> 
> Instead, you should use
> 
> 
> 
> ```bash
> 
> git clone https://github.com/riscv-collab/riscv-gnu-toolchain.git --branch 2022.08.08 --depth 1
> 
> ```
> 
> 
> 
> as [described in the README](https://github.com/riscv-collab/riscv-gnu-toolchain#getting-the-sources), and it works fine. Submodules will be automatically fetched on demand during build time and usually do not require `--recursive` or manual handling.

OK. Thanks.

TommyMurphyTM1234, 2022-09-27:
Incorrect approach to building the toolchain. Not an issue with the RISC-V GCC toolchain or build process.

closed 2022-09-27 by TommyMurphyTM1234