riscv-collab/riscv-gnu-toolchain #1110: GCC option 'print-multi-lib' cannot show the correct infomation?

xicongye, 2022-08-10:
1. steps:
```
$ cd riscv-gnu-toolchain
$ ./configure --enable-multilib && make
```

2. test
```
$ riscv64-unknown-elf-gcc --print-multi-lib
.;
rv32i/ilp32;@march=rv32i@mabi=ilp32
rv32im/ilp32;@march=rv32im@mabi=ilp32
rv32iac/ilp32;@march=rv32iac@mabi=ilp32
rv32imac/ilp32;@march=rv32imac@mabi=ilp32
rv32imafc/ilp32f;@march=rv32imafc@mabi=ilp32f
rv64imac/lp64;@march=rv64imac@mabi=lp64
```

3. conclusion

`rv64imafdc-lp64d` is lost. I check `configure.ac`, the default multilib is:
```
AS_IF([test "x$enable_multilib" != xno],
         [AC_SUBST(newlib_multilib_names,"rv32i-ilp32 rv32iac-ilp32 rv32im-ilp32 rv32imac-ilp32 rv32imafc-ilp32f rv64imac-lp64 rv64imafdc-lp64d")],
         [AC_SUBST(newlib_multilib_names,"$with_arch-$with_abi")])
```

TommyMurphyTM1234, 2022-08-11:
What is your initial `git clone` command?
Can you attach (not post) your build log?
Is there a reason that you're building from this repository rather than from upstream gcc etc.?

xicongye, 2022-08-11:
Hi, @TommyMurphyTM1234 

the build.log size is too large to upload, sorry about that.
maybe you can reproduce it on `Ubuntu20.04`, here is my build steps:

my `git clone` command is:
```
$ git clone https://github.com/riscv-collab/riscv-gnu-toolchain.git
$ cd riscv-gnu-toolchain
$ git submodule init
$ git submodule update
```

I use git to check my repository status:
```
$ cd riscv-gnu-toolchain
$ git submodule status 
 9826b03b747b841f5fc6de2054bf1ef3f5c4bdf3 glibc (glibc-2.33)
 85e0e3519655220688e757b9d5bfd314923548bd musl (v1.2.2)
 415fdd4279b85eeec9d54775ce13c5c412451e08 newlib (newlib-4.1.0)
 2efabd3e6604b8a9e8f70baf52f57696680c7855 pk (v1.0.0-89-g2efabd3)
 823a3f11fb8f04c3c3cc0f95f968fef1bfc6534f qemu (v7.0.0)
 20756b0fbe065a84710aa38f2457563b57546440 riscv-binutils (heads/riscv-binutils-2.38)
 4ea498a8e1fafeb568530d84db1880066478c86b riscv-dejagnu (heads/riscv-dejagnu-1.6)
 5964b5cd72721186ea2195a7be8d40cfe6554023 riscv-gcc (remotes/origin/riscv-gcc-11.1.0)
 5da071ef0965b8054310d8dde9975037b0467311 riscv-gdb (remotes/origin/fsf-gdb-10.1-with-sim)
 a0298a33e7b2091ba8d9f3a20838d96dc1164cac spike (v1.1.0-212-ga0298a33)
$ git status .
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean

$ git log -1
commit cb25bb862a3bf56d1577d7930bc41f259632ae24 (HEAD -> master, tag: 2022.08.08, origin/master, origin/HEAD)
Merge: 409b951 0824fc5
Author: Kito Cheng <kito.cheng@sifive.com>
Date:   Mon Aug 8 10:02:39 2022 +0800

    Merge pull request #1107 from wanlinwang/patch-1
    
    Update .gitmodules
```

building:

```
$ cd riscv-gnu-toolchain
$ ./configure --enable-multilib --prefix=xxx/install && make
```

run `--print-multi-lib` option:
```
$ cd xxx/install
$ ./riscv64-unknown-elf-gcc --print-multi-lib
.;
rv32i/ilp32;@march=rv32i@mabi=ilp32
rv32im/ilp32;@march=rv32im@mabi=ilp32
rv32iac/ilp32;@march=rv32iac@mabi=ilp32
rv32imac/ilp32;@march=rv32imac@mabi=ilp32
rv32imafc/ilp32f;@march=rv32imafc@mabi=ilp32f
rv64imac/lp64;@march=rv64imac@mabi=lp64
```

TommyMurphyTM1234, 2022-08-13:
Yes, I get the same results as you.
I haven't looked at the log file in detail or otherwise investigated further yet, but will do so soon.

TommyMurphyTM1234, 2022-08-13:
It looks like something has changed with the build process over recent months (maybe years?). In the past `--enable-multilib`, for a newlib/bare metal build of the tools, used to build the set of multilibs specified in the file `t-elf-multilib`

https://github.com/riscv-collab/riscv-gcc/blob/riscv-gcc-10.2.0/gcc/config/riscv/t-elf-multilib

But now it looks like the build calls `genmultilib` instead

https://github.com/riscv-collab/riscv-gcc/blob/riscv-gcc-10.2.0/gcc/genmultilib

But I haven't figured out yet how it arrives at the list of RISC-V newlib multilibs that actually gets built.

Maybe somebody with better knowledge of the build process might know offhand?

TommyMurphyTM1234, 2022-08-14:
> $ riscv64-unknown-elf-gcc --print-multi-lib
> .;
> rv32i/ilp32;@march=rv32i@mabi=ilp32
> rv32im/ilp32;@march=rv32im@mabi=ilp32
> rv32iac/ilp32;@march=rv32iac@mabi=ilp32 [...]

Parking my confusion about how the list of multilibs to be built is controlled, I see now what your issue is here. `rv64imafdc-lp64d` is not actually lost but is implied by the `.;` at the start of the multilib list and is the default configuration when `-march=... -mabi=...` are not specified. Unless otherwise overridden at configuration time, the toolchain defaults to `rv64imafdc-lp64d` aka `rv64gc-lp64d`, so `.;` in the multilib list implies that arch/abi combination.

xicongye, 2022-08-15:
> rv64imafdc-lp64d is not actually lost but is implied by the .; at the start of the multilib list and is the default configuration when -march=... -mabi=... are not specified. Unless otherwise [...]

Agree. But there is one more question, how do we know the default configuration about ` -march/-mabi` ?

TommyMurphyTM1234, 2022-08-15:
> Agree. But there is one more question, how do we know the default configuration about ` -march/-mabi` ?

One way is to run `riscv64-unknown-elf-gcc -dumpspecs` and you should get

```
...

*multilib_defaults:
march=rv64imafdc mabi=lp64d

...
```

Also, the configure or build logs when building the tools should show that these are the defaults.

TommyMurphyTM1234, 2022-08-15:
> It looks like something has changed with the build process over recent months (maybe years?). In the past `--enable-multilib`, for a newlib/bare metal build of the tools, used to build the set of [...]

I think I've figured it out.
To override the default set of multilibs specified in `configure.ac` I was able to use `./configure ... --with-multilib-generator="rv32i-ilp32--;rv32imafd-ilp32--"` which built these two and the compiler default mode `rv64imafdc-lp64d--`. As far as I can see `t-elf-multilib` is no longer used, the default set of multilibs does not match those specified there, and the file does not get updated at configure/build time?

TommyMurphyTM1234, 2022-09-27:
Closing this issue as the original confusion has been explained.

closed 2022-09-27 by TommyMurphyTM1234