riscv-collab/riscv-gnu-toolchain #1111: configure: error:  *** These critical programs are missing or too old: make during make linux

Omar-M-Yehia, 2022-08-10:
I have the following issue when I run make linux during toolchain build 

`checking build system type... x86_64-pc-linux-gnu
checking host system type... riscv64-unknown-linux-gnu
checking for riscv64-unknown-linux-gnu-gcc... riscv64-unknown-linux-gnu-gcc 
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether riscv64-unknown-linux-gnu-gcc  accepts -g... yes
checking for gcc... gcc
checking for riscv64-unknown-linux-gnu-readelf... riscv64-unknown-linux-gnu-readelf
checking whether we are using the GNU C++ compiler... no
checking whether this-is-not-the-compiler-youre-looking-for accepts -g... no
checking whether this-is-not-the-compiler-youre-looking-for can link programs... no
checking for sysdeps preconfigure fragments... aarch64 alpha arc arm csky hppa i386 m68k microblaze checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
mips nios2 powerpc riscv s390 sh checking for grep that handles long lines and -e... (cached) /bin/grep
checking for egrep... (cached) /bin/grep -E
sparc x86_64 
checking for use of fpu sysdeps directories... yes
checking for -fstack-protector... yes
checking for -fstack-protector-strong... yes
checking for -fstack-protector-all... yes
checking for assembler and linker STT_GNU_IFUNC support... yes
checking for gcc attribute ifunc support... yes
checking whether the linker supports textrels along with ifunc... yes
checking if compiler warns about alias for function with incompatible types... yes
checking sysdep dirs... sysdeps/unix/sysv/linux/riscv/rv64 sysdeps/unix/sysv/linux/riscv sysdeps/riscv/nptl sysdeps/unix/sysv/linux/generic sysdeps/unix/sysv/linux/wordsize-64 sysdeps/unix/sysv/linux sysdeps/nptl sysdeps/pthread sysdeps/gnu sysdeps/unix/inet sysdeps/unix/sysv sysdeps/unix sysdeps/posix sysdeps/riscv/rv64/rvd sysdeps/riscv/rv64/rvf sysdeps/riscv/rvf sysdeps/riscv/rvd sysdeps/riscv/rv64 sysdeps/riscv sysdeps/ieee754/ldbl-128 sysdeps/ieee754/dbl-64 sysdeps/ieee754/flt-32 sysdeps/wordsize-64 sysdeps/ieee754 sysdeps/generic
checking for a BSD-compatible install... /bin/install -c
checking whether ln -s works... yes
checking whether /home/omyehia/RISCV/lib/gcc/riscv64-unknown-linux-gnu/11.1.0/../../../../riscv64-unknown-linux-gnu/bin/as is GNU as... yes
checking whether /home/omyehia/RISCV/lib/gcc/riscv64-unknown-linux-gnu/11.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld is GNU ld... yes
checking for /home/omyehia/RISCV/lib/gcc/riscv64-unknown-linux-gnu/11.1.0/../../../../riscv64-unknown-linux-gnu/bin/as... /home/omyehia/RISCV/lib/gcc/riscv64-unknown-linux-gnu/11.1.0/../../../../riscv64-unknown-linux-gnu/bin/as
checking version of /home/omyehia/RISCV/lib/gcc/riscv64-unknown-linux-gnu/11.1.0/../../../../riscv64-unknown-linux-gnu/bin/as... 2.38, ok
checking for /home/omyehia/RISCV/lib/gcc/riscv64-unknown-linux-gnu/11.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld... /home/omyehia/RISCV/lib/gcc/riscv64-unknown-linux-gnu/11.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld
checking version of /home/omyehia/RISCV/lib/gcc/riscv64-unknown-linux-gnu/11.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld... 2.38, ok
checking for gnumake... no
checking for gmake... gmake
checking version of gmake... 3.82, bad
checking for gnumsgfmt... no
checking for gmsgfmt... no
checking for msgfmt... msgfmt
checking version of msgfmt... 0.19.8.1, ok
checking for makeinfo... makeinfo
checking version of makeinfo... 5.1, ok
checking for sed... /bin/sed
checking version of /bin/sed... 4.2.2, ok
checking for gawk... /bin/gawk
checking version of /bin/gawk... 4.0.2, ok
checking for bison... bison
checking version of bison... 3.0.4, ok
checking if riscv64-unknown-linux-gnu-gcc  is sufficient to build libc... yes
checking for riscv64-unknown-linux-gnu-nm... riscv64-unknown-linux-gnu-nm
checking for python3... python3
checking version of python3... 3.6.1, ok
configure: error: 
*** These critical programs are missing or too old: make
*** Check the INSTALL file for required versions.
make: *** [stamps/build-glibc-linux-rv64imafdc-lp64d] Error 1
`

 I installed the Developer Toolset , and running make linux have same issue

cmuellner, 2022-08-10:
Have you enabled the devtoolset?
See the instructions here: https://www.softwarecollections.org/en/scls/rhscl/devtoolset-7/

Omar-M-Yehia, 2022-08-10:
Yes I followed these instructions , it should be enabled and it doesn't work 

checking for gmsgfmt... no
checking for msgfmt... msgfmt
checking version of msgfmt... 0.19.8.1, ok
checking for makeinfo... makeinfo
checking version of makeinfo... 5.1, ok
checking for sed... /bin/sed
checking version of /bin/sed... 4.2.2, ok
checking for gawk... /bin/gawk
checking version of /bin/gawk... 4.0.2, ok
checking for bison... bison
checking version of bison... 3.0.4, ok
checking if riscv64-unknown-linux-gnu-gcc  is sufficient to build libc... yes
checking for riscv64-unknown-linux-gnu-nm... riscv64-unknown-linux-gnu-nm
checking for python3... python3
checking version of python3... 3.6.1, ok
configure: error: 
*** These critical programs are missing or too old: make
*** Check the INSTALL file for required versions.
make: *** [stamps/build-glibc-linux-rv64imafdc-lp64d] Error 1
**bash-4.2$** sudo yum list devtoolset-7\*
Loaded plugins: langpacks
Installed Packages
devtoolset-7.x86_64                                                                                                     7.1-4.el7                                                                                      @centos-sclo-rh
devtoolset-7-binutils.x86_64                                                                                            2.28-11.el7                                                                                    @centos-sclo-rh
devtoolset-7-dwz.x86_64                                                                                                 0.12-1.1.el7                                                                                   @centos-sclo-rh
devtoolset-7-dyninst.x86_64                                                                                             9.3.2-3.el7                                                                                    @centos-sclo-rh
devtoolset-7-elfutils.x86_64                                                                                            0.170-5.el7                                                                                    @centos-sclo-rh
devtoolset-7-elfutils-libelf.x86_64                                                                                     0.170-5.el7                                                                                    @centos-sclo-rh
devtoolset-7-elfutils-libs.x86_64                                                                                       0.170-5.el7                                                                                    @centos-sclo-rh
devtoolset-7-gcc.x86_64                                                                                                 7.3.1-5.16.el7                                                                                 @centos-sclo-rh
devtoolset-7-gcc-c++.x86_64                                                                                             7.3.1-5.16.el7                                                                                 @centos-sclo-rh
devtoolset-7-gcc-gfortran.x86_64                                                                                        7.3.1-5.16.el7                                                                                 @centos-sclo-rh

cmuellner, 2022-08-10:
Again, you need to enable the devtoolset as described in the linked page, not only install it:
```
$ gmake -v
GNU Make 3.82
Built for x86_64-redhat-linux-gnu
Copyright (C) 2010  Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
$ scl enable devtoolset-7 bash
$ gmake -v
GNU Make 4.2.1
Built for x86_64-redhat-linux-gnu
Copyright (C) 1988-2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

Omar-M-Yehia, 2022-08-10:
I opened a new terminal and enabled devtoolset and it is worked, thanks

closed 2022-08-10 by Omar-M-Yehia