riscv-collab/riscv-gnu-toolchain #1112: Does --with-multilib-generator render multilib-generator/t-*-multilib redundant?

TommyMurphyTM1234, 2022-08-15:
Apropos of this

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1110#issuecomment-1214883135

and the documentation about the use of `--with-multilib-generator` when configuring a RISC-V build of GCC:

* https://gcc.gnu.org/install/configure.html
* https://github.com/riscv-collab/riscv-gnu-toolchain#build-with-customized-multi-lib-configure

Does this mean that the following are now deprecated/redundant?

* https://github.com/riscv-collab/riscv-gcc/blob/riscv-gcc-10.2.0/gcc/config/riscv/multilib-generator
* https://github.com/riscv-collab/riscv-gcc/blob/riscv-gcc-10.2.0/gcc/config/riscv/t-elf-multilib
* https://github.com/riscv-collab/riscv-gcc/blob/riscv-gcc-10.2.0/gcc/config/riscv/t-linux-multilib

If so, should the be removed?
Is there anything else in this folder that is deprecated/redundant and which should be removed?

* https://github.com/riscv-collab/riscv-gcc/blob/riscv-gcc-10.2.0/gcc/config/riscv

kito-cheng, 2022-08-16:
That's mainly for designed for bare-metal target and didn't well verified with linux target, so t-linux-multilib still the recommended way in linux, and I suppose all extension specific optimization should be done via IFUNC in glibc.

For the bare-metal: Yes, --with-multilib-generator is the encourage way to configure with the multi-lib, I think we might consider retired t-elf-multilib and setup that by setting default value of `--with-multilib-generator` to deal with the different ISA spec version issue.

TommyMurphyTM1234, 2022-08-16:
Thanks a lot @kito-cheng
Maybe you could clarify something else for me please?
Since the `riscv-gnu-toolchain` repo is an "umbrella" repo that assists with building the full RISC-V toolchain including gcc, binutilts, gdb, newlib/glibc etc. what, if anything, is the analog/equivalent upstream in which the `--with-multilib-generator` changes to `configure.ac` will be upstreamed?
Or will they eventually be reflected in the gcc `configure.ac`?

https://gcc.gnu.org/git?p=gcc.git;a=blob;f=configure.ac;h=371def9b42122953255ca8a10c8ce93e590abf81;hb=HEAD

Or am I misunderstanding something? (Quite possible! 🙂)

Thanks

TommyMurphyTM1234, 2022-08-16:
Thanks @kito-cheng - so I take it that is not upstreamed in any shape or form right now? The reason I'm confused is that previously when building from upstream gcc et. al. repos (using [crosstool-NG](https://crosstool-ng.github.io/)  as it happens), when I enabled a multilib build I would get the list of multilibs specified in `t-elf-multilib`. But more recent builds, in the last week or so, using CT-NG result in the same smaller set of multilibs now specified by default in this repo's `configure.ac`. I'm still trawling the CT-NG build logs to try to understand how it arrives at the set of multilibs that it builds by default and how it, also, now seems to ignore `t-elf-multilib`.

TommyMurphyTM1234, 2022-08-16:
Since this issue is largely moot until such time as the `--with-multilib-generator` mechanism is upstreamed, in some shape or form, and any concomitant cleanups of the `gcc/config/riscv` directory are effected, I am closing the issue now. Thanks again @kito-cheng for your contributions. 👍

closed 2022-08-16 by TommyMurphyTM1234