riscv-collab/riscv-gnu-toolchain #1113: unrecognized opcode `c.li s4,-13'

billasurendra, 2022-08-17:
I am trying to cross compile linux-5.16.9 with the gcc-11.2.0 cross compiler (rv64imafd). Here I am getting following error.

Error:

arch/riscv/kernel/head.S: Assembler messages:
arch/riscv/kernel/head.S:51: Error: unrecognized opcode `c.li s4,-13'
make[2]: *** [scripts/Makefile.build:388: arch/riscv/kernel/head.o] Error 1
make[1]: *** [scripts/Makefile.build:549: arch/riscv/kernel] Error 2
make: *** [Makefile:1846: arch/riscv] Error 2

Please help me to resolve above error.

Thanks
Billa

TommyMurphyTM1234, 2022-08-17:
Your compilation target rv64imafd doesn't support the C compressed extension but the assembly source file in the Linux source tree is using a compressed instruction. I seem to recall that most or all RISC-V Linux ports target rvXXgc. So you'll either need to compile for a C capable target (i.e. rv64gc here) or try porting Linux to target non C devices. The former is obviously simpler.

billasurendra, 2022-08-17:
If I have cross compiled linux-5.4.1 version successfully with gcc-7.3.1
version (rv64imfd) then whey I can't compile latest version ?

TommyMurphyTM1234, 2022-08-17:
> If I have cross compiled linux-5.4.1 version successfully with gcc-7.3.1 version (rv64imfd) then whey I can't compile latest version ?

Maybe that version didn't use C compressed instructions?

TommyMurphyTM1234, 2022-08-17:
I can't find any Linux 5.4.1 tag but the 5.4 tag doesn't have any compressed instructions in the file that's causing you problems now.

https://github.com/torvalds/linux/blob/v5.4/arch/riscv/kernel/head.S

billasurendra, 2022-08-17:
My riscv hardware  is rv64imafd, so please tell me, How I can compile the
latest version of linux to rv64imafd.

TommyMurphyTM1234, 2022-08-17:
Maybe you could compile Linux with `CONFIG_EFI` disabled (presumably via the kernel menuconfig or configuration files)

https://github.com/torvalds/linux/blob/3cc40a443a04d52b0c95255dce264068b01e9bfe/arch/riscv/kernel/head.S#L28

But

1. I have no idea what impact this would have on the resulting kernel
2. There may be other compressed instructions elsewhere in the Linux RISC-V source tree

As per the discussion link that I posted above, you may need to port Linux (and related packages?) to support non C `rv64imafd` given that it seems that most or all existing ports expect C compressed instruction support.

TommyMurphyTM1234, 2022-08-17:
For what it's worth, the compressed instruction was added to that file here

https://github.com/torvalds/linux/commit/cb7d2dd5612a77a2597c00fce770a52c921e2ea5

billasurendra, 2022-08-17:
I have deleted **c.li s4, -13** line from head.S file then linux successfully compiled. is it correct way or not ?

TommyMurphyTM1234, 2022-08-17:
> I have deleted **c.li s4, -13** line from head.S file then linux successfully compiled. is it correct way or not ?

Arbitrarily deleting code to make things compile is obviously not the correct way to port code!

kito-cheng, 2022-08-18:
Yeah, I think it's kind of linux kernel issue rather than toolchain issue, I would strongly suggest you could drop a mail to RISC-V kernel[1] mailing list.

[1] https://lists.infradead.org/mailman/listinfo/linux-riscv

TommyMurphyTM1234, 2022-08-19:
Further discussion of this issue here, including clarification of why, after all, it's impossible to avoid the compressed `c.li s4, -13` instruction when the EFI kernel option is enabled.

https://groups.google.com/a/groups.riscv.org/g/sw-dev/c/6Ko22JeAlF0/m/lpvHa1E8AgAJ?utm_medium=email&utm_source=footer

TommyMurphyTM1234, 2022-09-27:
The original issue has been explained so closing this issue as resolved.

closed 2022-09-27 by TommyMurphyTM1234