riscv-collab/riscv-gnu-toolchain #1115: Macros defined in _newlib_version.h are not filled with actual value

yuzdoge, 2022-08-22:
Platform: Ubuntu22.04 for wsl2

I am trying to check some Macros defined in `_newlib_version.h`,  but I found that they are not filled with actual value. The following is the code snippet from the file  `/opt/riscv-newlib/riscv64-unknown-elf/include/_newlib_version.h`.
```c
/* _newlib_version.h.  Generated from _newlib_version.hin by configure.  */
/* Version macros for internal and downstream use. */
#ifndef _NEWLIB_VERSION_H__
#define _NEWLIB_VERSION_H__ 1

/* The newlib version in string format. */
#define _NEWLIB_VERSION "@NEWLIB_VERSION@"

/* The newlib major version number. */
#define __NEWLIB__ @NEWLIB_MAJOR_VERSION@

/* The newlib minor version number. */
#define __NEWLIB_MINOR__ @NEWLIB_MINOR_VERSION@

/* The newlib patch level. */
#define __NEWLIB_PATCHLEVEL__ @NEWLIB_PATCHLEVEL_VERSION@

#endif /* !_NEWLIB_VERSION_H__ */
```
I follow [Installation (Newlib/Linux multilib)](https://github.com/riscv-collab/riscv-gnu-toolchain#installation-newliblinux-multilib) to install the toolchain. According to the comment  `_newlib_version.h.  Generated from _newlib_version.hin by configure.`, I think that `@xxx@` should be replaced when I `make`. Do I neglect anything?

Thanks.

kito-cheng, 2022-08-22:
Not sure what's wrong? but I saw the _newlib_version.h in https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2022.08.19 has right pre-processed?

```
/* _newlib_version.h.  Generated from _newlib_version.hin by configure.  */
/* Version macros for internal and downstream use. */
#ifndef _NEWLIB_VERSION_H__
#define _NEWLIB_VERSION_H__ 1

#define _NEWLIB_VERSION "4.1.0"
#define __NEWLIB__ 4
#define __NEWLIB_MINOR__ 1
#define __NEWLIB_PATCHLEVEL__ 0

#endif /* !_NEWLIB_VERSION_H__ */

```

yuzdoge, 2022-08-22:
> Not sure what's wrong? but I saw the _newlib_version.h in https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2022.08.19 has right pre-processed?
>
> ```
> /* _newlib_version.h.  Generated [...]

Does it come from `riscv64-elf-ubuntu-20.04-nightly-2022.08.19-nightly.tar.gz` ?

closed 2022-08-22 by yuzdoge

TommyMurphyTM1234, 2022-08-22:
I've just completed a build of the tools from this repo and do not have the issue reported in the original post.

```
user@hornbill:~/Downloads/riscv-collab/riscv-gnu-toolchain-install$ ./configure --prefix=$HOME/Downloads/riscv-collab/riscv-gnu-toolchain-install
user@hornbill:~/Downloads/riscv-collab/riscv-gnu-toolchain-install$ make
user@hornbill:~/Downloads/riscv-collab/riscv-gnu-toolchain-install$ cd ../riscv-gnu-toolchain-install/
user@hornbill:~/Downloads/riscv-collab/riscv-gnu-toolchain-install$ find . -name "_newlib_version.h" -print
./riscv64-unknown-elf/include/_newlib_version.h
user@hornbill:~/Downloads/riscv-collab/riscv-gnu-toolchain-install$ cat riscv64-unknown-elf/include/_newlib_version.h 
/* _newlib_version.h.  Generated from _newlib_version.hin by configure.  */
/* Version macros for internal and downstream use. */
#ifndef _NEWLIB_VERSION_H__
#define _NEWLIB_VERSION_H__ 1

#define _NEWLIB_VERSION "4.1.0"
#define __NEWLIB__ 4
#define __NEWLIB_MINOR__ 1
#define __NEWLIB_PATCHLEVEL__ 0

#endif /* !_NEWLIB_VERSION_H__ */

```