riscv-collab/riscv-gnu-toolchain #1117: Wrong submodule name in gitmodules

AlpherJang, 2022-08-23:
When I install riscv-gnu-toolchain latest version with brew, I got an error with it, the error info like this:
==> Installing riscv/riscv/riscv-tools dependency: riscv-gnu-toolchain
==> git submodule update --init --recursive newlib
==> git submodule update --init --recursive riscv-binutils
Last 15 lines from /Users/evans/Library/Logs/Homebrew/riscv-gnu-toolchain/02.git:
2022-08-23 07:34:00 +0000

git
submodule
update
--init
--recursive
riscv-binutils

error: pathspec 'riscv-binutils' did not match any file(s) known to git

TommyMurphyTM1234, 2022-08-23:
I did a build from this repo about 12 hours ago and didn't have any issues. That was on Zorin OS (Ubuntu derivative). It doesn't look like anything changed overnight:

https://github.com/riscv-collab/riscv-gnu-toolchain/commits/master

So are you sure that it's not simply due to an intermittent/sporadic repo access problem?

Have you tried the build again?

AlpherJang, 2022-08-23:
> I did a build from this repo using the master branch about 12 hours ago and didn't have any issues. That was on Zorin OS (Ubuntu derivative). It doesn't look like anything changed overnight (Irish time):
> 
> https://github.com/riscv-collab/riscv-gnu-toolchain/commits/master
> 
> So are you sure that it's not simply due to an intermittent/sporadic repo access problem?
> 
> Have you tried the build again? Are you building from a branch other than master?

I have try to build many times, and always use master branch, maybe we can build it with ourself, but not suit for homebrew

pz9115, 2022-08-23:
The riscv-binutils had updated into binutils with upstream branch in .gitmodules, I had tested this before and it was passed.

Can you use follow commands and offer the info with git submodules?
```
git submodule sync

git submodule
```

TommyMurphyTM1234, 2022-08-23:
> The riscv-binutils had updated into binutils with upstream branch in .gitmodules, I had tested this before and it was passed.

@pz9115 - I presume that you're referring to this change in the last few hours? 

https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1116/files

I hadn't noticed that when replying above.

pz9115, 2022-08-23:
> @pz9115 - I presume that you're referring to this change in the last few hours?
>
> https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1116/files
>
> I hadn't noticed that when replying above.

Sorry, But which I pointed is this PR:

https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1108

It had been merged for few days.

AlpherJang, 2022-08-23:
I found the reason ,there has some problem in riscv-software-src/homebrew-riscv, in that repo update submodule with old name

pz9115, 2022-08-23:
> I found the reason ,there has some problem in riscv-software-src/homebrew-riscv, in that repo update submodule with old name

@AlpherJang 
I had opened a PR to fix it, thanks for your report.

AlpherJang, 2022-08-23:
maybe there exists same error with riscv-gcc, we need to fix this command

TommyMurphyTM1234, 2022-08-23:
> maybe there exists same error with riscv-gcc, we need to fix this command

Isn't `riscv-gcc` correct?

https://github.com/riscv-software-src/homebrew-riscv/blob/63b502d20210a278eacf8b6b6bae8b7c91e65c22/riscv-gnu-toolchain.rb#L56

https://github.com/riscv-collab/riscv-gnu-toolchain/blob/c39061968972ac2e4a10844aa54b90c8fb94125e/.gitmodules#L5

AlpherJang, 2022-08-24:
> sn't `riscv-gcc` correct

it should be change to gcc, not riscv-gcc

kito-cheng, 2022-08-24:
Fixed via https://github.com/riscv-collab/riscv-gnu-toolchain/commit/bbc2dbb6b3e1909bf4c94fa66f125e9b27d3e7a8

Vishal-Chdhry, 2022-08-25:
```
Last 15 lines from /Users/vishal/Library/Logs/Homebrew/riscv-gnu-toolchain/02.git:
2022-08-25 17:42:00 +0000

git
submodule
update
--init
--recursive
riscv-binutils

error: pathspec 'riscv-binutils' did not match any file(s) known to git

Do not report this issue to Homebrew/brew or Homebrew/core!

These open issues may also help:
Sync the binutils name with riscv-gnu-toolchain repo https://github.com/riscv-software-src/homebrew-riscv/pull/91
```
I am still getting this issue

TommyMurphyTM1234, 2022-08-25:
> ```
> Last 15 lines from /Users/vishal/Library/Logs/Homebrew/riscv-gnu-toolchain/02.git:
> 2022-08-25 17:42:00 +0000
>
> git
> submodule
> update
> --init
> --recursive
> riscv-binutils
>
> error: pathspec [...]

Isn't that an issue with the `homebrew-riscv` repo:

* https://github.com/riscv-software-src/homebrew-riscv

And already reported here?

* https://github.com/riscv-software-src/homebrew-riscv/issues/92
* https://github.com/riscv-software-src/homebrew-riscv/issues/90

Vishal-Chdhry, 2022-08-26:
> Isn't that an issue with the homebrew-riscv repo:
yeah, and the PR regarding this: https://github.com/riscv-software-src/homebrew-riscv/pull/91 pointed me to this issue

and the problem is basically the same, naming of riscv-binutils

the PR might have fixed it, but it needs to get merged

AlpherJang, 2022-08-26:
> e PR might have fixed it, but it needs to ge

yes, I had reported it with [https://github.com/riscv-software-src/homebrew-riscv/issues/90](https://github.com/riscv-software-src/homebrew-riscv/issues/90)

brandaogbs, 2022-08-28:
I've trying to install the `riscv-gnu-toolchain` (dependence for`riscv-tools`) on MB M1 (Apple Silicon) and had the same mentioned error. I've check the rename of two submodules `riscv-binutils`-> `binutils` and `riscv-gcc`-> `gcc`. 

I've tested the following steps using Homebrew:

1. Edit the brew formula: `brew edit riscv-gnu-toolchain`;
2. Rename the two submodules mentioned (remove the `riscv-` prefix):
```
 # need to pull in needed submodules (now that they are disabled above)
 system "git", "submodule", "update", "--init", "--recursive", "newlib"
 system "git", "submodule", "update", "--init", "--recursive", "binutils"
 system "git", "submodule", "update", "--init", "--recursive", "gcc"
```
3. Fix the "workaround for M1" paths:
```
# Workaround for M1
# See https://github.com/riscv/homebrew-riscv/issues/47
system "sed", "-i", ".bak", "s/.*=host-darwin.o$//", "gcc/gcc/config.host"
system "sed", "-i", ".bak", "s/.* x-darwin.$//", "gcc/gcc/config.host"
```
4. Then, if everything goes right it should be installed with: `brew install riscv-tools`.

I'll report on the Homebrew issue [riscv-software-src/homebrew-riscv/issues/90](https://github.com/riscv-software-src/homebrew-riscv/issues/90)

pz9115, 2022-08-29:
I had update the fix patch in  PR, thanks. 
https://github.com/riscv-software-src/homebrew-riscv/pull/91/files

LukeSamkharadze, 2022-09-20:
> I've trying to install the `riscv-gnu-toolchain` (dependence for`riscv-tools`) on MB M1 (Apple Silicon) and had the same mentioned error.
> 
> I've check the rename of two submodules `riscv-binutils`-> `binutils` and `riscv-gcc`-> `gcc`. I've tested the following steps using Homebrew:
> 
> 1. Edit the brew formula: `brew edit riscv-gnu-toolchain`;
> 2. Rename the two submodules mentioned (remove the `riscv-` prefix):
> 
> ```
>  # need to pull in needed submodules (now that they are disabled above)
>  system "git", "submodule", "update", "--init", "--recursive", "newlib"
>  system "git", "submodule", "update", "--init", "--recursive", "binutils"
>  system "git", "submodule", "update", "--init", "--recursive", "gcc"
> ```
> 
>  3. Fix the "workaround for M1" paths:
> 
> ```
> # Workaround for M1
> # See https://github.com/riscv/homebrew-riscv/issues/47
> system "sed", "-i", ".bak", "s/.*=host-darwin.o$//", "gcc/gcc/config.host"
> system "sed", "-i", ".bak", "s/.* x-darwin.$//", "gcc/gcc/config.host"
> ```
> 
>  4. Then, if everything went right, it should be installed with: `brew install riscv-tools`.
> 
> I'll report on the Homebrew issue [riscv-software-src/homebrew-riscv/issues/90](https://github.com/riscv-software-src/homebrew-riscv/issues/90)

Did you install `riscv-gnu-toolchain` seperately?

I did those changes, but still fails on `brew install riscv-tools` and it took about 1 hour

```
==> Installing riscv-software-src/riscv/riscv-tools dependency: riscv-isa-s
==> ../configure --prefix=/opt/homebrew/Cellar/riscv-isa-sim/main --with-target=../../riscv
==> make
Last 15 lines from /Users/lukesamkharadze/Library/Logs/Homebrew/riscv-isa-sim/02.make:
ld: in libspike_main.a(libriscv.a), archive member 'libriscv.a' with length 11617464 is not mach-o or llvm bitcode file 'libspike_main.a' for architecture arm64
ld: in libspike_main.a(libriscv.a), archive member 'libriscv.a' with length 11617464 is not mach-o or llvm bitcode file 'libspike_main.a' for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
clang: error: linker command failed with exit code 1 (use -v to see invocation)
ld: in libspike_main.a(libriscv.a), archive member 'libriscv.a' with length 11617464 is not mach-o or llvm bitcode file 'libspike_main.a' for architecture arm64
make: *** [spike] Error 1
make: *** [libcustomext.so] Error 1
ld: in libspike_dasm.a(libriscv.a), archive member 'libriscv.a' with length 11617464 is not mach-o or llvm bitcode file 'libspike_dasm.a' for architecture arm64
ld: in libspike_main.a(libriscv.a), archive member 'libriscv.a' with length 11617464 is not mach-o or llvm bitcode file 'libspike_main.a' for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [spike-dasm] Error 1
make: *** [spike-log-parser] Error 1
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [termios-xspike] Error 1

Do not report this issue to Homebrew/brew or Homebrew/core!
```

LukeSamkharadze, 2022-09-24:
Issue was fixed by downgrading `Command Line Tools for Xcode` to `13.4` and applying those `riscv-gnu-toolchain` changes

khokho, 2022-09-25:
Related issue to the problem @LukeSamkharadze was facing. Their fix might be useful context as well.
https://trac.macports.org/ticket/65838

grejioh, 2022-09-28:
> > I've trying to install the `riscv-gnu-toolchain` (dependence for`riscv-tools`) on MB M1 (Apple Silicon) and had the same mentioned error.
> > I've check the rename of two submodules `riscv-binutils`-> `binutils` and `riscv-gcc`-> `gcc`. I've tested the following steps using Homebrew:
> > 
> > 1. Edit the brew formula: `brew edit riscv-gnu-toolchain`;
> > 2. Rename the two submodules mentioned (remove the `riscv-` prefix):
> > 
> > ```
> >  # need to pull in needed submodules (now that they are disabled above)
> >  system "git", "submodule", "update", "--init", "--recursive", "newlib"
> >  system "git", "submodule", "update", "--init", "--recursive", "binutils"
> >  system "git", "submodule", "update", "--init", "--recursive", "gcc"
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 
> > 3. Fix the "workaround for M1" paths:
> > 
> > ```
> > # Workaround for M1
> > # See https://github.com/riscv/homebrew-riscv/issues/47
> > system "sed", "-i", ".bak", "s/.*=host-darwin.o$//", "gcc/gcc/config.host"
> > system "sed", "-i", ".bak", "s/.* x-darwin.$//", "gcc/gcc/config.host"
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 
> > 4. Then, if everything went right, it should be installed with: `brew install riscv-tools`.
> > 
> > I'll report on the Homebrew issue [riscv-software-src/homebrew-riscv/issues/90](https://github.com/riscv-software-src/homebrew-riscv/issues/90)
> 
> @brandaogbs did you install `riscv-gnu-toolchain` seperately?
> 
> I did those changes, but still fails on `brew install riscv-tools` and it took about 1 hour
> 
> ```
> ==> Installing riscv-software-src/riscv/riscv-tools dependency: riscv-isa-s
> ==> ../configure --prefix=/opt/homebrew/Cellar/riscv-isa-sim/main --with-target=../../riscv
> ==> make
> Last 15 lines from /Users/lukesamkharadze/Library/Logs/Homebrew/riscv-isa-sim/02.make:
> ld: in libspike_main.a(libriscv.a), archive member 'libriscv.a' with length 11617464 is not mach-o or llvm bitcode file 'libspike_main.a' for architecture arm64
> ld: in libspike_main.a(libriscv.a), archive member 'libriscv.a' with length 11617464 is not mach-o or llvm bitcode file 'libspike_main.a' for architecture arm64
> clang: error: linker command failed with exit code 1 (use -v to see invocation)
> clang: error: linker command failed with exit code 1 (use -v to see invocation)
> ld: in libspike_main.a(libriscv.a), archive member 'libriscv.a' with length 11617464 is not mach-o or llvm bitcode file 'libspike_main.a' for architecture arm64
> make: *** [spike] Error 1
> make: *** [libcustomext.so] Error 1
> ld: in libspike_dasm.a(libriscv.a), archive member 'libriscv.a' with length 11617464 is not mach-o or llvm bitcode file 'libspike_dasm.a' for architecture arm64
> ld: in libspike_main.a(libriscv.a), archive member 'libriscv.a' with length 11617464 is not mach-o or llvm bitcode file 'libspike_main.a' for architecture arm64
> clang: error: linker command failed with exit code 1 (use -v to see invocation)
> clang: error: linker command failed with exit code 1 (use -v to see invocation)
> make: *** [spike-dasm] Error 1
> make: *** [spike-log-parser] Error 1
> clang: error: linker command failed with exit code 1 (use -v to see invocation)
> make: *** [termios-xspike] Error 1
> 
> Do not report this issue to Homebrew/brew or Homebrew/core!
> ```

This issue seems to be fixed here.
[riscv-isa-sim/pull/1104](https://github.com/riscv-software-src/riscv-isa-sim/pull/1104#issue-1388975462)

TommyMurphyTM1234, 2022-09-29:
Can somebody with access to a macOS platform clarify if the issues raised here are now resolved so that this issue can be closed, please? Thanks.

aswaterman, 2022-09-29:
@TommyMurphyTM1234 AFAICS, there are two unrelated issues being discussed here.  The linker failure was a problem with the Spike Makefile, which I can confirm has been fixed.  The other, as you pointed out earlier, is a reported issue with homebrew-riscv.  IMO, we can close this issue in good conscience, as both issues are either fixed or reported to the appropriate place.

TommyMurphyTM1234, 2022-09-29:
Thanks a lot @aswaterman

closed 2022-09-29 by TommyMurphyTM1234