riscv-collab/riscv-gnu-toolchain #1118: [Question] ssh authentication for gitmodules

Incarnation-p-lee, 2022-08-23:
I notice there are some sub module of this repo, which take ssh authentication (well-known starts with `git://`) from their url(s). Given most of sub module leverage `https`, I am not sure why such kinds of sub modules take ssh.

As I understand, ssh authentication requires additional ad-hoc steps, for example, add `id_rsa.pub` to remote git host server or stuff like that. I can see there may be some unnecessary trouble shoot effort for each potential contributors or developers.

So may I proposal to change the url of these sub modules to `https` based? Please feel free to correct me if there is any misunderstanding, Thank you!

Incarnation-p-lee, 2022-08-23:
Looks there is one PR for this already but out of date, #1063.

kito-cheng, 2022-08-23:
`git://sourceware.org/git/*.git` should able access without any permission? so I am not sure what cause the problem?

TommyMurphyTM1234, 2022-08-23:
> I notice there are some sub module of this repo, which take ssh authentication (well-known starts with `git://`) from their url(s). 

I presume that you mean these ones?

* glibc: https://github.com/riscv-collab/riscv-gnu-toolchain/blob/c39061968972ac2e4a10844aa54b90c8fb94125e/.gitmodules#L11
* newlib: https://github.com/riscv-collab/riscv-gnu-toolchain/blob/c39061968972ac2e4a10844aa54b90c8fb94125e/.gitmodules#L18
* musl: https://github.com/riscv-collab/riscv-gnu-toolchain/blob/c39061968972ac2e4a10844aa54b90c8fb94125e/.gitmodules#L29

Perhaps these should be changed to 

* glibc: https://sourceware.org/git/glibc.git (https://www.gnu.org/software/libc/sources.html)
* newlib: no change (https://sourceware.org/newlib/download.html)
* musl: https://git.musl-libc.org/cgit (https://musl.libc.org/)

Incarnation-p-lee, 2022-08-24:
> `git://sourceware.org/git/*.git` should able access without any permission? so I am not sure what cause the problem?

Not sure if it related to the network environment, but I can have a try from my home network later. But for `git`, do I still need to add the `id_rsa.pub`?

Incarnation-p-lee, 2022-08-24:
> I presume that you mean these ones?
>
> * glibc: https://github.com/riscv-collab/riscv-gnu-toolchain/blob/c39061968972ac2e4a10844aa54b90c8fb94125e/.gitmodules#L11
> * newlib: [...]

Yes, these 3 repos. 
* glic => https://sourceware.org/git/glibc.git
* newlib => https://sourceware.org/git/newlib-cygwin.git
* musl => https://git.musl-libc.org/cgit

TommyMurphyTM1234, 2022-08-24:
> But for `git`, do I still need to add the `id_rsa.pub`?

I haven't had to do that in order to build the tools from this repo successfully.

Incarnation-p-lee, 2022-08-24:
> > But for `git`, do I still need to add the `id_rsa.pub`?
> 
> I haven't had to do that in order to build the tools from this repo successfully.

Yes, you are right. It may related to my dev environment network(which may be restricted by http proxy) I guess. I can build the tools include `SIM=spike` successfully from github action.

TommyMurphyTM1234, 2022-08-24:
> Yes, these 3 repos.
> 
> * glic => https://sourceware.org/git/glibc.git
> * newlib => https://sourceware.org/git/newlib-cygwin.git
> * musl => https://git.musl-libc.org/cgit

I'm not sure that the newlib one should be changed since the repo recommended by the project is `git://sourceware.org/git/newlib-cygwin.git`.

https://sourceware.org/newlib/download.html

In contrast, the glibc and musl projects do provide `https://` URLs.

Incarnation-p-lee, 2022-08-24:
> I'm not sure that the newlib one should be changed since the repo recommended by the project is `git://sourceware.org/git/newlib-cygwin.git`.
>
> https://sourceware.org/newlib/download.html
>
> In [...]

The `newlib` provides both `git`, `ssh` and `https` for the URLs [here](https://sourceware.org/git/newlib-cygwin.git).

[image]

Incarnation-p-lee, 2022-08-26:
@kito-cheng and @TommyMurphyTM1234 As the https switch PR merged, I will close this issue for now.

closed 2022-08-26 by Incarnation-p-lee