riscv-collab/riscv-gnu-toolchain #1120: How to run gcc testsuite for riscv32 using QEMU?

shubhamnarlawar77, 2022-08-24:
I have build riscv-gnu-toolchain. Then, I have build QEMU for riscv32 on x86 machine using below commands -

../configure --target-list=riscv32-softmmu --prefix=$HOME/riscv-gnu-toolchain/qemu/install

make -j12

How to run gcc testsuite for riscv32 using QEMU?

kito-cheng, 2022-08-24:
I would suggest you just using `make check` within the build folder of riscv-gnu-toolchain, that will build qemu and use a simple wrapper to run gcc testsuite with qemu.

And there is a configure option: `--with-qemu-src` for riscv-gnu-toolchain, that can specify your customized qemu source code if you don't want the built-in one.

TommyMurphyTM1234, 2022-09-29:
Question seems to have been adequately answered.

closed 2022-09-29 by TommyMurphyTM1234

shubhamnarlawar77, 2022-10-21:
> I would suggest you just using `make check` within the build folder of riscv-gnu-toolchain, that will build qemu and use a simple wrapper to run gcc testsuite with qemu.
>
> And there is a configure [...]

Your suggestion worked and it had build successfully.

Thanks for the reply!