riscv-collab/riscv-gnu-toolchain #1122: __sync_bool_compare_and_swap error while compilation

amr-25, 2022-08-25:
Hi,

I am compiling a c++ source using the command, riscv32-linux-g++ -march=rv32ima -mabi=ilp32 -std=c++11 -O3 -Wall -fopenmp src.c -o output. I get the following error posted below. One of the header files use _sync_bool_compare_and_swap which is causing the error. But it works fine when using g++ instead of riscv-linux-g++, Any thought on this? Thank you

/home/user/Litex/buildroot/output/host/lib/gcc/riscv32-buildroot-linux-gnu/10.4.0/../../../../riscv32-buildroot-linux-gnu/bin/ld: /tmp/ccYBJso0.o: in function `.L38':
bfs.cc:(.text+0x2f0): undefined reference to `__sync_bool_compare_and_swap_8'
/home/user/Litex/buildroot/output/host/lib/gcc/riscv32-buildroot-linux-gnu/10.4.0/../../../../riscv32-buildroot-linux-gnu/bin/ld: /tmp/ccYBJso0.o: in function `.L185':
bfs.cc:(.text._ZN11BuilderBaseIiiiLb1EE7MakeCSRERK7pvectorI8EdgePairIiiEEbPPPiS8_._omp_fn.0[_ZN11BuilderBaseIiiiLb1EE9MakeGraphEv]+0xb0): undefined reference to `__sync_fetch_and_add_8'
/home/user/Litex/buildroot/output/host/lib/gcc/riscv32-buildroot-linux-gnu/10.4.0/../../../../riscv32-buildroot-linux-gnu/bin/ld: /tmp/ccYBJso0.o: in function `.L170':
bfs.cc:(.text._ZN11BuilderBaseIiiiLb1EE7MakeCSRERK7pvectorI8EdgePairIiiEEbPPPiS8_._omp_fn.0[_ZN11BuilderBaseIiiiLb1EE9MakeGraphEv]+0x104): undefined reference to `__sync_fetch_and_add_8'
/home/user/Litex/buildroot/output/host/lib/gcc/riscv32-buildroot-linux-gnu/10.4.0/../../../../riscv32-buildroot-linux-gnu/bin/ld: /tmp/ccYBJso0.o: in function `.L184':
bfs.cc:(.text._ZN11BuilderBaseIiiiLb1EE7MakeCSRERK7pvectorI8EdgePairIiiEEbPPPiS8_._omp_fn.0[_ZN11BuilderBaseIiiiLb1EE9MakeGraphEv]+0x15c): undefined reference to `__sync_fetch_and_add_8'
collect2: error: ld returned 1 exit status

kito-cheng, 2022-08-25:
Could you try `-latomic` ?

amr-25, 2022-08-25:
yes. It shows the same error. Also, just mentioning that the compiler(riscv32-linux-g++) is built through buildroot linux.

cmuellner, 2022-08-25:
Does this also happen if you use GCC' atomic builtins instead of the legacy sync builtins?
See https://gcc.gnu.org/onlinedocs/gcc/_005f_005fatomic-Builtins.html

amr-25, 2022-08-25:
No it doesn't happen. Errors are gone. I don't understand the extra arguments such as memory_ordering in the GCC atomic builtins. I need to learn what those mean, but can it be used as an alternative? If yes, what is the difference between the sync builtins and gcc builtins? Thanks

closed 2022-08-30 by amr-25