riscv-collab/riscv-gnu-toolchain #1123: Floating Point Square Root

HamzaShabbir517, 2022-08-25:
Hello,
I have configured the risc-v toolchain using --enable-multilib option. As we have a hardware that support Floating Point Division and Square Root instruction. While running a C program which contain  math library i am getting the following error shown in snap below. The command which i am using to compile the program are as follow:

GCC_PREFIX=riscv64-unknown-elf
$(GCC_PREFIX)-gcc -Wl,-m,elf32lriscv -Wl,--discard-none -T$(LINK) -o $(TEST).exe $(OFILES) -nostartfiles -lm $(TEST_LIBS)
$(GCC_PREFIX)-objcopy -O verilog  $(TEST).exe program.hex

[image]

Please provide the solution for this error.

cmuellner, 2022-08-25:
Have you tried with `GCC_PREFIX=riscv32-unknown-elf`?

HamzaShabbir517, 2022-08-25:
yes i tried but for that also i got lots of error. Below is the snap of the errors
[image]

cmuellner, 2022-08-25:
Before you had a mismatch of rv32 and rv64.
Now you have a mismatch of the floating-point ABI.

Did you add the flags `-march=rv32imafc -mabi=ilp32f`?

HamzaShabbir517, 2022-08-25:
yes i used the following cflags while compiling the C code. i have also mention the ABI as mention below

TEST_CFLAGS = -mabi=ilp32f -march=rv32imafc -g
ABI = -mabi=ilp32f -march=rv32imafc

HamzaShabbir517, 2022-08-25:
just for better understanding here is the snap of Makefile which build the test using compiler
[image]

TommyMurphyTM1234, 2022-08-25:
I suspect that the relevant multilibs were not built so the linker falls back to the default libs for the toolchain and these are incompatible?

How did you configure your toolchain?

What do you get from

`riscv32-unknown-elf-gcc --print-multi-lib`

And

`riscv32-unknown-elf-gcc -dumpspecs`

TommyMurphyTM1234, 2022-08-25:
When linking, I also think that you should also be passing `-march=rv32imafc -mabi=ilp32f` instead of `-Wl,-m,elf32lriscv`.

TommyMurphyTM1234, 2022-08-25:
If you are always targeting only `rv32imafc/ilp32f` then it might be simpler to just build a toolchain that targets that arch/abi by default rather than building a multilib toolchain?

`./configure --with-arch=rv32imafc --with-abi=ilp32f --prefix=...`

Depending on what version of the tools you are building you may also need to take cognizance of the earlier separation of the `Zicsr` and `Zifencei` extensions out from the I integer base extension:

`./configure --with-arch=rv32imafc_Zicsr_Zifencei --with-abi=ilp32f --prefix=...`

HamzaShabbir517, 2022-08-25:
Actually my main target is RV32GCV that's why i build the toolchain with multilib. After making some changes in Make file i was able to run the C program but now the issue is i am facing is that my hardware support Floating Point Square root instruction but the compiler is not using fsqrt.s instruction directly rather it is using multiple function to calculate the C program. Below are the snap of C code and the assembly file generated.
[image]
[image]
[image]
[image]

TommyMurphyTM1234, 2022-08-25:
> now the issue is i am facing is that my hardware support Floating Point Square root instruction but the compiler is not using fsqrt.s instruction directly rather it is using multiple function to calculate the C program. 

That probably means that the newlib used hasn't been "tuned" to use RISC-V floating point instructions when available.

HamzaShabbir517, 2022-08-25:
so what is the possible way to tuned it so that it can used fsqrt instructions ?

TommyMurphyTM1234, 2022-08-25:
I'm not sure if the details but you'd have to check the newlib project for info on how to port/tune it for a specific target. Or maybe somebody else has done some work on doing this for RISC-V?

I think that picolibc already has RISC-V FP support in case it's feasible to build that and use it instead of newlib?

https://github.com/picolibc/picolibc

HamzaShabbir517, 2022-08-25:
ok thanks. Do i have to build this on existing newlib or do i have to build it seperately?

TommyMurphyTM1234, 2022-08-25:
> ok thanks. Do i have to build this on existing newlib or do i have to build it separately?

I don't really understand your question but newlib and picolibc are two different C standard library implementations for embedded/resource constrained targets so you'll generally use one or the other. You should look at the picolibc project site/repo for information on how to build and use it.

closed 2022-08-25 by HamzaShabbir517

TommyMurphyTM1234, 2022-08-26:
Actually, I just noticed that newlib *does* have some support for floating point instructions here:

https://sourceware.org/git/?p=newlib-cygwin.git;a=tree;f=newlib/libm/machine/riscv;h=7643a9d51619f7c06dfc505b76dd42d302546820;hb=HEAD

For example:

https://sourceware.org/git/?p=newlib-cygwin.git;a=blob;f=newlib/libm/machine/riscv/ef_sqrt.c;h=1f378f547df436f31ed9d048d9fb7ec95bff2a1d;hb=HEAD

However, I'm not sure if/how `sqrtf()` should get mapped to `__ieee754_sqrtf()`?

HamzaShabbir517, 2022-08-27:
Yes after rebuilding the toolchain and using -lm flag i was able to generate the sqrt.s instruction. And you are also right it does not use sqrt directly. It used sqrt function in which there are other few instructions as well and from there it jump to --ieee754_sqrtf() and there it use fsqrt.s. 
[image]
Adding the snap for the reference

aswaterman, 2022-08-27:
All of those extra instructions are implied by -fmath-errno. They should disappear under -fno-math-errno, IIRC.

HamzaShabbir517, 2022-08-27:
is this some flag which can be used while compiling?

aswaterman, 2022-08-27:
Yeah, It's a gcc command-line argument.

merldsu, 2022-08-27:
ok thanks i will try it out .

Is there any way to add csr instructions in C program as i want to update my frm csr to test the same code on different rounding modes but the code generated from C does not include any csr instructions

HamzaShabbir517, 2022-08-27:
ok thanks i will try it out .

Is there any way to add csr instructions in C program as i want to update my frm csr to test the same code on different rounding modes but the code generated from C does not include any csr instructions

HamzaShabbir517, 2022-08-27:
@TommyMurphyTM1234 As i mention that i am targeting to get a toolchain which support RV32IMFCV as my hardware support all of these extensions. So to support the Vector Extension is the rvv-next branch is the correct branch to checkout and build a toolchain from that using enable-multilib as a configure parameter?

TommyMurphyTM1234, 2022-08-27:
> ok thanks i will try it out .
> 
> Is there any way to add csr instructions in C program as i want to update my frm csr to test the same code on different rounding modes but the code generated from C does not include any csr instructions

You can use gcc inline assembly code for this.
Something like this:

https://stackoverflow.com/questions/58947716/how-to-interact-with-risc-v-csrs-by-using-gcc-c-code

TommyMurphyTM1234, 2022-08-27:
> @TommyMurphyTM1234 As i mention that i am targeting to get a toolchain which support RV32IMFCV as my hardware support all of these extensions. So to support the Vector Extension is the rvv-next branch is the correct branch to checkout and build a toolchain from that using enable-multilib as a configure parameter? Will it be able to compile the C code for vector instructions?

You might want to look at this issue for guidance on building a V extension enabled gcc toolchain.

* https://github.com/riscv-collab/riscv-gcc/issues/353

If you are always targeting `rv32imfcv` then it might make sense to configure and build your toolchain as follows:

```
./configure --with-architecture=rv32imfcv --with-abi=ilp32f
make
```

Depending on the source base used (I'm not clear on what that is for RVV support) you might also need to take cognizance of the fact that the `Zicsr` and `Zifencei` extensions were split out from the original I (Integer) base extension recently - so you may need to do:

```
./configure --with-architecture=rv32imfcv_Zicsr_Zifencei --with-abi=ilp32f
make
```

HamzaShabbir517, 2022-08-27:
ok thanks i will look into it

HamzaShabbir517, 2022-08-27:
@TommyMurphyTM1234 i tried to build the toolchain as u mention above but it is giving me this error
[image]

TommyMurphyTM1234, 2022-08-27:
> @TommyMurphyTM1234 i tried to build the toolchain as u mention above but it is giving me this error [image]

How exactly?
What `configure` command did you use?
If you included `--with-arch=rv32imfcv_Zicsr_Zifencei` then maybe try it without `_Zicsr_Zifencei`.
As I said, I'm not exactly sure which source case/branch/commit requires this for CSR and `fencei` support.

HamzaShabbir517, 2022-08-28:
@TommyMurphyTM1234 yes i was able to build the toolchain. 

I have another question is there any way to tell the compiler to not to generate the specific instructions which are not supported by the hardware?

TommyMurphyTM1234, 2022-08-28:
> @TommyMurphyTM1234 yes i was able to build the toolchain.

So the previous issue that you mentioned is no longer a problem?

> I have another question is there any way to tell the compiler to not to generate the specific instructions which are not supported by the hardware?

Why do you want to do this if you're always targeting `rv32imfcv` and you built an `rv32imfcv` toolchain?

If you want to do this then you will need to build a multilib toolchain, you may need to adjust the default set of multilibs built depending on what you want to target, and then when compiling your code you'll need to specify the required architecture/abi using `-march=... -mabi= ...`.

HamzaShabbir517, 2022-08-28:
Let me rephrase my question i am not talking about other extensions. 
As we know there are many instruction in F extension like almost 25. Just suppose my hardware does not support FMUL instruction but if i compile the code the compiler will generate the FMUL instruction so is there any way i can stop that. Just a question.

TommyMurphyTM1234, 2022-08-28:
> Let me rephrase my question i am not talking about other extensions. As we know there are many instruction in F extension like almost 25. Just suppose my hardware does not support FMUL instruction but if i compile the code the compiler will generate the FMUL instruction so is there any way i can stop that. Just a question.
> 
> It like instruction within a suppprted extension is not supported.

No. If you have implemented a custom extension then you need to customise the toolchain yourself to support it. Implementing only a subset of a ratified standard extension, as you have done, is tantamount to implementing a custom extension.

HamzaShabbir517, 2022-08-28:
Thanks for clearing things out. Really appreciate.