riscv-collab/riscv-gnu-toolchain #1126: Cannot build multilib Linux toolchain for rv32imafd

Plociennik, 2022-09-15:
Dear all,

I need to compile the Linux multilib toolchain supporting the `rv32imafd/ilp32d` architecture/ABI combination. I tried several ways of doing this but cannot make this work. I am compiling on Ubuntu 22.04 with GCC 11.2.

When I do `./configure --enable-multilib` and then `make linux` at least I get a multilib Linux toolchain, but `-print-multi-lib` says that the following combinations are supported:

```
lib32/ilp32;@march=rv32imac@mabi=ilp32
lib32/ilp32d;@march=rv32imafdc@mabi=ilp32d
lib64/lp64;@march=rv64imac@mabi=lp64
```

so the combination `rv32imafd/ilp32d` I need is not there. I tried to edit the line

```
glibc_multilib_names="rv32imac-ilp32 rv32imafdc-ilp32d rv64imac-lp64 rv64imafdc-lp64d"
```

in `configure` so that it contains `rv32imafd-ilp32d` instead of `rv32imafdc-ilp32d` but I get the same set of supported multilibs as before, i.e., `-print-multi-lib` still outputs `rv32imafdc@mabi=ilp32d` instead of `rv32imafd@mabi=ilp32d`. Also, when I compile an executable using `riscv64-unknown-linux-gnu-g++ -march=rv32imafd -mabi=ilp32d` the created executable seems to still use compressed instructions, even though I do not specify the `c` flag.

I also wonder why there are four multilib names configured, but only three are output by `-print-multi-lib`.

I also tried the `--with-multilib-generator` option, but then `make linux` crashes with `home/kai/home/software/riscv-gnu-toolchain/build-gcc-linux-stage2/./gcc/xgcc ...` outputting

```
...sysroot/usr/include/gnu/stubs.h:8:11: fatal error: gnu/stubs-ilp32.h: No such file or directory
    8 | # include <gnu/stubs-ilp32.h>
```

I would greatly appreciate any hint on how I can build the multilib Linux toolchain supporting `rv32imafd/ilp32d` since for technical reasons in my project, I need this specific kind of build.

TommyMurphyTM1234, 2022-09-15:
> Dear all,
>
> I need to compile the Linux multilib toolchain supporting the `rv32imafd/ilp32d` architecture/ABI combination. I tried several ways of doing this but cannot make this work. I am compiling [...]

That's the default set of multilibs (including those shared between arch/abi combinations). 

* https://github.com/gcc-mirror/gcc/blob/master/gcc/config/riscv/t-elf-multilib

If you want a different set then you need to use the `multilib-generator` script to generate an updated `t-elf-multilib`.

* https://github.com/gcc-mirror/gcc/blob/master/gcc/config/riscv/multilib-generator

TommyMurphyTM1234, 2022-09-15:
The default set of multilibs is configured here.

* https://github.com/gcc-mirror/gcc/blob/master/gcc/config/riscv/t-linux-multilib

Note that several are shared by different arch/abi combinations.

There are actually 4 multilibs listed but one is referenced as `.` and represents the toolchain default arch/abi (usually `rv64gc/lp64`.

You probably need to create a custom `t-linux-multilib` using the `multilib-generator` script.

* https://github.com/gcc-mirror/gcc/blob/master/gcc/config/riscv/multilib-generator

TommyMurphyTM1234, 2022-09-15:
If a specific `rv32imafd/ilp32` toolchain will not suffice and you do need a multilib toolchain supporting these libs:

```
rv32imac-ilp32
rv32imafdc-ilp32d
rv64imac-lp64
rv64imafdc-lp64d
```

then, after cloning the sources but before building the toolchain do this:

```
cd riscv-gnu-toolchain
pushd .
cd gcc/gcc/config/riscv
./multilib-generator rv32imac-ilp32-- rv32imafdc-ilp32d-- rv64imac-lp64-- rv64imafdc-lp64d-- > t-linux-multilib
popd
./configure --enable-multilib --prefix=...
make linux
```

TommyMurphyTM1234, 2022-09-15:
Actually, for some reason when I do the build above it eventually fails when building glibc because, I think, it tries to build it using `-march=rv32imafdc -mabi=ilp32d` but this does not match the `rv32imafdc_zicsr/ilp32d` multilib that was built...
```
...
a - elf/dl-sym.os
a - elf/dl-error.os
a - elf/libc_early_init.os
riscv64-unknown-linux-gnu-gcc -march=rv32imafdc -mabi=ilp32d   -nostdlib -nostartfiles -r -o /home/user/Downloads/riscv-linux/riscv-gnu-toolchain/build-glibc-linux-rv32imafdc-ilp32d/elf/librtld.map.o -Wl,--defsym=calloc=0 -Wl,--defsym=free=0 -Wl,--defsym=malloc=0 -Wl,--defsym=realloc=0 -Wl,--defsym=__stack_chk_fail=0 -Wl,--defsym=__stack_chk_fail_local=0 \
	'-Wl,-(' /home/user/Downloads/riscv-linux/riscv-gnu-toolchain/build-glibc-linux-rv32imafdc-ilp32d/elf/dl-allobjs.os /home/user/Downloads/riscv-linux/riscv-gnu-toolchain/build-glibc-linux-rv32imafdc-ilp32d/libc_pic.a -lgcc '-Wl,-)' -Wl,-Map,/home/user/Downloads/riscv-linux/riscv-gnu-toolchain/build-glibc-linux-rv32imafdc-ilp32d/elf/librtld.mapT
/home/user/Downloads/riscv-linux/riscv-gnu-toolchain/riscv-linux-toolchain/lib/gcc/riscv64-unknown-linux-gnu/12.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/user/Downloads/riscv-linux/riscv-gnu-toolchain/riscv-linux-toolchain/lib/gcc/riscv64-unknown-linux-gnu/12.1.0/libgcc.a(div.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/home/user/Downloads/riscv-linux/riscv-gnu-toolchain/riscv-linux-toolchain/lib/gcc/riscv64-unknown-linux-gnu/12.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: failed to merge target specific data of file /home/user/Downloads/riscv-linux/riscv-gnu-toolchain/riscv-linux-toolchain/lib/gcc/riscv64-unknown-linux-gnu/12.1.0/libgcc.a(div.o)
/home/user/Downloads/riscv-linux/riscv-gnu-toolchain/riscv-linux-toolchain/lib/gcc/riscv64-unknown-linux-gnu/12.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: attempt to do relocatable link with elf64-littleriscv input and elf32-littleriscv output
/home/user/Downloads/riscv-linux/riscv-gnu-toolchain/riscv-linux-toolchain/lib/gcc/riscv64-unknown-linux-gnu/12.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/user/Downloads/riscv-linux/riscv-gnu-toolchain/riscv-linux-toolchain/lib/gcc/riscv64-unknown-linux-gnu/12.1.0/libgcc.a(div.o): file class ELFCLASS64 incompatible with ELFCLASS32
/home/user/Downloads/riscv-linux/riscv-gnu-toolchain/riscv-linux-toolchain/lib/gcc/riscv64-unknown-linux-gnu/12.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: final link failed: file in wrong format
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:541: /home/user/Downloads/riscv-linux/riscv-gnu-toolchain/build-glibc-linux-rv32imafdc-ilp32d/elf/librtld.map] Error 1
make[3]: Leaving directory '/home/user/Downloads/riscv-linux/riscv-gnu-toolchain/glibc/elf'
make[2]: *** [Makefile:479: elf/subdir_lib] Error 2
make[2]: Leaving directory '/home/user/Downloads/riscv-linux/riscv-gnu-toolchain/glibc'
make[1]: *** [Makefile:9: all] Error 2
make[1]: Leaving directory '/home/user/Downloads/riscv-linux/riscv-gnu-toolchain/build-glibc-linux-rv32imafdc-ilp32d'
make: *** [Makefile:353: stamps/build-glibc-linux-rv32imafdc-ilp32d] Error 2

```

Plociennik, 2022-09-16:
Thank you very much @TommyMurphyTM1234 for your detailed explanations and hints! This is very helpful for me, and I will try to make it work considering the things you mentioned.

For certain technical reasons, in our project, even though we will only be targeting `rv32imafd/ilp32d`, we actually need a multilib GCC. I hope I can make it work using your hints.

As you mentioned above, for me, the multilib glibc build also has crashed sometimes due to some compiler invocation using `rv32...c` even though `c` should not be present. I hence looked for `rv32imafdc` in the overall source code and found that in `glibc/scripts/build-many-glibcs.py`, some compiler invocations seem to be generated according to directory names if I understand correctly. So maybe besides making `t-linux-multilib` have the correct contents and specifying the right multilib names in top-level `configure`, one also needs to adapt `build-many-glibcs.py` so that the correct compiler flags are set. In the current form of that file I don't see the necessary `--with-arch=rv32imafd` for my case anywhere, so maybe this is the problem. I will try today.

Plociennik, 2022-09-16:
Short update/question: I tried to run `multilib-generator` and noticed that even with the same arguments that are reported to have been used for generating `t-linux-multilib` (at the beginning of that file), it generated a different output, including different `MULTILIB_DIRNAMES`, the use of the new `zicsr` flag, and different `MULTILIB_REUSE` values.

Hence, the `t-linux-multilib` file in the repo seems to have been generated with an older version of `multilib-generator`, but since the build works in principle with the "old style" `t-linux-multilib`  and since our target processor does not have `zicsr`, I tried to manually edit the `t-linux-multilib` file to make a `rv32imafd` version being built. However, this does not work yet, since only the `build-gcc-linux-stage1` target is being built, not the actual stage 2 targets. Maybe this has something to do with the fact that I edited the configurations in `build-many-glibcs.py` to provide a configuration for `rv32imafd` which seems to be missing, I don't know yet.

Now, a short question on the contents of the `t-linux-multilib` file: In the comments of the `multilib-generator` script, it says the following:

```
# Each argument to this script is of the form
#  <primary arch>-<abi>-<additional arches>-<extensions>
# Example 1:
#  rv32imafd-ilp32d-rv32g-c,v
# means that, in addition to rv32imafd, these configurations can also use the
# rv32imafd-ilp32d libraries: rv32imafdc, rv32imafdv, rv32g, rv32gc, rv32gv
```

This makes sense to me, since if I am e.g. targeting `rv32imafdc` when compiling user code, I can make use of `rv32imafd` libraries, since the latter do not contain compressed instructions, while the target processor would be able to execute them. Hence, the capability of the processor to execute these instructions is simply not used.

Now, let's have a look at the `t-elf-multilib` and `t-linux-multilib` files. In the former, I see the line

```
MULTILIB_REUSE = march.rv32i/mabi.ilp32=march.rv32ic/mabi.ilp32
```

which would mean that when targeting the "right hand side" architecture `rv32ic`, the libraries for the "left hand side" primary architecture `rv32i` can be used, so "compile target -> used libraries" goes from right to left here. All the lines under `MULTILIB_REUSE` follow this logic, since the right hand side architecture always has a superset of the instruction extensions of the left hand side one.

In `t-linux-multilib`, this seems to be reversed, since I can e.g. see the line

```
MULTILIB_REUSE = march.rv32imac/mabi.ilp32=march.rv32ima/mabi.ilp32
```

which using the same "right to left" logic as above would mean that when I target `rv32ima` when compiling user code, the libraries for the primary architecture `rv32imac` would be used. However, since the libraries contain compressed intructions and the processor I am targeting does not have this ISA extension, I think the targeted processor will not be able to execute the generated binary.

Do I misunderstand something here, or is this possibly a bug in the scripts? And also, do you know whether I have to re-run certain scripts or edit some other files consistently when editing `t-linux-multilib`?

Any help would be greatly appreciated.

TommyMurphyTM1234, 2022-09-16:
> Short update/question: I tried to run `multilib-generator` and noticed that even with the same arguments that are reported to have been used for generating `t-linux-multilib` (at the beginning of [...]

Yes, the default `t-elf-multilib` seems to predate some recent changes that reflect the separation of `zicsr` extension out from the base `I` (integer) ISA. But `multilib-generator` and the toolchain build scripts/makefiles seem to be up to date with this change. Hence there seems to be a mismatch.

> Now, let's have a look at the `t-elf-multilib` and `t-linux-multilib` files. In the former, I see the line
>
> ```
> MULTILIB_REUSE = march.rv32i/mabi.ilp32=march.rv32ic/mabi.ilp32
> ```
>
> which would mean [...]

I noticed that too and thought that I might be reading the "mapping" the wrong way, but it does seem to be wrong.

I'll hopefully get a chance later today to play with this again in case it helps at all.

Plociennik, 2022-09-16:
Thank you very much for your help!

I am working on this at the moment, if I find out anything helpful, I'll post it here.

Plociennik, 2022-09-19:
It seems I was able to make it work, doing two things.

Firstly, I changed the `glibc_multilib_names` line in `configure` to contain `rv32imafd-ilp32d` instead of `rv32imafdc-ilp32d`.

Secondly, I edited `gcc/gcc/config/riscv/t-linux-multilib`. I removed all `rv32` patterns from `MULTILIB_REUSE` (don't know whether this was relevant, maybe not) and I changed `MULTILIB_REQUIRED` to contain `march=rv32imafd/mabi=ilp32d` instead of `march=rv32imafdc/mabi=ilp32d`.

Using this, the build finishes and seems to correctly create what I want, namely a `rv32imafd` library instead of a `rv32imafdc` one.

TommyMurphyTM1234, 2022-09-19:
That's great @Plociennik. But it does seem like this is an issue in the build mechanism that ultimately needs a "proper" fix as there seems to be some mismatch between parts that are aware of the splitting of `zicsr` (and `zifencei`?) extensions out from the base `I` (integer) ISA/extension and parts that are not?

Plociennik, 2022-09-19:
Thanks for your comment! Yes, first of all it would be nice to at least have some clear description of how to add/change the targeted architecture/ABI combinations, since it was not trivial for me to build my required `rv32imafd/ilp32d` one. It seems that adding more complete support for this has been discussed, but it is technically hard to implement this in a general, configurable way, cf. e.g. [this issue](https://github.com/riscv-collab/riscv-gnu-toolchain/issues/425). Hence, at least having some short README describing the necessary steps for users would be helpful.

Secondly, you might be right when stating that there might be additional problems w.r.t. `zicsr` I did not yet notice. That's why I wrote "It seems I was able to make it work". As I found out, this is a question of RISCV ISA specification 2.0 vs 2.1 which in my case should not make any problems (?) but I am not sure. I will investigate this by trying to compile and link user code and execute it on our simulation platforms.

MarkGoncharovAl, 2023-03-01:
I want to add an example to show the problem connected with ```zicsr```.

I have multilib name, such as ```rv32imfc-ilp32f-rv32imfc_zifencei-zicsr```

If I use ```--with-multilib-generator``` option I have folder in 

```
riscv64-unknown-elf/lib: ls -d */*/
./rv32imfc/ilp32f
```

If I run ```python3 ./multilib-generator``` by hands and copy to config/riscv/t-elf-multilib, I have

```
riscv64-unknown-elf/lib: ls -d */*/
./rv32imfc_zicsr/ilp32f
```

I wanted to create independent issue, but it seems like this one is almost the same.

TommyMurphyTM1234, 2023-03-01:
> I want to add an example to show the problem connected with `zicsr`.
>
> I have multilib name, such as `rv32imfc-ilp32f-rv32imfc_zifencei-zicsr`
>
> If I use `--with-multilib-generator` option I have [...]

I'm not clear what your question is.
Please clarify.

There are still various problems extant due to the switch to canonicalised architecture names (e.g. `rv32gc` versus `rv32imafdc_zicsr_zifencei`) late last year and not all parts of the toolchain infrastructure deal with this correctly.
That may the problem that you're hitting, but I'm not really sure...

TommyMurphyTM1234, 2023-08-18:
> I noticed that too and thought that I might be reading the "mapping" the wrong way, but it does seem to be wrong.

FWIW I logged this GCC bug upstream on this issue:

* [Bug 111065](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=111065) - [RISCV] t-linux-multilib specifies incorrect multilib reuse patterns

TommyMurphyTM1234, 2023-10-15:
> If a specific `rv32imafd/ilp32d` toolchain will not suffice and you do need a multilib toolchain supporting these libs:
> 
> ```
> rv32imac-ilp32
> rv32imafdc-ilp32d
> rv64imac-lp64
> rv64imafdc-lp64d
> ```
> 
> then do this:
> 
> ```
> cd ~/Downloads
> git clone https://github.com/riscv/riscv-gnu-toolchain --recursive
> cd riscv-gnu-toolchain
> pushd .
> cd gcc/gcc/config/riscv
> ./multilib-generator rv32imac-ilp32-- rv32imafdc-ilp32d-- rv64imac-lp64-- rv64imafdc-lp64d-- > t-linux-multilib
> popd
> ./configure --enable-multilib --prefix=`pwd`/riscv-linux-toolchain
> make linux
> ```
> 
> Note that the `multilib-generator` command line above generates the following `t-linux-multilib` and the fact that hard float architectures have the `zicsr` extension automatically added. This may impact the `arch/abi` details that you need to pass when compiling:
> 
> ```
> # This file was generated by multilib-generator with the command:
> #  ./multilib-generator rv32imac-ilp32-- rv32imafdc-ilp32d-- rv64imac-lp64-- rv64imafdc-lp64d--
> MULTILIB_OPTIONS = march=rv32imac/march=rv32imafdc_zicsr/march=rv64imac/march=rv64imafdc_zicsr mabi=ilp32/mabi=ilp32d/mabi=lp64/mabi=lp64d 
> MULTILIB_DIRNAMES = rv32imac \
> rv32imafdc_zicsr \
> rv64imac \
> rv64imafdc_zicsr ilp32 \
> ilp32d \
> lp64 \
> lp64d 
> MULTILIB_REQUIRED = march=rv32imac/mabi=ilp32 \
> march=rv32imafdc_zicsr/mabi=ilp32d \
> march=rv64imac/mabi=lp64 \
> march=rv64imafdc_zicsr/mabi=lp64d
> MULTILIB_REUSE = 
> ```

Building the tools as above from the latest commit:

* https://github.com/riscv-collab/riscv-gnu-toolchain/commit/6b1324367b879a9b89437846827b48151b26b412

no longer results in this issue:

> Actually, for some reason when I do the build above it eventually fails when building glibc because, I think, it tries to build it using `-march=rv32imafdc -mabi=ilp32d` but this does not match the [...]

This is presumably a result of changes to the way that canonicalised architecture designators are handled and matched to the most appropriate compatible version when an exact match is not available.

As such, I am closing this issue off as resolved.

closed 2023-10-15 by TommyMurphyTM1234