riscv-collab/riscv-gnu-toolchain #1127: Errors when using members of std

kwonhyunjeong, 2022-09-15:
Dear developers,

I use riscv64-unknown-elf-g++ for cross compiling a simple c++ code as follows.
```
#include <iostream>
#include <cmath>
#include <mutex>
int main(){
        float a = 1;
        float b = 2;
        std::cout << std::fmin(a, b) << std::endl;

        std::mutex _mu;
        return 0;
}
```
When I compile the code using the following command,
```
riscv64-unknown-elf-g++ test.cpp -o test.o -std=c++11 -I/workdir/riscv/install/riscv64-unknown-elf/include/c++/12.1.0 -I/workdir/riscv/install/riscv64-unknown-elf/include/c++/12.1.0/riscv64-unknown-elf -I/workdir/riscv/install/riscv64-unknown
```
It yields errors related to std::fmin and std::mutex.
```
test.cpp: In function 'int main()':
test.cpp:7:27: error: 'fmin' is not a member of 'std'; did you mean 'min'?
    7 |         std::cout << std::fmin(a, b) << std::endl;
      |                           ^~~~
      |                           min
test.cpp:9:14: error: 'mutex' is not a member of 'std'
    9 |         std::mutex _mu;
      |              ^~~~~
test.cpp:4:1: note: 'std::mutex' is defined in header '<mutex>'; did you forget to '#include <mutex>'?
    3 | #include <mutex>
  +++ |+#include <mutex>
    4 | int main(){
```
I think it is because the members of std are defined only when certain variables are defined, such as _GLIBCXX_HAS_GTHREADS.

Can anyone recommend any methods or flags to be added to the command above?
Sincerely,

kwonhyunjeong, 2022-09-15:
Dear @TommyMurphyTM1234, It exactly yields the same errors as above.

TommyMurphyTM1234, 2022-09-15:
Ok, it'll be a few hours before I can check locally but maybe someone else will pitch in before then.

TommyMurphyTM1234, 2022-09-15:
OK - I get the same behaviour with the toolchain built from this repo and with the toolchain built using [crosstool-NG](https://crosstool-ng.github.io/) from all upstream sources.

cmuellner, 2022-09-15:
crosstool-ng RISC-V toolchains are also available via [compiler explorer](https://godbolt.org/).
The RV64 toolchains there are linux/glibc toolchains and the RV32 toolchains are ELF/newlib toolchains.
I can reproduce the issue with compiler explorer and all RV32 toolchains there.

The issue itself is not RISC-V specific (e.g. you can easily reproduce this with with an ARM ELF/newlib toolchain).
You basically already pinpointed the issue (`_GLIBCXX_HAS_GTHREADS` is not set when using newlib), which is out of the scope of this repo.

TommyMurphyTM1234, 2022-09-15:
> crosstool-ng RISC-V toolchains are also available via [compiler explorer](https://godbolt.org/). The RV64 toolchains there are linux/glibc toolchains and the RV32 toolchains are ELF/newlib toolchains. I can reproduce the issue with compiler explorer and all RV32 toolchains there.

Thanks @cmuellner - I just happened to have a CT-NG generated toolchain handy from some recent experimentation with CT-NG but it's good to be aware that such toolchains are available more conveniently for experimentation from Compiler Explorer. 👍

cylindrical2002, 2022-09-16:
> crosstool-ng RISC-V toolchains are also available via [compiler explorer](https://godbolt.org/). The RV64 toolchains there are linux/glibc toolchains and the RV32 toolchains are ELF/newlib toolchains. I can reproduce the issue with compiler explorer and all RV32 toolchains there.
> 
> The issue itself is not RISC-V specific (e.g. you can easily reproduce this with with an ARM ELF/newlib toolchain). You basically already pinpointed the issue ( is not set when using newlib), which is out of the scope of this repo.`_GLIBCXX_HAS_GTHREADS`

Sorry I don't quite understand you and I met the same problem. How can I set _GLIBCXX_HAS_GTHREADS when using newlib, do I need to compile it again? Or maybe I just need to add a few lines in my CMakeList.txt?

TommyMurphyTM1234, 2022-09-16:
@cylindrical2002 - does your code strictly need to be compiled with a bare metal/newlib toolchain or could you use a Linux toolchain? As @cmuellner says above, the latter will compile your code, but on the basis that any ultimate executable will be a Linux user space application program. The bare metal/newlib toolchain produces code/executables for running on bare metal/RTOS rather than Linux. A different way to pose my question is - what is your ultimate RISC-V target? If you do need to use the features in your code that depend on `_GLIBCXX_HAS_GTHREADS` then I get the impression from this and other previous similar discussions that there is no easy way to do that with the existing RISC-V bare metal/newlib toolchain.

cylindrical2002, 2022-09-18:
@TommyMurphyTM1234 Thank you for your answer. But as a beginner, I don't really understand you well. Maybe I need to rise a new issue for my problem. Thanks a lot again.

TommyMurphyTM1234, 2022-09-18:
> @TommyMurphyTM1234 Thank you for your answer. But as a beginner, I don't really understand you well. Maybe I need to rise a new issue for my problem. Thanks a lot again.

What is the nature of the RISC-V hardware/board or simulator that you're compiling code for? Is it bare metal or running an RTOS, or is it running Linux?

cylindrical2002, 2022-09-18:
@TommyMurphyTM1234 I am compiling it on Linux, maybe you can see #1129 for more information.

TommyMurphyTM1234, 2022-09-18:
> @TommyMurphyTM1234 I am compiling it on Linux, maybe you can see #1129 for more information.

Yes, but what's the nature of your RISC-V target that you're CROSS compiling for?

TommyMurphyTM1234, 2022-11-10:
Closing due to lack of updates/activity.

closed 2022-11-10 by TommyMurphyTM1234