riscv-collab/riscv-gnu-toolchain #1129: Problem when compiling a simple program with riscv-unknown-elf-g++.

cylindrical2002, 2022-09-18:
I compile a simple program with NumCpp for a simple test. But it pops up a lot of compilation errors. This my CMake and simple program.
```CMake
cmake_minimum_required(VERSION 3.14)
set (CMAKE_CXX_COMPILER "/home/ubuntu/RISCV/bin/riscv64/bin/riscv64-unknown-elf-g++")
set (CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++14")
project("rvtorch" CXX)
message("${CMAKE_CXX_FLAGS}")
add_executable(${PROJECT_NAME} main.cpp)
set (BOOST_ROOT "/home/ubuntu/boost/elfbin")
set (BOOST_INCLUDEDIR "${BOOST_ROOT}/include")
set (BOOST_LIBRARYDIR "${BOOST_ROOT}/lib")
set (BOOST_MIN_VERSION "1.80.0")
set (Boost_NO_BOOST_CMAKE ON)
FIND_PACKAGE(Boost ${BOOST_MIN_VERSION} REQUIRED)
find_package(NumCpp 2.6.2 REQUIRED)
target_link_libraries(${PROJECT_NAME}
        NumCpp::NumCpp
)
```

```C++
#include "NumCpp.hpp"
#include <cstdio>
#include <iostream>
using namespace std;
int main() {
    nc::NdArray<float> a = {{1, 2}, {3, 4}};
    nc::NdArray<float> b = {{5, 6}, {7, 8}};
    auto c = a * b;
    cout << c.shape() << endl;
    cout << c << endl;
    return 0;
}
```
It can pass the configuration of CMakeList.txt, and riscv64-unknown-elf-g++ is good in the environment (Ubuntu20.04 on WSL 2). But when I start to compile it, the error message poped up.

```
Scanning dependencies of target rvtorch
[ 50%] Building CXX object CMakeFiles/rvtorch.dir/main.cpp.o
In file included from /usr/local/include/NumCpp/Functions/deg2rad.hpp:30,
                 from /usr/local/include/NumCpp/Coordinates/Dec.hpp:36,
                 from /usr/local/include/NumCpp/Coordinates/Coordinate.hpp:34,
                 from /usr/local/include/NumCpp/Coordinates.hpp:30,
                 from /usr/local/include/NumCpp.hpp:44,
                 from /mnt/d/Code/rvten/main.cpp:5:
/usr/local/include/NumCpp/Core/Constants.hpp:44:37: error: 'nan' is not a member of 'std'; did you mean 'tan'?
   44 |         const double     nan = std::nan("1");                           ///< NaN
      |                                     ^~~
      |                                     tan
In file included from /usr/local/include/NumCpp/NdArray.hpp:30,
                 from /usr/local/include/NumCpp/Functions/deg2rad.hpp:33,
                 from /usr/local/include/NumCpp/Coordinates/Dec.hpp:36,
                 from /usr/local/include/NumCpp/Coordinates/Coordinate.hpp:34,
                 from /usr/local/include/NumCpp/Coordinates.hpp:30,
                 from /usr/local/include/NumCpp.hpp:44,
                 from /mnt/d/Code/rvten/main.cpp:5:
/usr/local/include/NumCpp/NdArray/NdArrayCore.hpp: In lambda function:
/usr/local/include/NumCpp/NdArray/NdArrayCore.hpp:4249:46: error: 'nearbyint' is not a member of 'std'; did you mean 'nearbyint'?
 4249 |             { return static_cast<dtype>(std::nearbyint(static_cast<double>(value) * multFactor) / multFactor); };
      |                                              ^~~~~~~~~
In file included from /home/ubuntu/RISCV/bin/riscv64/riscv64-unknown-elf/include/c++/11.1.0/cmath:45,
                 from /usr/local/include/NumCpp/Coordinates/Coordinate.hpp:30,
                 from /usr/local/include/NumCpp/Coordinates.hpp:30,
                 from /usr/local/include/NumCpp.hpp:44,
                 from /mnt/d/Code/rvten/main.cpp:5:
/home/ubuntu/RISCV/bin/riscv64/riscv64-unknown-elf/include/math.h:315:15: note: 'nearbyint' declared here
  315 | extern double nearbyint (double);
      |               ^~~~~~~~~
In file included from /usr/local/include/NumCpp/Functions.hpp:70,
                 from /usr/local/include/NumCpp.hpp:47,
[... 201 lines cut ...]
/usr/local/include/NumCpp/Vector/Vec2.hpp:222:30: note:   candidate expects 3 arguments, 2 provided
  222 |             return std::hypot(x, y);
      |                    ~~~~~~~~~~^~~~~~
In file included from /usr/local/include/NumCpp/Coordinates/Coordinate.hpp:30,
                 from /usr/local/include/NumCpp/Coordinates.hpp:30,
                 from /usr/local/include/NumCpp.hpp:44,
                 from /mnt/d/Code/rvten/main.cpp:5:
/home/ubuntu/RISCV/bin/riscv64/riscv64-unknown-elf/include/c++/11.1.0/cmath:1868:3: note: candidate: 'float std::hypot(float, float, float)'
 1868 |   hypot(float __x, float __y, float __z)
      |   ^~~~~
/home/ubuntu/RISCV/bin/riscv64/riscv64-unknown-elf/include/c++/11.1.0/cmath:1868:3: note:   candidate expects 3 arguments, 2 provided
/home/ubuntu/RISCV/bin/riscv64/riscv64-unknown-elf/include/c++/11.1.0/cmath:1872:3: note: candidate: 'double std::hypot(double, double, double)'
 1872 |   hypot(double __x, double __y, double __z)
      |   ^~~~~
/home/ubuntu/RISCV/bin/riscv64/riscv64-unknown-elf/include/c++/11.1.0/cmath:1872:3: note:   candidate expects 3 arguments, 2 provided
/home/ubuntu/RISCV/bin/riscv64/riscv64-unknown-elf/include/c++/11.1.0/cmath:1876:3: note: candidate: 'long double std::hypot(long double, long double, long double)'
 1876 |   hypot(long double __x, long double __y, long double __z)
      |   ^~~~~
/home/ubuntu/RISCV/bin/riscv64/riscv64-unknown-elf/include/c++/11.1.0/cmath:1876:3: note:   candidate expects 3 arguments, 2 provided
make[2]: *** [CMakeFiles/rvtorch.dir/build.make:63: CMakeFiles/rvtorch.dir/main.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/rvtorch.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

I think it might be the error in the std library. But as a beginner of riscv, I can't solve it after seeking several days

cylindrical2002, 2022-09-18:
By the way, if I changes to the riscv64-unknown-linux-gnu-g++, it works! But I can't use it because out platform only supports riscv64-unknown-elf-g++.

TommyMurphyTM1234, 2022-09-18:
Duplicate of 

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1127

I don't really understand why you've logged it a second time...

TommyMurphyTM1234, 2022-09-18:
> By the way, if I change to the `riscv64-unknown-linux-gnu-g++`, it works! But I can't use it because our platform only supports `riscv64-unknown-elf-g++` (I can not show you how our platform built because it is another project, but I am sure if I can use `riscv-unknown-elf-g++` to compile the program, it can be supported on our platform).

So your RISC-V target platform is not running Linux?

Is that's the case then I don't think that there's any quick/easy solution here as already explained in https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1127.

cylindrical2002, 2022-09-23:
Yeah, The final platform we would run the program on is not actually a running Linux, but maybe it is okay for us to run the project on qemu, could you give me any advise about how to run it on qemu?

TommyMurphyTM1234, 2022-09-23:
> could you give me any advice about how to run it on qemu?

* https://www.qemu.org/docs/master/system/target-riscv.html
* https://wiki.qemu.org/Documentation/Platforms/RISCV
* https://risc-v-getting-started-guide.readthedocs.io/en/latest/linux-qemu.html

TommyMurphyTM1234, 2022-09-27:
Duplicate of https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1127

closed 2022-09-27 by TommyMurphyTM1234