riscv-collab/riscv-gnu-toolchain #1130: What is the meaning of ubuntu-18.04 and ubuntu-20.04 in release

thomas-coding, 2022-09-19:
Hi 

From
https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2022.08.26

```
riscv32-elf-ubuntu-18.04-nightly-2022.08.26-nightly.tar.gz
riscv32-elf-ubuntu-20.04-nightly-2022.08.26-nightly.tar.gz
```
There ard two riscv32 release, named "ubuntu-18.04" and "ubuntu-20.04", what's the difference between them?

Thanks.

pz9115, 2022-09-19:
I think this package mean generated from different version of ubuntu machine, that may have different default version of native gcc, libs. So you can choose the version that your machine componented.

TommyMurphyTM1234, 2022-09-19:
That's correct. You can see the details of the nightly build GitHub action here:

* https://github.com/riscv-collab/riscv-gnu-toolchain/actions/runs/3079611853/workflow

thomas-coding, 2022-09-20:
OK, I see, Thanks all.