riscv-collab/riscv-gnu-toolchain #1135: __builtin_bswap32 undefined

xnox, 2022-10-03:
I am building kernel code that uses __builtin_bswap32 on all other architectures that expands into some inline instructions, i.e. movbe and some such, but on riscv64 I ended up with a symbol dependency on bswap32 which in kernel module code is undefined.

Is that expected that __builtin_bswap32 is a "library function"?

TommyMurphyTM1234, 2022-10-03:
Hi @xnox - I presume that you're using the newlib bare metal toolchain if you're compiling kernel code? Is that kernel/code available publicly?

xnox, 2022-10-03:
let me experiment more to figure out for sure where it is going south; in case it is generated by some other code.

xnox, 2022-10-03:
failing to reproduce this now =/ horum.

TommyMurphyTM1234, 2022-10-03:
Thanks for the update @xnox - if it crops up again and you can reproduce it please reopen and update this issue.

closed 2022-10-03 by TommyMurphyTM1234