riscv-collab/riscv-gnu-toolchain #1136: --enable-multilib build fails for make musl

TommyMurphyTM1234, 2022-10-03:
`--enable-multilib` works for `make [newlib]` and `make linux` but fails for `make musl`.
```
git clone https://github.com/riscv/riscv-gnu-toolchain musl-multilib
cd musl-multilib
./configure --enable-multilib --prefix=`pwd`/installed-tools
date > build.log; make musl 2>&1 | tee -a build.log; date >> build.log

/bin/bash /home/tommy/Downloads/riscv-tools/musl-multilib/gcc/libgcc/../mkinstalldirs /home/tommy/Downloads/riscv-tools/musl-multilib/installed-tools/lib/gcc/riscv64-unknown-linux-musl/12.2.0/include
/usr/bin/install -c -m 644 unwind.h /home/tommy/Downloads/riscv-tools/musl-multilib/installed-tools/lib/gcc/riscv64-unknown-linux-musl/12.2.0/include
/bin/bash /home/tommy/Downloads/riscv-tools/musl-multilib/gcc/libgcc/../mkinstalldirs /home/tommy/Downloads/riscv-tools/musl-multilib/installed-tools/lib/gcc/riscv64-unknown-linux-musl/12.2.0/include
/usr/bin/install -c -m 644 /home/tommy/Downloads/riscv-tools/musl-multilib/gcc/libgcc/gcov.h /home/tommy/Downloads/riscv-tools/musl-multilib/installed-tools/lib/gcc/riscv64-unknown-linux-musl/12.2.0/include
make[2]: Leaving directory '/home/tommy/Downloads/riscv-tools/musl-multilib/build-gcc-musl-stage1/riscv64-unknown-linux-musl/libgcc'
make[1]: Leaving directory '/home/tommy/Downloads/riscv-tools/musl-multilib/build-gcc-musl-stage1'
mkdir -p stamps/ && touch stamps/build-gcc-musl-stage1
rm -rf stamps/build-musl-linux build-musl-linux
mkdir build-musl-linux
cd build-musl-linux && \
	CC="riscv64-unknown-linux-musl-gcc " \
	CXX="riscv64-unknown-linux-musl-g++ " \
	CFLAGS="  -mcmodel=medlow -g -O2 " \
	CXXFLAGS="  -mcmodel=medlow -g -O2 " \
	ASFLAGS=" -mcmodel=medlow " \
	/home/tommy/Downloads/riscv-tools/musl-multilib/musl/configure \
	--host=riscv64-unknown-linux-musl \
	--prefix=/home/tommy/Downloads/riscv-tools/musl-multilib/installed-tools/sysroot \
	--disable-werror \
	--enable-shared \
	
/bin/sh: 2: /home/tommy/Downloads/riscv-tools/musl-multilib/musl/configure: not found
make: *** [Makefile:761: stamps/build-musl-linux] Error 127
```
Build log: [build.log.tar.gz](https://github.com/riscv-collab/riscv-gnu-toolchain/files/9700495/build.log.tar.gz)
Branch: master
Commit: c2857f9c03867f9fe9ee8e44b7cf57dd83020455

TommyMurphyTM1234, 2022-10-06:
Perhaps this is because it seems that the musl toolchain can only support 64 bit RISC-V targets - see here:

* https://github.com/riscv-collab/riscv-gnu-toolchain#installation-newliblinux-multilib

but the default set of multilibs includes some rv32 architectures?

* https://gcc.gnu.org/git/?p=gcc.git;a=blob;f=gcc/config/riscv/t-linux-multilib;h=298547fee38b2b4062e7854c539aabd292927857;hb=HEAD

Perhaps a multilib build with only rv64 architecture libs would work?

TommyMurphyTM1234, 2022-10-06:
Oh, I just noticed this comment here:

* https://github.com/riscv-collab/riscv-gnu-toolchain#installation-newliblinux-multilib

> The musl compiler (`riscv64-unknown-linux-musl-`) will only be able to target 64-bit systems due to limitations in the upstream musl architecture support. The `--enable-multilib` flag therefore does not actually enable multilib support for musl libc.

But I guess that the build shouldn't fail?

florianhofhammer, 2022-10-14:
Hi,

The `musl` submodule is not initialized in your case, this has nothing to do with the multilib flags.
I've added auto-initialization in #1144. In the meantime, just do a `git submodule update --init musl` in the base directory before you invoke `make`, this should resolve the issue.

Hope this helps :)

TommyMurphyTM1234, 2022-10-15:
Thanks @florianhofhammer - I tried that and the build succeeds but there are no multilibs as far as I can see - just support for the toolchain default arch/abi `rv64gc/lp64d`:
```
git clone https://github.com/riscv/riscv-gnu-toolchain musl-multilib
cd musl-multilib/
git submodule update --init musl
./configure --prefix=`pwd`/installed-tools --enable-multilib
date > build.log; make musl 2>&1 | tee -a build.log; date >> build.log
cd installed-tools/bin/
./riscv64-unknown-linux-musl-gcc -print-multi-lib
.;
./riscv64-unknown-linux-musl-gcc -dumpspecs
...
*multilib_defaults:
march=rv64imafdc mabi=lp64d
...
```
Branch: [master](https://github.com/riscv-collab/riscv-gnu-toolchain)
Commit: c63d1b66120f860e3f6b16019c77d38b918b44ec
Build log: [build.log.tar.gz](https://github.com/riscv-collab/riscv-gnu-toolchain/files/9792318/build.log.tar.gz)
Dumpspecs: [dumpspecs.txt](https://github.com/riscv-collab/riscv-gnu-toolchain/files/9792320/dumpspecs.txt)

TommyMurphyTM1234, 2022-10-15:
I guess that the info cited here 

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1136#issuecomment-1270709965

means that the multilibs are not expected to be built for the musl toolchain?
That being the case I'm closing this issue.

closed 2022-10-15 by TommyMurphyTM1234