riscv-collab/riscv-gnu-toolchain #1143: Unable to `li` some symbols

jgcodes2020, 2022-10-12:
Here's a sample program:
```asm
.section .text
.global _start

SYS_write=64
SYS_exit=93

SIGTRAP=5

_start:
  li a7, SYS_write
  li a0, 1
  la a1, msg
  li a2, msg.size
  ecall
  
  li a7, SYS_exit
  li a0, 0
  ecall

.section .rodata
msg: 
  .ascii "Hello, world!\n"
  .equ msg.size, (. - msg)
```
This fails at `li a2, msg.size`, as `msg.size` is not an integer constant. If I instead replace `li` with `la`, it does work, however it treats msg.size as a PC-relative address. I would like it to be treated as an immediate value.

jgcodes2020, 2022-10-13:
> Label subtraction is a bit inflexible on RISC-V because of linker relaxation

What is "linker relaxation" by the way?

TommyMurphyTM1234, 2022-10-13:
> What is "linker relaxation" by the way?

This might help explain:

* https://www.sifive.com/blog/all-aboard-part-3-linker-relaxation-in-riscv-toolchain

TommyMurphyTM1234, 2022-10-15:
@aswaterman, @cmuellner, @kito-cheng - should this issue remain open (as an enhancement issue?), be closed, be logged upstream, or something else? Thanks.

TommyMurphyTM1234, 2023-10-16:
No activity in over a year so I am closing this issue.

closed 2023-10-16 by TommyMurphyTM1234