riscv-collab/riscv-gnu-toolchain #1146: error: looser exception specification on overriding virtual function

TONG369, 2022-10-18:
Hello!
I have a problem when compile code. The problem is error: looser exception specification on overriding virtual function

[image]

but，when it compile in x86_64 , is pass!

how How to solve it?

TommyMurphyTM1234, 2022-10-18:
When reporting issues you really need to provide/attach a reproducible test case (not just a code snippet) and a make/build log (not just a screenshot). If you search for the error message in question you'll see that it's a legitimate C++ error and you'll find why it happens and how to avoid it. The fact that the same code (?) compiles for another target without error seems odd but may be explicable (e.g. because of different default or explicit compilation flags).

TommyMurphyTM1234, 2022-10-18:
I can't get your code snippet to compile even for x86_64 using Compiler Explorer:

* https://godbolt.org/z/zbqePrdTf

TONG369, 2022-10-18:
sorry , is my falut!

I use the riscv   toolchain that comes with yocto! The version is 11.2.0 。

The code is from :
https://github.com/Cambricon/easydk/blob/6ae970adbbde9a2b0818c8f9a078327860621622/infer_server/include/infer_server.h#L442

>

TONG369, 2022-10-19:
@TommyMurphyTM1234 
@arunthomas 
@gkelly 
@scottj97 

when I use riscv64-unknown-linux-gnu-g++ ,   verison is    11.2.    The error is still occur!

TommyMurphyTM1234, 2022-10-19:
Can you create a simpler smaller standalone test case that doesn't require the compilation of the full library that you're trying to use?

TONG369, 2022-10-19:
> Can you create a simpler smaller standalone test case that doesn't require the compilation of the full library that you're trying to use?

sorry, I can't

cmuellner, 2022-10-19:
The "looser exception specification on overriding virtual function" error is caused by the C++ frontend of GCC.
I can't see how this can be caused by RISC-V specific parts of GCC.

Also, I don't see how this issue relates to this repository. You probably need to report this issue to the project that provided you the Yocto sources that you are using.

closed 2022-10-19 by cmuellner

TONG369, 2022-10-19:
> The "looser exception specification on overriding virtual function" error is caused by the C++ frontend of GCC. I can't see how this can be caused by RISC-V specific parts of GCC.
>
> Also, I don't see [...]

@cmuellner 

hello , this error in x86 is disapper!

riscv64-unknown-linux-gnu-g++ , verison is 11.2. The error is still occur!

cmuellner, 2022-10-19:
I got that, but let me ask the following questions:

* Is your toolchain built with the build instructions in this repository?
* Is your x86 build using the exact same sources and build steps as your riscv64 toolchain, or are there some meta layers changing the sources/build instructions?
* Is the easydk package (in the used version) ready to be compiled for RISC-V?

This repository provides build scripts to build and test a RISC-V toolchain. It is not a support forum for all build issues that could be related to RISC-V. However, we will support you if you can provide a reduced test case that can reproduce the issue. But don't expect us to build a Yocto system with an unknown configuration and an outdated version of GCC.

TommyMurphyTM1234, 2022-10-19:
> hello , this error in x86 is disapper!
> 
> riscv64-unknown-linux-gnu-g++ , verison is 11.2. The error is still occur!

As I said before, this could be because of different compilation flags/options for the x86 Vs RISC-V cases. Without more info (e.g. the x86 build log) is impossible to say if this is a reasonable comparison.