riscv-collab/riscv-gnu-toolchain #1147: How to use "make report-linux SIM=qemu " correctly?

yangyang885, 2022-10-21:
I would like to run the testsuite using Makefile. My configure is `./configure --prefix=/home/yaojy/Desktop/work/riscv-gnu-toolchain/build-yao --with-gcc-src=/home/yaojy/Desktop/work/gcc-11.2.0` . And I used `make linux` to make it. when I do `make report-linux SIM=qemu` , outputting is:
`/home/yaojy/Desktop/work/riscv-gnu-toolchain/scripts/testsuite-filter gcc glibc /home/yaojy/Desktop/work/riscv-gnu-toolchain/test/allowlist ` `   'find build-gcc-linux-stage2/gcc/testsuite/ -name *.sum |paste -sd "," -' ` `
/home/yaojy/Desktop/work/riscv-gnu-toolchain/scripts/testsuite-filter <toolname> <libc> <white-list-base-dir> <testsuite.sum>`
And when I do `find build-gcc-linux-stage2/gcc/testsuite/ -name *.sum |paste -sd "," -`, there were no results.
Is something wrong with it?

cmuellner, 2022-10-22:
I attempted to reproduce this issue with the external gcc branch having the following top commit `7ca38856` and using the following commands:
```
./configure --with-gcc-src=/home/cm/src/gcc/gcc-11.2.0  --prefix=/opt/riscv-rvi
make -j$(nproc) linux
make -j$(nproc) report-linux SIM=qemu
```

All commands succeed and the 
The output ends with:
```
[...]
FAIL: gcc.dg/tree-ssa/tailcall-7-run.c (test for excess errors)

               ========= Summary of gcc testsuite =========
                            | # of unexpected case / # of unique unexpected case
                            |          gcc |          g++ |     gfortran |
 rv64imafdc/  lp64d/ medlow |   77 /    18 |   16 /     2 |   51 /    36 |

```

The find command outputs the following:
```
$ find build-gcc-linux-stage2/gcc/testsuite/ -name *.sum |paste -sd "," -
build-gcc-linux-stage2/gcc/testsuite/g++/g++.sum,build-gcc-linux-stage2/gcc/testsuite/gfortran/gfortran.sum,build-gcc-linux-stage2/gcc/testsuite/gcc/gcc.sum
```

So I cannot reproduce your behaviour. Please have a look at your console output, maybe there are errors because of missing host dependencies.

TommyMurphyTM1234, 2022-10-22:
I get slightly different results.

I downloaded the gcc 11.2.0 tarball from here 

* https://bigsearcher.com/mirrors/gcc/releases/gcc-11.2.0/

and extracted it into `$HOME/Downloads/gcc-11.2.0`.

I then built the tools and report as follows:
```
cd ~/Downloads/riscv-tools
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1147
cd issue-1147
git log | head -1
commit c63d1b66120f860e3f6b16019c77d38b918b44ec
./configure --prefix=`pwd`/installed-tools --with-gcc-src=$HOME/Downloads/gcc-11.2.0
date > build.log && make linux 2>&1 | tee -a build.log && date >> build.log
date > report.log && make report-linux SIM=qemu 2>&1 | tee -a report.log && date >> report.log
```
The report ends with:
```
FAIL: gfortran.dg/structure_constructor_11.f90   -O3 -g  (test for excess errors)
FAIL: gfortran.dg/structure_constructor_11.f90   -Os  (test for excess errors)
FAIL: gfortran.dg/unlimited_polymorphic_19.f90   -O0  (test for excess errors)

               ========= Summary of gcc testsuite =========
                            | # of unexpected case / # of unique unexpected case
                            |          gcc |          g++ |     gfortran |
 rv64imafdc/  lp64d/ medlow |   77 /    18 |   16 /     2 |   51 /    36 |
make: *** [Makefile:951: report-gcc-linux] Error 1
```
And the find command in the original issue report post works as follows:
```
find build-gcc-linux-stage2/gcc/testsuite/ -name *.sum |paste -sd "," -
build-gcc-linux-stage2/gcc/testsuite/g++/g++.sum,build-gcc-linux-stage2/gcc/testsuite/gcc/gcc.sum,build-gcc-linux-stage2/gcc/testsuite/gfortran/gfortran.sum
```
Build log: [build.log.tar.gz](https://github.com/riscv-collab/riscv-gnu-toolchain/files/9844841/build.log.tar.gz)
Report log: [report.log.tar.gz](https://github.com/riscv-collab/riscv-gnu-toolchain/files/9844842/report.log.tar.gz)

yangyang885, 2022-10-23:
Thank you very much. I reinstall it again and in the `riscv-gnu-toolchain/gcc`  ,I use `git log` check the versoin ,it shows:
`commit 7ca388565af176bd4efd4f8db1e5e9e11e98ef45 (HEAD, tag: releases/gcc-11.2.0)`
then I use the following commands:
`./configure --prefix=/home/yaojy/Desktop/work/riscv-gnu-toolchain/build-yao --with-arch=rv64gv --with-abi=lp64d`
`time make linux  2>&1 | tee build.log`
`time make report-linux SIM=qemu  2>&1 | tee report.log`
The report ends with:
`               ========= Summary of gcc testsuite =========`
                  `          | # of unexpected case / # of unique unexpected case`
                   `         |          gcc |          g++ |     gfortran |`
   `  rv64gv/  lp64d/ medlow |   77 /    18 |   16 /     2 |   51 /    36 |`
`make: *** [Makefile:951: report-gcc-linux] Error 1`
and using `find build-gcc-linux-stage2/gcc/testsuite/ -name *.sum |paste -sd "," -`, it shows:
`build-gcc-linux-stage2/gcc/testsuite/g++/g++.sum,build-gcc-linux-stage2/gcc/testsuite/gfortran/gfortran.sum,build-gcc-linux-stage2/gcc/testsuite/gcc/gcc.sum`

But  why `make: *** [Makefile:951: report-gcc-linux] Error 1` appears. Is it normal to be prompted like this?

cmuellner, 2022-10-23:
The GCC test suite will fail in the case of at least one failing test case.
In the `riscv-gnu-toolchain` `Makefile` does not distinguish between error causes (e.g. test fails or errors during test execution because of missing dependencies) and thus reports an error.

yangyang885, 2022-10-23:
> The GCC test suite will fail in the case of at least one failing test case. In the `riscv-gnu-toolchain` `Makefile` does not distinguish between error causes (e.g. test fails or errors during test execution because of missing dependencies) and thus reports an error.

Thank you very much. I got it.

closed 2022-10-23 by yangyang885