riscv-collab/riscv-gnu-toolchain #1148: vfmacc_vf_f32m8 and vse32_v_f32m8 can't be compiled in rrv-next?

Zissi-Lei, 2022-10-27:
Hi, I use the branch rvv-next to compile rvv-1.0 intrinsic:
...
vfloat32m8_t v_x, v_y;
       for (size_t i = n; i > 0; i -= nvl) {
       nvl = vsetvl_e32m8(i);
       v_x = vle32_v_f32m8(x, nvl);
       v_y = vle32_v_f32m8(y, nvl);
       vfmacc_vf_f32m8(v_y, a, v_x, nvl);
       vse32_v_f32m8(y, v_y, nvl);
       x += nvl;
       y += nvl;
}
...
I found that the compiler skip the vfmacc_vf_f32m8 and vse32_v_f32m8  instructions as if they do not exist:
...
    30aa:	02868c27          	vs1r.v	v24,(a3)
               vfmacc_vf_f32m1(v_y, a, v_x, nvl);
               vse32_v_f32m1(y, v_y, nvl);
    30ae:	c2202773          	csrr	a4,vlenb
...
Does rvv-next support these instructions? Thanks for your time.

TommyMurphyTM1234, 2022-10-27:
I cannot compile your code as provided.
Please provide a reproducible test case.
And also details of how you built the toolchain.
Thanks.

zhongjuzhe, 2022-10-27:
rvv-next support all v1.0 intrinsics. Here is the testcase for the intrinsics you ask:
https://github.com/riscv-collab/riscv-gcc/blob/riscv-gcc-rvv-next/gcc/testsuite/gcc.target/riscv/rvv/intrinsic/alu/vfmacc.c
https://github.com/riscv-collab/riscv-gcc/blob/riscv-gcc-rvv-next/gcc/testsuite/gcc.target/riscv/rvv/intrinsic/load-store/vsex.c

Zissi-Lei, 2022-10-28:
OK, thanks. I found that I made a mistake in the code. Now I can compile it and run successfully!

closed 2022-10-28 by Zissi-Lei