riscv-collab/riscv-gnu-toolchain #1149: cfi_label and cfi_startproc in start.S

alexfanqi, 2022-10-29:
Hi,

Can I get some help to clarify the `.cfi_label dummy` directive [here](https://github.com/bminor/glibc/blob/8c77e26ba8b360c851b2b9485bb4431aacc51ad1/sysdeps/riscv/start.S#L48)? In https://github.com/ziglang/zig/issues/3340, zig vendors part of glibc header, and it fails to compile because llvm does not support .cfi_label. 

The thing confuses me is the macro ENTRY already includes a cfi_startproc, which I thought should start the FDE and makes cfi_label redundant.

I have also read [Bug 23125](https://sourceware.org/bugzilla/show_bug.cgi?id=23125) that introduces this line. The original patch in first comment didn't include cfi_label.

Is it safe to simply remove cfi_label, or there are other workarounds?

aswaterman, 2022-10-29:
I know this is not an answer to your question, but note that RISC-V is not alone in using the `.cfi_label` directive in this context; see https://github.com/bminor/glibc/search?q=cfi_label

As a matter of curiosity, I'd also like to understand why this directive is necessary.

TommyMurphyTM1234, 2022-10-29:
Shouldn't this issue be logged against the upstream glibc and not here?

alexfanqi, 2022-10-30:
>Anyway, doesn't this explain it?

>bminor/glibc@2dd12ba

I read this commit message before. It explains the reason for undefining ra and replacing tail, but doesn't mention cfi_label. Only comment I can find is cfi_label forces start of fde, which seems already done by cfi_startproc.

> RISC-V is not alone in using the .cfi_label directive in this context; see https://github.com/bminor/glibc/search?q=cfi_label
I noticed all arches using cfi_label come after riscv, so I tried to ask in riscv community first. I will lodge a bug to upstream while waiting for more help here.

TommyMurphyTM1234, 2022-11-04:
> I will lodge a bug to upstream while waiting for more help here.

Can you post a link here to the upstream bug please?

alexfanqi, 2022-11-04:
Sorry, upstream bug https://sourceware.org/bugzilla/show_bug.cgi?id=29749

TommyMurphyTM1234, 2022-11-05:
> Sorry, upstream bug https://sourceware.org/bugzilla/show_bug.cgi?id=29749

Thanks @alexfanqi

TommyMurphyTM1234, 2022-11-09:
This post seems to address the original question:

https://sourceware.org/bugzilla/show_bug.cgi?id=29749#c2

As such, can this issue be closed?

alexfanqi, 2022-11-10:
still want to check some details of cfi_label, but since the discussion is in upstream bugzilla. I think this issue can be closed.

closed 2022-11-10 by alexfanqi

TommyMurphyTM1234, 2022-11-10:
> still want to check some details of cfi_label, but since the discussion is in upstream bugzilla. I think this issue can be closed.

Thanks @alexfanqi