riscv-collab/riscv-gnu-toolchain #1151: la rd, symbol instruction get  additional 0x1_0000_0000 offset

pengluyou, 2022-10-31:
I used toolchain riscv64-unknown-elf-xx tried to put code section in high region which start from  0x80200000, and RW excute section and ZI region at low address at 0x200000, the compile and link can pass, but the la symbol instruction be comiled as:
    la      sp, __stackheap_end__
    80200008:	80040117          	auipc	sp,0x80040
    8020000c:	ff810113          	addi	sp,sp,-8 # 240000 <__stack>
actually it get 0x1_0024_0000

and also in C code access global value be comiled as:
    *_driver = &driver; // driver is an global var
    80200228:	80000797          	auipc	a5,0x80000
    8020022c:	52078793          	addi	a5,a5,1312 # 200748 <driver>
then get _driver ponit to 0x100200748

that will cause the calculated addresses get an additional 0x1_0000_0000 offset.

TommyMurphyTM1234, 2022-10-31:
You need to provide more information if you want assistance.
For example...

1. How you built or obtained your toolchain
2. Reproducible test case and instructions on how you're compiling it

pengluyou, 2022-11-01:
Hi TommyMurphy
Thanks so much!

1. How you built or obtained your toolchain?
I just git clone tool chain from this github(git@github.com:riscv-collab/riscv-gnu-toolchain.git) and followed the README.md newlib section to build build and install riscv64-unknown-elf-xx tools

2.Reproducible test case and instructions on how you're compiling it
below  copy part of the code because the issue happened at the beginning:
_.global Reset_Handler
Reset_Handler:
    /* disable ie and clear all interrupts */
...
    la      sp, __stackheap_end__

    /* Load d_ata section */
    la      a0, __data_load__
    la      a1, __data_start__
    la      a2, __data_end__
    bgeu    a1, a2, 2f_
1:_
...
and the link file as below,
_ENTRY(Reset_Handler)
MEMORY {
    mem0 (rx) : ORIGIN = 0x80200000, LENGTH = (64 * 1024)
    mem1 (w) : ORIGIN = 0x80220000, LENGTH = (64 * 1024)
}
    REGION_ALIAS("r", mem0);
    REGION_ALIAS("w", mem1);
    REGION_ALIAS("x", mem0);
SECTIONS {
    .text : {
        . = ALIGN(0x8) ;
        __stext = . ;
        KEEP(*startup.o(*.text*))
        KEEP(*startup.o(*.vectors*))
        *(.text .text.*)
    } > x
... ...
.stackheap (NOLOAD) : {
        __stackheap_start__ = ABSOLUTE(.);
        . = ORIGIN(w) + LENGTH(w);
        __stackheap_end__ = ABSOLUTE(.);
    } > w
    __data_end__ = __bss_start__;
    __bss_end__ = __stackheap_start__;
    __stack = __stackheap_end__;
}

At first I put RW and ZI section after CODE/RO section  it can work normally， from the objdump file
the instruction which set sp is correct:
 la      sp, __stackheap_end__
    80200008:	00030117          	auipc	sp,0x30
    8020000c:	ff810113          	addi	sp,sp,-8 # 80230000 <__stack>

but after I modified the RW and ZI section to low address as:
  mem1 (w) : ORIGIN = 0x200000, LENGTH = (64 * 1024)
then find the objdump file, this instruction change to 
    la      sp, __stackheap_end__
    80200008:	80010117          	auipc	sp,0x80010
    8020000c:	ff810113          	addi	sp,sp,-8 # 210000 <__stack>
after instruction at 8020000C, the sp will be 0x100210000 which is not exist, the expected value is 0x210000.

and below list the build flag:
-Werror -Wall -Wextra -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-missing-field-initializers -ffunction-sections -fdata-sections -g -std=gnu11 -fno-exceptions -Og -mcmodel=medany -MD -MP
and link flag:
-Wl,--gc-sections -Wl,--cref -Wl,--undefined=arch_exceptions -Wl,-n -Wl,--start-group -lc -lgcc

TommyMurphyTM1234, 2022-11-04:
That's not a reproducible test case.
It's just a code snippet.

TommyMurphyTM1234, 2022-11-11:
No reproducible test case provided so closing this.
If/when a reproducible test case is available please attach it and reopen the issue.

closed 2022-11-11 by TommyMurphyTM1234