riscv-collab/riscv-gnu-toolchain #1153: Test questions about rvv

yangyang885, 2022-11-05:
I installed the toolchain with following command:
`git clone https://github.com/riscv/riscv-gnu-toolchain rv64-linux-rvv-next -b rvv-next`
`./configure --prefix=/home/yaojy/Desktop/rv64-linux-rvv-next/installed_location --with-arch=rv64gcv --with-abi=lp64d`[config.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/9942938/config.log)
`make linux`
and test it with command:
`make report-linux SIM=qemu`
[make-report-linux-qemu.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/9942935/make-report-linux-qemu.log)
```
========= Summary of gcc testsuite =========
                            | # of unexpected case / # of unique unexpected case
                            |          gcc |          g++ |     gfortran |
    rv64gcv/  lp64d/ medlow |23118 /  6724 | 9920 /  2479 |17963 /  3028 |
```
In the report, there are many fails. It seems that the testsuite can not run on qemu. Is the result correct?
In the file of toolchain, `riscv-gcc` git version is `42df3464` and `qemu` git version is `553032db`.
I'm more interested in the support of gcc to rvv (contains compile and execute function), how can i test it?

zhongjuzhe, 2022-11-05:
rvv-next is well tested and reliable for VLEN >= 128bit.  Your result is incorrect. These fails should not happen.
Please make sure you are using correct riscv-gcc-rvv-next branch, the latest commit:
https://github.com/riscv-collab/riscv-gcc/commit/eb96155f49deefcde0b054f6c02aa67c607ac482

Besides, make sure you enable RVV ISA in QEMU.

TommyMurphyTM1234, 2022-11-05:
> Please make sure you are using correct riscv-gcc-rvv-next branch, the latest commit: [riscv-collab/riscv-gcc@eb96155](https://github.com/riscv-collab/riscv-gcc/commit/eb96155f49deefcde0b054f6c02aa67c607ac482)

Does that mean that this is NOT the correct way to build the latest GCC toolchain with vector support?
```git clone https://github.com/riscv/riscv-gnu-toolchain rv64-linux-rvv-next -b rvv-next
./configure --prefix=/home/yaojy/Desktop/rv64-linux-rvv-next/installed_location --with-arch=rv64gcv --with-abi=lp64d
```
If it's not then what IS the correct way?

> Besides, make sure you enable RVV ISA in QEMU.

How does one do that in the context of using make to run the GCC test suite?

yangyang885, 2022-11-05:
>Besides, make sure you enable RVV ISA in  QEMU.

I have the same question as @TommyMurphyTM1234 , and how can I do to enable RVV ISA in QEMU? @zhongjuzhe

zhongjuzhe, 2022-11-06:
Here is an example to enable RVV in QEMU:
qemu-riscv64 -cpu rv64,v=true,vlen=256,elen=64,vext_spec=v1.0,Zfh=true

To get latest RVV support, I suggest you:
cd riscv-gcc
git checkout  riscv-gcc-rvv-next
git pull

Then build the toolchain again.

TommyMurphyTM1234, 2022-11-06:
> To get latest RVV support, I suggest you: cd riscv-gcc git checkout riscv-gcc-rvv-next git pull
> 
> Then build the toolchain again.

Hi @zhongjuzhe 

Do you mean:
```
git clone --recurse https://github.com/riscv-collab/riscv-gnu-toolchain rvv
cd rvv/riscv-gcc
git checkout riscv-gcc-rvv-next
git pull
```
or
```
git clone --recurse https://github.com/riscv-collab/riscv-gnu-toolchain rvv -b rvv-next
cd rvv/riscv-gcc
git checkout riscv-gcc-rvv-next
git pull
```
or something else?

pz9115, 2022-11-07:
> I installed the toolchain with following command: `git clone https://github.com/riscv/riscv-gnu-toolchain rv64-linux-rvv-next -b rvv-next` `./configure [...]

It's seems that you want use qemu as the dejagnu test simulator, First you need to update qemu to the version with support vector extensions: https://github.com/qemu/qemu.git (l saw the qemu commit log was two years before, I will open an PR to update it)

Then you need to modify the simulator script in [riscv-gnu-toolchain/scripts/wrapper/qemu/](https://github.com/riscv-collab/riscv-gnu-toolchain/tree/rvv-next/scripts/wrapper/qemu), change the default qemu option like @zhongjuzhe said
>Here is an example to enable RVV in QEMU:
qemu-riscv64 -cpu rv64,v=true,vlen=256,elen=64,vext_spec=v1.0,Zfh=true

After that re-build and make report this, make sure your qemu option had opened all vector that you configure enabled.

pz9115, 2022-11-07:
> > I installed the toolchain with following command: `git clone https://github.com/riscv/riscv-gnu-toolchain rv64-linux-rvv-next -b rvv-next` `./configure --prefix=/home/yaojy/Desktop/rv64-linux-rvv-next/installed_location --with-arch=rv64gcv --with-abi=lp64d`[config.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/9942938/config.log) `make linux` and test it with command: `make report-linux SIM=qemu` [make-report-linux-qemu.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/9942935/make-report-linux-qemu.log)
> > ```
> > ========= Summary of gcc testsuite =========
> >                             | # of unexpected case / # of unique unexpected case
> >                             |          gcc |          g++ |     gfortran |
> >     rv64gcv/  lp64d/ medlow |23118 /  6724 | 9920 /  2479 |17963 /  3028 |
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > In the report, there are many fails. It seems that the testsuite can not run on qemu. Is the result correct? In the file of toolchain, `riscv-gcc` git version is `42df3464` and `qemu` git version is `553032db`. I'm more interested in the support of gcc to rvv (contains compile and execute function), how can i test it?
> 
> It's seems that you want to use qemu as the dejagnu test simulator, First you need to update qemu to the version with support vector extensions: https://github.com/qemu/qemu.git (l saw the qemu commit log was two years before, I will open an PR to update it)
> 
> Then you need to modify the simulator script in [riscv-gnu-toolchain/scripts/wrapper/qemu/](https://github.com/riscv-collab/riscv-gnu-toolchain/tree/rvv-next/scripts/wrapper/qemu), change the default qemu option like @zhongjuzhe said
> 
> > Here is an example to enable RVV in QEMU:
> > qemu-riscv64 -cpu rv64,v=true,vlen=256,elen=64,vext_spec=v1.0,Zfh=true
> 
> After that re-build and do make report, make sure your qemu option had opened all vector extensions that you configure enabled.

yangyang885, 2022-11-07:
> I installed the toolchain with following command:
git clone https://github.com/riscv/riscv-gnu-toolchain rv64-linux-rvv-next -b rvv-next
./configure --prefix=/home/yaojy/Desktop/rv64-linux-rvv-next/installed_location --with-arch=rv64gcv --with-abi=lp64d[config.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/9942938/config.log)
make linux
and test it with command:
make report-linux SIM=qemu
[make-report-linux-qemu.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/9942935/make-report-linux-qemu.log)
> ========= Summary of gcc testsuite =========
>                             | # of unexpected case / # of unique unexpected case
>                             |          gcc |          g++ |     gfortran [...]

I rebuild the toolchain with commands:
`cd rv64-linux-rvv-next`
`make clean`
`./configure --prefix=/home/yaojy/Desktop/rv64-linux-rvv-next/installed_location --with-arch=rv64gcv --with-abi=lp64d`[config.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/9947497/config.log)
`cd rv64-linux-rvv-next/riscv-gcc`
`git fetch`
`git checkout eb96155`
`cd rv64-linux-rvv-next/`
`git rm qemu`
`git clone https://github.com/qemu/qemu.git`
`cd rv64-linux-rvv-next/`
`make linux | tee make-linux.log `[make-linux.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/9947508/make-linux.log)
And the riscv-gcc git commit is `eb96155`, `qemu-riscv64 --version` is `qemu-riscv64 version 7.1.50 (v7.1.0-1676-g6295a58ad1)` 
I modify the simulator script in` rv64-linux-rvv-next/scripts/wrapper/qemu/` as follows:
`qemu-riscv$xlen -r 5.10 -cpu rv64,v=true,vlen=256,elen=64,vext_spec=v1.0,Zfh=true "${qemu_args[@]}" -L ${RISC_V_SYSROOT} "$@"`[qemu.zip](https://github.com/riscv-collab/riscv-gnu-toolchain/files/9947530/qemu.zip)
then:
`make  report-linux SIM=qemu | tee makereport.log`[makereport.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/9947515/makereport.log)
But there are still many fails in the report. Is this the correct result? 
It seems that it is not running corrently. Did I not enable qemu option correctly or something else?

pz9115, 2022-11-07:
> I rebuild the toolchain with commands: `cd rv64-linux-rvv-next` `make clean` `./configure --prefix=/home/yaojy/Desktop/rv64-linux-rvv-next/installed_location --with-arch=rv64gcv [...]

I saw your log and checked some testcases, they are forced to support VLEN=128 only, maybe you can try set qemu args with:

-cpu rv64,v=true,Zfh=true

yangyang885, 2022-11-09:
Thanks a lot.
I re-build  the toolchain with the follow commands:
`./configure --prefix=/home/yaojy/Desktop/rv64-linux-rvv-next/installed_location --with-arch=rv64gcv_zfh --with-abi=lp64d`
`make linux`
And I set qemu args with :  -cpu rv64,v=true,Zfh=true. 
Then
 `make report-linux SIM=qemu | tee make-report-linux-qemu.log`[make-report-linux-qemu.zip](https://github.com/riscv-collab/riscv-gnu-toolchain/files/9966394/make-report-linux-qemu.zip)
It seems that it works.

TommyMurphyTM1234, 2022-11-09:
Closing as resolved.

closed 2022-11-09 by TommyMurphyTM1234

TommyMurphyTM1234, 2022-11-09:
> To get latest RVV support, I suggest you: cd riscv-gcc git checkout riscv-gcc-rvv-next git pull

@zhongjuzhe - Is this expected?
```
git clone  https://github.com/riscv-collab/riscv-gnu-toolchain rvv -b rvv-next --recursive
cd rvv/riscv-gcc
git checkout riscv-gcc-rvv-next
git pull
Already up to date.
```
If so, then why the advice to explicitly do this?
```
cd rvv/riscv-gcc
git checkout riscv-gcc-rvv-next
git pull
```

zhongjuzhe, 2022-11-09:
Yes

***@***.***
 
From: Tommy Murphy
Date: 2022-11-09 18:59
To: riscv-collab/riscv-gnu-toolchain
CC: Ju-Zhe Zhong; Mention
Subject: Re: [riscv-collab/riscv-gnu-toolchain] Test questions about rvv (Issue #1153)
To get latest RVV support, I suggest you: cd riscv-gcc git checkout riscv-gcc-rvv-next git pull
@zhongjuzhe - Is this expected?
git clone  https://github.com/riscv-collab/riscv-gnu-toolchain rvv -b rvv-next --recursive
cd rvv/riscv-gcc
git checkout riscv-gcc-rvv-next
git pull
Already up to date.

If so, then why the advice to explicitly do this?
cd rvv/riscv-gcc
git checkout riscv-gcc-rvv-next
git pull