riscv-collab/riscv-gnu-toolchain #1159: additional .2byte instructions are generated in asm & elf though not specified in .S

kumarbrk9, 2022-11-15:
Hi,
I'm using gcc version 11.1.0
While generating ".asm" from ".S", I see the toolchain adds some .2byte instructions which are not specified anywhere. That too this issue I can see is specific to last label present in ".S". I can see these getting executed in ".elf" as well.

Example:
I have added a new label in ".S" at end with just one instruction as follows.
end_dummy:
        nop

In ".asm", I can see the following
000018c <end_dummy>:
    000018c:	00000013          	nop
    0000190:	4641                	.2byte	0x4641
    0000194:	0000                	.2byte	0x0
    0000198:	7200                	.2byte	0x7200
    000019c:	7369                	.2byte	0x7369

It is expected that it should be just 000018c generated. But not sure why all these unwanted instructions are generated.
command using:
$RV64DISASM -d test.elf > test.asm

Thanks in advance.

Regards
Kumar

TommyMurphyTM1234, 2022-11-15:
Please provide a self-contained reproducible test case and instructions for compiling it to reproduce the issue.

TommyMurphyTM1234, 2022-11-16:
Any update @kumarbrk9?
Without a reproducible test case there's not much that can be done and this issue will be closed.

RaviB18, 2022-11-16:
I'll post the files and steps to reproduce the issue. Please expect a day. Sorry for the inconvenience.
Thanks in advance.

kumarbrk9, 2022-11-16:
Sure, I'll post the codes and also steps to reproduce the issue. Please expect some delay. Sorry for the inconvenience.
Thanks in advance.

kumarbrk9, 2022-11-17:
[Required_files.zip](https://github.com/riscv-collab/riscv-gnu-toolchain/files/10028200/Required_files.zip)
Hi, Following attachment has the required files and I've added a file "steps_to_reproduce_issue.txt".
As mentioned, I'm using gcc version 11.1.0
Earlier, I used to have "end_irq" at end and the junk/unnecessary instructions used to get generated for this label. Then I added an extra label "end_dummy" at end and those junk/unnecessary instructions are added to this label.
**So, I understood that label at end in irq.S has this issue.**

TommyMurphyTM1234, 2022-11-17:
I'm using the latest pre-built toolchain from here:

https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2022.11.16

but seem to get the same results.

And the extra label seems to be a red herring as whether it's present or not I get the same additional data in the ELF file.

This data looks like some sort of header or metadata but I cannot figure out what it is so far - maybe somebody else knows? It seems to be the target (RISC-V obviously) and arch. 

Apologies for embedding a screenshot but I can't think of an easier way offhand right now.

[image]

TommyMurphyTM1234, 2022-11-17:
Actually - maybe it's the contents of the `.riscv_attributes` section?
```
tommy@pelican:~/Downloads/riscv/bin$ ./riscv64-unknown-elf-objdump -h testing.elf

testing.elf:     file format elf64-littleriscv

Sections:
Idx Name          Size      VMA               LMA               File off  Algn
  0 .intr_handler 000006bf  000052e62a002000  000052e62a002000  00001000  2**2
                  CONTENTS, ALLOC, LOAD, CODE
  1 .text         000008c0  000052e62a0026c0  000052e62a0026c0  000016c0  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  2 .data         00000048  000052e62a804000  000052e62a804000  00002000  2**6
                  CONTENTS, ALLOC, LOAD, DATA
  3 .region_0     00001000  000052e62a804048  000052e62a804048  00002048  2**0
                  CONTENTS, ALLOC, LOAD, DATA
  4 .region_1     00001000  000052e62a805048  000052e62a805048  00003048  2**0
                  CONTENTS, ALLOC, LOAD, DATA
  5 .user_stack   00000fa0  000052e62a806048  000052e62a806048  00004048  2**2
                  CONTENTS, ALLOC, LOAD, DATA
  6 .riscv.attributes 00000047  0000000000000000  0000000000000000  00004fe8  2**0
                  CONTENTS, READONLY
```

TommyMurphyTM1234, 2022-11-17:
This section may be specified here although I cannot immediately correlate the hex dump above to the documentation...

https://github.com/riscv-non-isa/riscv-elf-psabi-doc/blob/master/riscv-elf.adoc#rv-section-type

RaviB18, 2022-11-17:
As mentioned earlier, whichever label is at the end in irq.S, will have these generated. Not sure of how it is happening. Any specific hacks to get rid of it?
Thanks in advance.

TommyMurphyTM1234, 2022-11-17:
> As mentioned earlier, whichever label is at the end in irq.S, will have these generated. 

My point was that even without any level the same data appears - at least with the latest nightly build of the toolchain that I'm using.

> Not sure of how it is happening

I suspect that the linker adds this section as a matter of course for the convenience of other tools that may need to understand the nature and target of the ELF file. But I'm still not fully clear on the origin, nature and purpose of this data.

> Any specific hacks to get rid of it?

Why do you want to eliminate it? Is not a loadable section so doesn't take up any space at runtime.

RaviB18, 2022-11-17:
I did not want any compressed instructions in my test. Because of this issue, error pops up as the additional instructions are compressed version.

TommyMurphyTM1234, 2022-11-17:
> I did not want any compressed instructions in my test. Because of this issue, error pops up as the additional instructions are compressed version.

The data in question is not instructions but metadata as far as I can tell.

cmuellner, 2022-11-17:
If you use `objdump -D` to look into irq.o, then you can see where this is coming from:
```
[...]
0000000000000674 <end_dummy>:
 674:   00000013                nop

Disassembly of section .riscv.attributes:

0000000000000000 <.riscv.attributes>:
   0:   4f41                    .2byte  0x4f41
   2:   0000                    .2byte  0x0
   4:   7200                    .2byte  0x7200
   6:   7369                    .2byte  0x7369
[...]
```

So the remaining question is: if the resulting `testing.elf` has a `.riscv.attributes` section, then why is the same data also appended to the `.intr_handler` section? The answer is, because you told the linker to do so.

You have the following section in your linker script:
```
[...]
  .intr_handler :
  {
    irq.o
  } >I_IRQ
[...]
```

This will simply put everything from `irq.o` into the `.intr_handler` section of `testing.elf`.
If you want to just take the `.intr_handler` section from `irq.o` you may want to use this instead:
```
  .intr_handler :
  {
    irq.o (.intr_handler)
  } >I_IRQ
```
With this you will get this in your `testing.elf`:
```
000052e62a004f34 <end_dummy>:
    52e62a004f34:       00000013                nop

Disassembly of section .data:

000052e62a804000 <_fdata>:
        ...
```

TommyMurphyTM1234, 2022-11-17:
Thanks @cmuellner - so the `.riscv.attributes` section is emitted at the end of every object file?
And the `.riscv.atributes` section is documented here: https://github.com/riscv-non-isa/riscv-elf-psabi-doc/blob/master/riscv-elf.adoc#rv-section-type
And, in this case, the linker script is erroneously incorporating the `.riscv.attributes` section for the `irq.o` object file into the ultimate ELF file?
As such, I'm closing this as not a valid issue but rather a mistaken linker script.

BTW @RaviB18 / @kumarbrk9 the `crt0.S` file's initialisation of the `$gp` register probably should be done with relaxation disabled:
```
.option push
.option norelax
la    gp, _gp
.option pop
```

closed 2022-11-17 by TommyMurphyTM1234