riscv-collab/riscv-gnu-toolchain #1163: Spike can't simulate RVV instructions compiled by riscv-gcc

tak-ka3, 2022-12-02:
I want to simulate programs composed of RISC-V Vector instructions using spike simulator. But I couldn't do it, so if you know the way to execute correctly, please tell me. 

After `spike --varch=vlen:128,elen:64 --isa=rv64gcv pk hello` is executed, nothing is output and program never ends like unlimit loop. That is the following. https://github.com/riscv-software-src/riscv-isa-sim/issues/1152#issuecomment-1323280293

## MY ENV
- riscv64-unknown-elf-gcc
```bash
git clone  https://github.com/riscv-collab/riscv-gnu-toolchain rvv -b rvv-next --recursive
cd rvv/riscv-gcc
git checkout remotes/origin/riscv-gcc-rvv-next
cd .. && mkdir build && cd build
../configure --prefix=/home/takumi/riscv64_github --with-arch=rv64gcv --with-abi=lp64d
make -j8
```
- spike
```bash
git clone git@github.com:riscv-software-src/riscv-isa-sim.git
cd riscv-isa-sim && mkdir build && cd build
../configure --prefix=/home/takumi/riscv64_github
make -j8
make install
```
- pk
```bash
git clone git@github.com:riscv-software-src/riscv-pk.git
cd riscv-pk && mkdir build && cd build
../configure --prefix=/home/takumi/riscv64_github --host=riscv64-unknown-elf --with-arch=rv64gcv
make -j8
make install
```

tak-ka3, 2022-12-02:
Even if I use the latest `master` branch at riscv-gnu-toolchain repo and try to use `#include riscv_vector.h", some errors are output like this and cannot compile RVV programs.
```
/home/takumi/riscv64_github/lib/gcc/riscv64-unknown-elf/12.2.0/include/riscv_vector.h:39:9: error: unknown type name '__fp16'
   39 | typedef __fp16 float16_t;
      |         ^~~~~~
/home/takumi/riscv64_github/lib/gcc/riscv64-unknown-elf/12.2.0/include/riscv_vector.h:43:9: error: unknown type name '__fp16'
   43 | typedef __fp16 __float16_t;
      |         ^~~~~~
```

pz9115, 2022-12-02:
Since RVV is still under developing, I think you can try to use `master` branch and update gcc to latest commit.

```
cd gcc
git fetch origin
git pull origin master
```

Then rebuild the riscv-gnu-toolchain with your set, my env show that could include `riscv_vector.h` correctly, hope this help.

TommyMurphyTM1234, 2022-12-02:
@tak-ka3, how exactly are you building your toolchain?
Please provide a small test program with which to reproduce your problem.
Thanks.

tak-ka3, 2022-12-02:
As you said I updated gcc to latest commit, and then I got to compile RVV code including `riscv_vector.h`!
The `releases/gcc-12` was updated just only about 3 months, so I thought there was no problem for using that branch. I look forward to seeing the current `master` branch merged into the new release branch.
Anyway, I had a very hard time to compiling that code, so I really appreciate your advice!

zhongjuzhe, 2022-12-02:
So far the full vector support still need to use code in riscv-gcc-rvv-next. For FP16 support, you need to use -march=rv64gcv_zfh

tak-ka3, 2022-12-02:
Thank you for your advice. 
Do you mean that `riscv-gcc-rvv-next` branch supports FP16, but spike cannot work for some reasons as my spike simulator didn't work, and if we use spike, we have to use the latest `master` branch?

zhongjuzhe, 2022-12-02:
riscv-gcc-rvv-next support all features of RVV including FP16.
You mean master branch of spike ?  So far I am using the latest spike, it works fine for me.

***@***.***
 
From: Takumi
Date: 2022-12-02 20:05
To: riscv-collab/riscv-gnu-toolchain
CC: Ju-Zhe Zhong; Comment
Subject: Re: [riscv-collab/riscv-gnu-toolchain] Spike can't simulate RVV instructions compiled by riscv-gcc (Issue #1163)
Thank you for your advice.
Do you mean that riscv-gcc-rvv-next branch supports FP16, but spike cannot work for some reasons as my spike simulator didn't work, and if we use spike, we have to use the latest master branch?

TommyMurphyTM1234, 2022-12-02:
> As you said I updated gcc to latest commit, and then I got to compile RVV code including `riscv_vector.h`! The `releases/gcc-12` was updated just only about 3 months, so I thought there was no problem for using that branch. I look forward to seeing the current `master` branch merged into a new release branch. Anyway, I had a very hard time to compiling that code, so I really appreciate your advice!

I take it that this issue can be closed so?

ryan037, 2022-12-05:
@tak-ka3 
My spike have the same problem. Did you resolve it?

tak-ka3, 2022-12-05:
@ryan037 
Yes I resolved it, but I cannot still build riscv-gcc using `rvv` branch supporting FP16. Even if I add option like `-march=rv64gcv_zfh`, I cannot build. 
If there is no problem for you to use RVV which isn't supporting FP16, you should use `master` branch in the riscv-gnu-toolchain rerpogitory and update gcc like https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1163#issuecomment-1334844022.

hanm2019, 2022-12-22:
@tak-ka3 I meet the same problem too. How did you solve it?

I config my env as follow:
```
export RISCV=/home/ubuntu/riscv

git clone  https://github.com/riscv-collab/riscv-gnu-toolchain rvv -b rvv-next --recursive
cd rvv/riscv-gcc
git checkout remotes/origin/riscv-gcc-rvv-next
cd .. && mkdir build && cd build
../configure --prefix=$RISCV --with-arch=rv64gcv --with-abi=lp64d
make -j8

git clone git@github.com:riscv-software-src/riscv-isa-sim.git
cd riscv-isa-sim && mkdir build && cd build
../configure --prefix=$RISCV --with-isa=RV64GCV
make -j8
make install

git clone git@github.com:riscv-software-src/riscv-pk.git
cd riscv-pk && mkdir build && cd build
../configure --prefix=$RISCV --host=riscv64-unknown-elf --with-arch=rv64gcv
make -j8
make install
```

then I use
```
riscv64-unknown-elf-gcc -march=rv64gcv -c hello.cpp -o hello
spike --isa=rv64gcv pk hello
```

to run, and It seem stucked

tak-ka3, 2022-12-22:
@hanm2019 
If there is no problem for you to use RVV which isn't supporting FP16, you should use master branch in the riscv-gnu-toolchain rerpogitory and update riscv-gcc like https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1163#issuecomment-1334844022.

But I don't still know the way to build the env where RVV instructions supporting FP16 are executetd. In other words, I can't build using option like `-march=rv64gcv_zfh`.

hanm2019, 2022-12-28:
thanks, I finally use the master branch of Qeum to run the RVV program

TommyMurphyTM1234, 2022-12-28:
Can this issue be closed?
I'm confused that it started off about spike but now seems to have moved on to QEMU.
In any case, I'm not sure that it has anything to do with the riscv-gnu-toolchain itself?

closed 2022-12-28 by tak-ka3