riscv-collab/riscv-gnu-toolchain #1164: RISC-V and Image Calssification with OpenCV, C++ Error

kahretsinanil, 2022-12-04:
Hi,

I'm trying to perform image classification method with opencv c++ and trying to run it on riscv-isa on linux. After that qemu-riscv64 will show the output.
I have installed the toolchain and I can perform simple image processing task with RISCV but when I try to run the image classification code with DNN, i got some errors.
I have tried to run he pretrained YOLOv5 weights but I got these errors.

`root@ubuntu:/home/anil/Desktop/YOLOv5# riscv64-unknown-linux-gnu-g++ main.cpp -o new -I /home/anil/Desktop/project/opencv/include/ -I /home/anil/Desktop/project/opencv/modules/core/include -I /home/anil/Desktop/project/opencv/build/install/include/opencv4 -I /home/anil/Desktop/project/opencv/modules/dnn/include/ -I /home/anil/Desktop/project/opencv/modules/imgproc/include/ -I /home/anil/Desktop/project/opencv/modules/highgui/include -I /home/anil/Desktop/project/opencv/modules/core/include/opencv2/core/cuda -L /home/anil/Desktop/project/opencv/build/lib -lopencv_core -lopencv_imgcodecs -lopencv_imgproc
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /tmp/ccxBXtGn.o: in function `.L0 ':
main.cpp:(.text+0x1f4): undefined reference to `cv::dnn::dnn4_v20220524::blobFromImage(cv::_InputArray const&, cv::_OutputArray const&, double, cv::Size_<int> const&, cv::Scalar_<double> const&, bool, bool, int)'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /tmp/ccxBXtGn.o: in function `.LEHE4':
main.cpp:(.text+0x268): undefined reference to `cv::dnn::dnn4_v20220524::Net::setInput(cv::_InputArray const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, double, cv::Scalar_<double> const&)'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: main.cpp:(.text+0x2ae): undefined reference to `cv::dnn::dnn4_v20220524::Net::getUnconnectedOutLayersNames[abi:cxx11]() const'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /tmp/ccxBXtGn.o: in function `.LEHB6':
main.cpp:(.text+0x2c6): undefined reference to `cv::dnn::dnn4_v20220524::Net::forward(cv::_OutputArray const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&)'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /tmp/ccxBXtGn.o: in function `.LEHB12':
main.cpp:(.text+0x614): undefined reference to `cv::dnn::dnn4_v20220524::NMSBoxes(std::vector<cv::Rect_<int>, std::allocator<cv::Rect_<int> > > const&, std::vector<float, std::allocator<float> > const&, float, float, std::vector<int, std::allocator<int> >&, float, int)'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /tmp/ccxBXtGn.o: in function `.LEHE20':
main.cpp:(.text+0x9b2): undefined reference to `cv::dnn::dnn4_v20220524::Net::Net()'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /tmp/ccxBXtGn.o: in function `.L81':
main.cpp:(.text+0xa2e): undefined reference to `cv::dnn::dnn4_v20220524::readNet(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /tmp/ccxBXtGn.o: in function `.L79':
main.cpp:(.text+0xa4c): undefined reference to `cv::dnn::dnn4_v20220524::Net::~Net()'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /tmp/ccxBXtGn.o: in function `.LEHE23':
main.cpp:(.text+0xb3c): undefined reference to `cv::dnn::dnn4_v20220524::Net::getPerfProfile(std::vector<double, std::allocator<double> >&)'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /tmp/ccxBXtGn.o: in function `.LEHE29':
main.cpp:(.text+0xc04): undefined reference to `cv::imshow(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, cv::_InputArray const&)'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /tmp/ccxBXtGn.o: in function `.LEHE30':
main.cpp:(.text+0xc2c): undefined reference to `cv::waitKey(int)'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /tmp/ccxBXtGn.o: in function `.LEHE31':
main.cpp:(.text+0xc64): undefined reference to `cv::dnn::dnn4_v20220524::Net::~Net()'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /tmp/ccxBXtGn.o: in function `.LEHB35':
main.cpp:(.text+0xd96): undefined reference to `cv::dnn::dnn4_v20220524::Net::~Net()'
`

TommyMurphyTM1234, 2022-12-04:
Aren't these just link errors that you need to work through, figure out, and fix yourself? I don't see any evidence that these are necessarily an issue with the riscv-gnu-toolchain tools themselves. If you believe that they are then please provide a reproducible test case (ideally a minimal one that doesn't require the compilation of the full OpenCV and other dependencies) and instructions on how to reproduce the issue for investigative purposes. Thanks.

kahretsinanil, 2022-12-04:
> Aren't these just build/link errors that you need to work through, figure out, and fix yourself? I don't see any evidence that these are necessarily an issue with the riscv-gnu-toolchain tools themselves. If you believe that they are then please provide a reproducible test case (ideally a minimal one that doesn't require the compilation of the full OpenCV and other dependencies) and instructions on how to reproduce the issue for investigative purposes. Thanks.

Hi, thanks for the quick response.

I think that might be a problem with gnu toolchain because when I try to compile a simple edge detection code, and it works without any problem, that's why I opened that issue.
If there is no problem working of riscv gnu toolchain with whole opencv library of course I will try to fix these errors.

The outputs of the simple edge detection can be shown below

`root@ubuntu:/home/anil/Desktop/deneme# riscv64-unknown-linux-gnu-g++ edge.cpp -o edge-detector -I /home/anil/Desktop/project/opencv/include/ -I /home/anil/Desktop/project/opencv/modules/core/include -I /home/anil/Desktop/project/opencv/build/install/include/opencv4 -I /home/anil/Desktop/project/opencv/modules/dnn/include/ -I /home/anil/Desktop/project/opencv/modules/imgproc/include/ -I /home/anil/Desktop/project/opencv/modules/highgui/include -I /home/anil/Desktop/project/opencv/modules/core/include/opencv2/core/cuda -L /home/anil/Desktop/project/opencv/build/lib -lopencv_core -lopencv_imgcodecs -lopencv_imgproc
root@ubuntu:/home/anil/Desktop/deneme# cd edge
edge.cpp       edge-detector  
`

TommyMurphyTM1234, 2022-12-04:
You haven't shown anything that looks like a toolchain problem as opposed to a build/link problem. You need to do this and provide a reproducible test case. You should probably ask about OpenCV build/link issues on the OpenCV project/forums.

TommyMurphyTM1234, 2022-12-15:
No update so closing as not reproducible.
If it's still an issue please post a reproducible test case/instructions and reopen it.

closed 2022-12-15 by TommyMurphyTM1234