riscv-collab/riscv-gnu-toolchain #1165: expat is missing ir ununusabe

yy172, 2022-12-06:
When I use --with-host=x86_64-w64-mingw32 to cross-compile under LINUX, both binutils and gcc step2 have been successfully compiled, but when entering the compilation gdb, an error is detected:     expat is missing ir ununusabe;  How do I specify the library or what are the correct cross compile options

TommyMurphyTM1234, 2022-12-06:
This doesn't identify any issue with the riscv-gnu-toolchain tools. It's simply a toolchain build issue. 

Maybe try using crosstool-NG to cross compile from Linux to Windows

https://crosstool-ng.github.io/

Or the xPack approach (or prebuilt binaries if applicable).

https://xpack.github.io/riscv-none-elf-gcc/

TommyMurphyTM1234, 2022-12-15:
No update so closing as not reproducible.
If it is still and issue please provide a reproducible test case/instruction and reopen it.

closed 2022-12-15 by TommyMurphyTM1234