riscv-collab/riscv-gnu-toolchain #1169: README - add "that is writeable"

CarlFK, 2022-12-25:
Now that I got an error, it makes sense to make sure the install path is writable, but the readme seemed like I could just blindly cut/paste the commands.  

at a minimum, add "that is writable" to "pick an install path" here:

`To build the Linux cross-compiler, pick an install path. If you choose, say, /opt/riscv, then add /opt/riscv/bin to your PATH now. Then, simply run the following command:`  https://github.com/riscv-collab/riscv-gnu-toolchain#installation-linux

```
mkdir -p /opt/riscv/.test || \
	(echo "Sorry, you don't have permission to write to" \
	 "'/opt/riscv'. Please make sure that the location is " \
	 "writable or use --prefix to specify another destination.'" \
	 && exit 1)
mkdir: cannot create directory ‘/opt/riscv’: Permission denied
Sorry, you don't have permission to write to '/opt/riscv'. Please make sure that the location is  writable or use --prefix to specify another destination.'
```

TommyMurphyTM1234, 2022-12-25:
I don't really see what the problem is. If you choose an install path that cannot be written then you get a clearly explanatory message and can try again to a writeable path - or use `sudo` as is common when installing anything to a "system" path.

However, if you think that the readme needs clarification, then feel free to submit a change via a pull request.

TommyMurphyTM1234, 2023-01-04:
@CarlFK - any update/PR?

TommyMurphyTM1234, 2023-01-21:
The same comment should be added to the bare metal/newlib toolchain instructions. And the full stop/period is in the wrong place.

lazyparser, 2023-02-12:
> @CarlFK - any update/PR?

Seems fixed in another PR that updates README:
https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1182

TommyMurphyTM1234, 2023-02-17:
Yes, it's updated in one place (but not another) in the `README.md` but it was also always flagged up front by the `Makefile` so it's arguably moot in my opinion.

*

closed 2023-02-17 by TommyMurphyTM1234