riscv-collab/riscv-gnu-toolchain #1170: docker method: fail to build the "simple-display" testcase

vishwanatha-hattera, 2022-12-25:
I followed the steps from the link ("Alternative Setup: Docker").  Able to build and run the sw/basic-c using "make sim".  But same is not working for "sw/simple-display", Below is the log.

3da74e0444f3:~/riscv-vp/sw/simple-display$ make sim
riscv32-unknown-elf-g++ -g -std=c++14 main.cpp libDisplay.cpp -o main -march=rv32ima -mabi=ilp32
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /tmp/ccpkpNDJ.o: can't link soft-float modules with double-float modules
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: failed to merge target specific data of file /tmp/ccpkpNDJ.o
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /tmp/ccieDnlA.o: can't link soft-float modules with double-float modules
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: failed to merge target specific data of file /tmp/ccieDnlA.o
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /tmp/ccpkpNDJ.o: in function `activeWait(float)':
/home/riscv-vp/riscv-vp/sw/simple-display/main.cpp:17: undefined reference to `__floatsisf'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /home/riscv-vp/riscv-vp/sw/simple-display/main.cpp:17: undefined reference to `__mulsf3'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /home/riscv-vp/riscv-vp/sw/simple-display/main.cpp:17: undefined reference to `__ltsf2'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /tmp/ccpkpNDJ.o: in function `drawFunnyRects()':
/home/riscv-vp/riscv-vp/sw/simple-display/main.cpp:52: undefined reference to `__mulsf3'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /tmp/ccpkpNDJ.o: in function `progressBar(bool, Framebuffer::PointF, Framebuffer::PointF, unsigned short, unsigned short, float)':
/home/riscv-vp/riscv-vp/sw/simple-display/main.cpp:57: undefined reference to `__mulsf3'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /tmp/ccpkpNDJ.o: in function `drawFunnyBar(bool)':
/home/riscv-vp/riscv-vp/sw/simple-display/main.cpp:81: undefined reference to `__extendsfdf2'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /home/riscv-vp/riscv-vp/sw/simple-display/main.cpp:84: undefined reference to `__adddf3'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /home/riscv-vp/riscv-vp/sw/simple-display/main.cpp:84: undefined reference to `__truncdfsf2'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /home/riscv-vp/riscv-vp/sw/simple-display/main.cpp:84: undefined reference to `__lesf2'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /home/riscv-vp/riscv-vp/sw/simple-display/main.cpp:83: undefined reference to `__extendsfdf2'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /home/riscv-vp/riscv-vp/sw/simple-display/main.cpp:89: undefined reference to `__subdf3'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /home/riscv-vp/riscv-vp/sw/simple-display/main.cpp:90: undefined reference to `__truncdfsf2'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /home/riscv-vp/riscv-vp/sw/simple-display/main.cpp:90: undefined reference to `__gesf2'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /tmp/ccpkpNDJ.o: in function `Framebuffer::PointF::PointF(Framebuffer::Point)':
/home/riscv-vp/riscv-vp/sw/simple-display/main.cpp:117: undefined reference to `__floatunsisf'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /home/riscv-vp/riscv-vp/sw/simple-display/main.cpp:117: undefined reference to `__floatunsisf'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /tmp/ccpkpNDJ.o: in function `.L0 ':
/home/riscv-vp/riscv-vp/sw/simple-display/main.cpp:117: undefined reference to `__addsf3'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /tmp/ccpkpNDJ.o: in function `operator+(Framebuffer::PointF, Framebuffer::PointF)':
/home/riscv-vp/riscv-vp/sw/simple-display/../../env/basic/vp-display/framebuffer.h:84: undefined reference to `__addsf3'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /tmp/ccieDnlA.o: in function `display::drawRect(Framebuffer::Type, Framebuffer::PointF, Framebuffer::PointF, unsigned short)':
/home/riscv-vp/riscv-vp/sw/simple-display/libDisplay.cpp:31: undefined reference to `__gtsf2'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /home/riscv-vp/riscv-vp/sw/simple-display/libDisplay.cpp:33: undefined reference to `__gtsf2'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /tmp/ccieDnlA.o: in function `display::fillRect(Framebuffer::Type, Framebuffer::PointF, Framebuffer::PointF, unsigned short)':
/home/riscv-vp/riscv-vp/sw/simple-display/libDisplay.cpp:44: undefined reference to `__eqsf2'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /home/riscv-vp/riscv-vp/sw/simple-display/libDisplay.cpp:44: undefined reference to `__eqsf2'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /home/riscv-vp/riscv-vp/sw/simple-display/libDisplay.cpp:45: undefined reference to `__gtsf2'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /home/riscv-vp/riscv-vp/sw/simple-display/libDisplay.cpp:48: undefined reference to `__gtsf2'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /home/riscv-vp/riscv-vp/sw/simple-display/libDisplay.cpp:51: undefined reference to `__subsf3'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /home/riscv-vp/riscv-vp/sw/simple-display/libDisplay.cpp:52: undefined reference to `__subsf3'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /home/riscv-vp/riscv-vp/sw/simple-display/libDisplay.cpp:52: undefined reference to `__gtsf2'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /home/riscv-vp/riscv-vp/sw/simple-display/libDisplay.cpp:54: undefined reference to `__floatunsisf'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /home/riscv-vp/riscv-vp/sw/simple-display/libDisplay.cpp:56: undefined reference to `__floatunsisf'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /home/riscv-vp/riscv-vp/sw/simple-display/libDisplay.cpp:57: undefined reference to `__floatsisf'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /home/riscv-vp/riscv-vp/sw/simple-display/libDisplay.cpp:57: undefined reference to `__lesf2'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /home/riscv-vp/riscv-vp/sw/simple-display/libDisplay.cpp:56: undefined reference to `__floatunsisf'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /home/riscv-vp/riscv-vp/sw/simple-display/libDisplay.cpp:56: undefined reference to `__floatunsisf'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /home/riscv-vp/riscv-vp/sw/simple-display/libDisplay.cpp:62: undefined reference to `__floatsisf'
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: /home/riscv-vp/riscv-vp/sw/simple-display/libDisplay.cpp:62: undefined reference to `__lesf2'
collect2: error: ld returned 1 exit status
make: *** [Makefile:2: all] Error 1
3da74e0444f3:~/riscv-vp/sw/simple-display$ `
`

TommyMurphyTM1234, 2022-12-25:
You haven't really given sufficient info about how you configured and built your toolchain but this error:

`can't link soft-float modules with double-float modules`

and the fact that you're targeting `-march=rv32ima -mabi=ilp32` suggests that you may have configured and built the toolchain for, say, `rv32imafdc`, and without multilibs - in particular the multilibs for `rv32ima/ilp32` so the toolchain tries to link its default libs for hard double float, thus causing the error.

hvishu, 2022-12-26:
Sorry, this might not the right forum to raise the issue.

I was following the steps mentioned in https://github.com/agra-uni-bremen/riscv-vp.  And faced this issue.  I have submitted the issue now in that project https://github.com/agra-uni-bremen/riscv-vp/issues/26.

Just for your information, the configuration used was "./configure --prefix=$(pwd)/../riscv-gnu-toolchain-dist-rv32imac-ilp32 --with-arch=rv32imac --with-abi=ilp32"

Will get the response from that project and update here.

TommyMurphyTM1234, 2022-12-26:
> Just for your information, the configuration used was `./configure --prefix=$(pwd)/../riscv-gnu-toolchain-dist-rv32imac-ilp32 --with-arch=rv32imac --with-abi=ilp32`

Your toolchain was installed into `$(pwd)/../riscv-gnu-toolchain-dist-rv32imac-ilp32` but your build log output seems to indicate that a different path is being used to find the toolchain and associated libs:

`/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld`

Are you sure that you didn't build or install another RISC-V toolchain in your Docker environment and that's the one that's being picked up? And it may be configured for something like `rv32imafdc/ilp32d`?

Cirromulus, 2022-12-26:
I don't think that this repo here is the correct place for this issue.

> Your toolchain was installed into `$(pwd)/../riscv-gnu-toolchain-dist-rv32imac-ilp32` but your build log output seems to indicate that a different path is being used to find the toolchain and associated libs:

This indeed seems correct.
While not being strictly necessary, the `simple-display` example program uses and links to soft-float arithmetics (`-march=rv32ima -mabi=ilp32`).
You are using the system's riscv32-compiler (`/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld`) which comes on nearly every system with hard-float libraries.

> Just for your information, the configuration used was "./configure --prefix=$(pwd)/../riscv-gnu-toolchain-dist-rv32imac-ilp32 --with-arch=rv32imac --with-abi=ilp32"

Then you probably just did not update your `PATH` to actually use this newly built toolchain which then will have soft-float libraries.

TommyMurphyTM1234, 2022-12-26:
Closing this as it's almost certainly a toolchain environment error and not an issue with the riscv-gnu-toolchain tools themselves.

closed 2022-12-26 by TommyMurphyTM1234