riscv-collab/riscv-gnu-toolchain #1172: GCC still use libgcc functions with flag `-fno-builtin` or `-ffreestanding`

cheyao, 2023-01-07:
As you can see with this godbolt example here:
https://godbolt.org/z/5ehTPcaM9
GCC produces the call to the libgcc function `__divdi3` even though there is the flags `-fno-builtin` and `-ffreestanding`, which is quite frustrating as I want my OS to be freestanding (and can't get the linker to link to libgcc XD).

Is there someway to remove the builtin functions? Thanks!

TommyMurphyTM1234, 2023-01-07:
There are many existing discussions elsewhere (e.g. StackOverflow, GCC issue tracker etc.) discussing (and sometimes, arguing over!) the nature, purpose, and rationale for these GCC "helper" functions. And their use even in the presence of, say, `-ffreestanding -fno-builtin`. (Actually `-ffreestanding` already implies `fno-builtin` but the latter is irrelevant in the context of these functions anyway as far as I know.).
* https://gcc.gnu.org/onlinedocs/gccint/Integer-library-routines.html

So, this is not a RISC-V or riscv-gnu-toolchain specific issue but is a general GCC issue.

In case it's of any help, you should be able to link them statically using `-lstatic-libgcc -lgcc`.

cheyao, 2023-01-07:
Ah thanks!

But now the problem is that the helper functions use the C extension (or the one which permets 16bit instruction), which my cpu doesn't support. Is there someway to let libgcc only use the rv64id extension?

TommyMurphyTM1234, 2023-01-07:
> Ah thanks!
> 
> But now the problem is that the helper functions use the C extension (or the one which permets 16bit instruction), which my cpu doesn't support. Is there someway to let libgcc only use the rv64id extension?

I can't see from the Compiler Explorer link what flags you're passing (I'm on a mobile device and it doesn't render very well). Are you passing `-march=rv64id` and maybe also `-mabi=lp64d` although the latter may be implied by default by the architecture?

Bear in mind that some (most?) multilib toolchains may not bundle the multilib for this architecture without customisation if the multilib list at configure time. But if you're not linking libraries or startup code that may not matter.

TommyMurphyTM1234, 2023-01-07:
> Ah thanks!
> 
> But now the problem is that the helper functions use the C extension (or the one which permets 16bit instruction), which my cpu doesn't support. Is there someway to let libgcc only use the rv64id extension?

Specifying `-march=rv64id` should avoid the use/generation of C (compressed) instructions. Does adding `-mabi=lp64d` make any difference? Even though I think it should be implied `-march= rv64id`?

Perhaps the issue is that there are no multilibs for `rv64id/lp64d` and it's using the toolchain default `rv64gc/lp64d` versions?

TommyMurphyTM1234, 2023-01-07:
BTW does Compiler Explorer use only the Linux RISC-V toolchains and not the bare metal/newlib ones? Would the latter be more appropriate for your needs? I'm not sure if/how using the bare metal toolchain might affect the specific issue that you're having here.

cheyao, 2023-01-07:
> Specifying `-march=rv64id` should avoid the use/generation of C (compressed) instructions. Does adding `-mabi=lp64d` make any difference - even though I think it should already be implied by `-march=rv64id`?
> 
> Perhaps the issue is that there are no multilibs for `rv64id/lp64d` and it's falling back to using the toolchain default (most likely `rv64gc/lp64d`) versions (libs, startup code etc.)? `rv64id/lp64d` is not in the default multilib list for either toolchain (Linux or bare metal) as far as I recall.

I tried with -mabi=lp64d, but still the same result. I looked into the gcc folder and there is only the `rv64imac` version of libgcc, do you know how can I get the `rv64i`? Or should I just switch to clang :P

> Is the original Compiler Explorer link still the test that you are performing or have you changed anything?

Nope, i'm on my local machine (MacOS), toolchain installed with brew.

This is my link command: `-L/usr/local/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/12.2.0 -l:libgcc.a -nostdlib`

TommyMurphyTM1234, 2023-01-07:
> do you know how can I get the `rv64i`?

You could build your own toolchain configured using `-with-arch=rv64i`. But did you actually mean `rv64id`?

Alternatively, you could build the `rv64gc/lp64d` toolchain with multilib (including `rv64i/lp64` or `rv64id/lp64d`) support.

But, there are ongoing issues with building certain toolchains due to charges to support canonicalized architecture specifiers, so I have no idea offhand if this would work.

TommyMurphyTM1234, 2023-01-07:
> This is my link command: `-L/usr/local/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/12.2.0 -l:libgcc.a -nostdlib`

Do you actually have to specify the link directory using `-L`?
Why aren't you using `-static-libgcc`?
Since you're not specifying `-nostartfiles` I presume that it's linking the default `rv64gc/lp64d` ones and that's where your compressed instructions are coming from?
A list/map file would clarify the taxonomy and provenance of sections of the linked ELF file.

cheyao, 2023-01-07:
> Do you actually have to specify the link directory using `-L`? 

Yes I have to, or ld will give the error `riscv64-unknown-elf-ld: cannot find -lgcc: file format not recognized`

> Why aren't you using `-static-libgcc`? 

Well adding that command gives me the error
```
riscv64-unknown-elf-ld: Warning: grouped short command line options are deprecated: -static-libgcc
riscv64-unknown-elf-ld: Error: unable to disambiguate: -static-libgcc (did you mean --static-libgcc ?)
```
> Since you're not specifying `-nostartfiles` I presume that it's linking the default `rv64gc/lp64d` ones and that's where your compressed instructions are coming from? 

Well there isn't a `-nostartfiles` flag: `Error: unable to disambiguate: -nostartfiles (did you mean --nostartfiles ?)`

> Or are you sure that they're coming from the "helper" functions? A list/map file would clarify the contents and provenance of sections of the linked ELF file.

I'm sure that they are from the helper functions, I made a objdump to check

TommyMurphyTM1234, 2023-01-07:
I made some mistakes with some of the flags.
I can never remember which use `-` and which use `--`.

I'm trying to build the bare metal toolchain with `configure --with-arch=rv64i --with-abi=lp64` to see how that goes...

TommyMurphyTM1234, 2023-01-07:
Can you clarify if you target is `rv64i` or `rv64id` or something else, please? The thread is a bit confusing on this point.

cheyao, 2023-01-07:
> I still suspect that your test is causing the rv64i[d?]/lp64[d?] program to get linked with rv64gc/lp64d objects, thus leaving you with compressed instructions.

Yea maybe that's the problem, but gcc only left me with the rv64imac libgcc, as you can see, here is the contents of the lib dir where I installed gcc (`/usr/local/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/12.2.0`)

```
$ ls
crtbegin.o    include       libgcc.a      rv32iac       rv64imac
crtend.o      include-fixed libgcov.a     rv32im
crti.o        install-tools plugin        rv32imac
crtn.o        libgcc        rv32i         rv32imafc
```

> Can you clarify if your target is `rv64i` or `rv64id` or something else, please? The thread is a bit confusing on this point.

My target is `rv64i`, but gcc requires at least the d expasion for floating point numbers, so in the flags i'm passing `rv64id`

> And also, if you actually need to use the RISC-V Linux toolchain or if you could/should be using the bare metal one.

I think I should use the bare metal one, which, should be the one that homebrew installed (as the toolchain name is `riscv64-unknown-elf-`)

> So, what is the nature of your target RISC-V architecture and OS (if any)?

None :P

TommyMurphyTM1234, 2023-01-07:
> Yea maybe that's the problem, but gcc only left me with the rv64imac libgcc, as you can see, here is the contents of the lib dir where I installed gcc [...]

That's because whoever built that toolchain only bundled those specific multilibs.

> My target is `rv64i`, but gcc requires at least the d expasion for floating point numbers, so in the flags i'm passing `rv64id`

That's not correct. The D extension means hard double float. (And I think it also implies or requires F?). If you're specifying `-march=rv64id` then you're telling the compiler that the target hardware supports hard double float. `rv64i` does not. So you have a mismatch between the `-march` and the actual target.

When my `rv64i/lp64` toolchain is built I'll be able to experiment...

TommyMurphyTM1234, 2023-01-07:
Also, the `__divdi3` mentioned originally has nothing to do with float/double and the example code in the Compiler Explorer link doesn't even use float/double!

If your target is `rv64i` then you need a toolchain configured specifically for `rv64i/lp64` or a multilib toolchain that bundles the multilibs/startup code/etc. for that.

TommyMurphyTM1234, 2023-01-07:
So, I built a toolchain as follows:
```
cd ~/Downloads
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools --with-arch=rv64i --with-abi=lp64
make
```
and then I compiled, linked and disassembled the test code (the `#include` lines in the original are unnecessary).
```
cd installed-tools/bin
gedit test.c 
...
./riscv64-unknown-elf-gcc test.c -o test.elf
./riscv64-unknown-elf-objdump -dS test.elf > test.dis
```
The `test.c` and `test.dis` files are attached (renamed to `*.txt` so that they can be attached here).
There are no compressed instructions in the disasembled file - in particular in the implementation of `__divsi3` or any other helper functions.
There seems to be no issue here once the correct toolchain is used properly.

* [test.c.txt](https://github.com/riscv-collab/riscv-gnu-toolchain/files/10367117/test.c.txt)
* [test.dis.txt](https://github.com/riscv-collab/riscv-gnu-toolchain/files/10367120/test.dis.txt)

```

TommyMurphyTM1234, 2023-01-07:
Closing as not a valid issue.

closed 2023-01-07 by TommyMurphyTM1234