riscv-collab/riscv-gnu-toolchain #1174: Build hangs at cloning riscv-gnu-toolchain/gcc

Plociennik, 2023-01-09:
Dear all,

I tried to build `riscv-gnu-toolchain` as described in the `README`, i.e., I tried to execute

```
git clone https://github.com/riscv/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=<my prefix> --with-arch=rv32ima --with-abi=ilp32
make
```

This results in the following output:

```
cd /home/kai/home/tmp/riscv-gnu-toolchain && \
flock `git rev-parse --git-dir`/config git submodule init /home/kai/home/tmp/riscv-gnu-toolchain/gcc/ && \
flock `git rev-parse --git-dir`/config git submodule update /home/kai/home/tmp/riscv-gnu-toolchain/gcc/
Submodule 'gcc' (https://gcc.gnu.org/git/gcc.git) registered for path 'gcc'
Cloning into '/home/kai/home/tmp/riscv-gnu-toolchain/gcc'...
```

At this point, the build hangs, i.e., cloning `riscv-gnu-toolchain/gcc` does not work.

Last year, I was able to successfully clone and build the toolchain. Back then, I used

```
git clone --recurse-submodules https://github.com/riscv/riscv-gnu-toolchain
```

to clone the repo. However, the setup seems to have changed, i.e., submodules now seem to be cloned lazily when calling `make`, as stated in the `README`. I am unsure what the problem is or whether I can safely return to using some kind of explicit recursive cloning, since I suppose this could interfere with the commands executed in the `Makefile`.

Any hints on this would be greatly appreciated.

TommyMurphyTM1234, 2023-01-09:
It's working fine for me - but just takes a while.
What happens if you manually try to clone the gcc repo just as a test?
```
git clone --progress --verbose https://gcc.gnu.org/git/gcc.git -b releases/gcc-12
```

TommyMurphyTM1234, 2023-01-09:
This is what I got for a manual clone of the gcc repo:
```
user@hornbill:~/Downloads$ git clone --progress --verbose https://gcc.gnu.org/git/gcc.git -b releases/gcc-12
Cloning into 'gcc'...
POST git-upload-pack (gzip 12515 to 6298 bytes)
remote: Enumerating objects: 182661, done.
remote: Counting objects: 100% (182661/182661), done.
remote: Compressing objects: 100% (16363/16363), done.
remote: Total 2791320 (delta 174835), reused 168944 (delta 166181), pack-reused 2608659
Receiving objects: 100% (2791320/2791320), 1.06 GiB | 4.74 MiB/s, done.
Resolving deltas: 100% (2290749/2290749), done.
Updating files: 100% (116086/116086), done.
```
I've also kicked off a tools build and it's proceeding normally having cloned the gcc repo (albeit slowly).

TommyMurphyTM1234, 2023-01-09:
Closing as not a riscv-gnu-toolchain issue.
Possibly an issue with (transient?) sluggishness with the upstream gcc repo.

closed 2023-01-09 by TommyMurphyTM1234

Plociennik, 2023-01-10:
Thanks for your hints! I tried it again today, and after about ten minutes the build continues. For some reason, this took even longer yesterday, I think, but now it works.

TommyMurphyTM1234, 2023-01-10:
I guess it's just a (hopefully transient/temporary) slowdown of the GCC repo. In case it helps, to avoid cloning every toolchain build you could clone once and use `--with-gcc-src`. But then it's up to you to keep the sources up to date.

* https://github.com/riscv-collab/riscv-gnu-toolchain#use-source-tree-other-than-riscv-gnu-toolchain