riscv-collab/riscv-gnu-toolchain #1175: ld -shared not supported when I build riscv-linux

ardeal, 2023-01-10:
Hi,

my envrionment:
ubuntu 20.04,
toolchain: riscv32-elf-ubuntu-20.04-nightly-2022.12.17-nightly.tar.gz
riscv-linux link: https://github.com/CSL-KU/riscv-linux

I would like to build riscv-linux and then run it on embedded RISCV MCU. I am experiencing the following issue

where should I add `static` option in Makefile? 
or, should I change toolchain option?

```C

  CALL    scripts/checksyscalls.sh
  CHK     scripts/mod/devicetable-offsets.h
  CHK     include/generated/compile.h
  VDSOLD  arch/riscv/kernel/vdso/vdso.so.dbg
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: -shared not supported
collect2: error: ld returned 1 exit status
  OBJCOPY arch/riscv/kernel/vdso/vdso.so
riscv32-unknown-elf-objcopy: 'arch/riscv/kernel/vdso/vdso.so.dbg': No such file
make[2]: *** [arch/riscv/kernel/vdso/Makefile:48: arch/riscv/kernel/vdso/vdso.so] Error 1
make[1]: *** [scripts/Makefile.build:575: arch/riscv/kernel/vdso] Error 2
make: *** [Makefile:1018: arch/riscv/kernel] Error 2

```

TommyMurphyTM1234, 2023-01-11:
On what basis do you think that this is a riscv-gnu-toolchain issue and not a build issue with the Linux port that you're trying to build and which you should be asking about here?

* https://github.com/CSL-KU/riscv-linux

ardeal, 2023-01-11:
@TommyMurphyTM1234 
Thank you for your reply!
I don't think this is a riscv-gnu-toolchain issue. I think it should be a issue about how to config static lib in Makefile, or how to config it in toolchain.

I searched "riscv linux ld shared not supported"  in Google, and got the following link:
https://github.com/riscv-collab/riscv-gnu-toolchain/issues/335
someone othere asked the same issue in this repo.

I did asked some questions in https://github.com/CSL-KU/riscv-linux, but I didn't get the reply.

Dear @TommyMurphyTM1234 ,
Do you have a good solution about the issue I am experiencing?

TommyMurphyTM1234, 2023-01-11:
> @TommyMurphyTM1234 Thank you for your reply! I don't think this is a riscv-gnu-toolchain issue.

I agree. So I am consequently closing this issue.

> I think it should be a issue about how to config static lib in Makefile, or how to config it in toolchain.

This is not the place to ask for general compilation issues or how to build a Linux port totally unrelated to this repo.

> I searched "riscv linux ld shared not supported" in Google, and got the following link: #335 someone othere asked the same issue in this repo.

And did you need the answer here?

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/335#issuecomment-388638448

> I did asked some questions in https://github.com/CSL-KU/riscv-linux, but I didn't get the reply.

I don't see where you asked over there. All I see is this:

* https://github.com/CSL-KU/riscv-linux/issues/2

> Dear @TommyMurphyTM1234 , Do you have a good solution about the issue I am experiencing?

Your question is already answered here as far as I can see - you need to use the RISC-V Linux rather than the bare metal toolchain to build a Linux kernel.

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/335#issuecomment-388638448
*

closed 2023-01-11 by TommyMurphyTM1234

ardeal, 2023-01-12:
> I don't see where you asked over there. All I see is this:

@TommyMurphyTM1234 
Thank you!

I just now submitted the question at: https://github.com/CSL-KU/riscv-linux/issues/3

Could you please have a look at it?

TommyMurphyTM1234, 2023-01-12:
> > I don't see where you asked over there. All I see is this:
> 
> @TommyMurphyTM1234 Thank you!
> 
> I just now submitted the question at: [CSL-KU/riscv-linux#3](https://github.com/CSL-KU/riscv-linux/issues/3)
> 
> Could you please have a look at it?

Your question has already been addressed. You need to use the RISC-V Linux rather than bare metal toolchain. You can download binary builds from here:

* https://github.com/riscv-collab/riscv-gnu-toolchain/releases

ardeal, 2023-01-12:
@TommyMurphyTM1234 
my question is:
 what is RISC-V Linux?
could you please help to tell me the exact name/string in the toolchain name?

TommyMurphyTM1234, 2023-01-12:
> @TommyMurphyTM1234 my question is: what is RISC-V Linux? 

I don't know what you mean. There are various Linux versions/ports for RISC-V. There isn't just one.

> could you please help to tell me the exact name/string in the toolchain name?

The RISC-V Linux toolchain archives here are the ones named `riscvXX-glibc-*` and `riscvXX-musl-*` depending on which standard library implementation is used.

* https://github.com/riscv-collab/riscv-gnu-toolchain/releases

The tools are named with the prefix `riscvXX-unknown-linux-gnu-*` - for example, `riscv64-unknown-linux-gnu-gcc`.

There are plenty of resources elsewhere on how to use the various RISC-V toolchains.

ardeal, 2023-01-12:
@TommyMurphyTM1234 
I got it. Thank you!
your comments: 
The RISC-V Linux toolchain archives here are the ones named riscvXX-glibc-* and riscvXX-musl-* depending on which standard library implementation is used.

match my questions.

TommyMurphyTM1234, 2023-01-12:
> @TommyMurphyTM1234 I got it. Thank you! your comments: The RISC-V Linux toolchain archives here are the ones named riscvXX-glibc-* and riscvXX-musl-* depending on which standard library implementation is used.
> 
> match my questions.

Glad it helped.