riscv-collab/riscv-gnu-toolchain #1177: rvv-next: vectorization of single complex produces incorrect result

angsch, 2023-01-15:
Hello,

I have built the riscv64-unknown-linux-gnu compiler based on the rvv-next branch. Depending on the compiler optimization level, different results are computed for the routine below. The routine takes a complex vector and divides every entry by a complex scalar.

```
// main.c
#include <stdio.h>
#include <math.h>

typedef struct scomplex
{
    float  real;
    float  imag;
} scomplex;

void cinvscalv_rv64iv(int n, scomplex* restrict alpha, scomplex* restrict x) {
    scomplex alpha_conj; 
    alpha_conj.real = (*alpha).real;
    alpha_conj.imag = (*alpha).imag;

    float s;
    if (fabs(alpha_conj.real) > fabs(alpha_conj.imag))
        s = fabs(alpha_conj.real);
    else
        s = fabs(alpha_conj.imag);

    float ar_s = alpha_conj.real / s;
    float ai_s = alpha_conj.imag / s;
    float temp = ar_s * alpha_conj.real + ai_s * alpha_conj.imag;

    for ( int i = 0; i < n; ++i ) {
        float xrt = x[i].real;
        x[i].real = ( xrt * ar_s + x[i].imag * ai_s ) / temp;
        x[i].imag = ( x[i].imag * ar_s - xrt * ai_s ) / temp;
    } (note -O2)
}

int main(int argc, char **argv) {
    int n = 10;

    scomplex v[n];
    for (int k = 0; k < n; k++) {
        v[k].real = 1.5*k; v[k].imag = 1.0;
    }

    printf("Input vector v:\n");
    for (int k = 0; k < n; k++) {
        printf("%.6e + %.6ei\n", v[k].real, v[k].imag);
    }

    scomplex alpha_conj;
    alpha_conj.real = 1.0;
    alpha_conj.imag = -1.3;
    cinvscalv_rv64iv(n, &alpha_conj, v);

    printf("\n\nOutput: v[k] <- v[k] / (1-1.3I)\n");
    for (int k = 0; k < n; k++) {
        printf("%.6e + %.6ei\n", v[k].real, v[k].imag);
    }
}
```

### Correct behavior at -O1

Compile: `riscv64-unknown-linux-gnu-gcc -march=rv64iv -Wall -O1 main.c -static`
Run: `qemu-riscv64 -cpu rv64,vext_spec=v1.0,v=true,vlen=128 -B 0x100000 ./a.out`
Correct output:
```
[...]
Output: v[k] <- v[k] / (1-1.3I)
-4.832714e-01 + 3.717472e-01i
7.434944e-02 + 1.096654e+00i
6.319702e-01 + 1.821561e+00i
1.189591e+00 + 2.546468e+00i
1.747212e+00 + 3.271375e+00i
2.304833e+00 + 3.996283e+00i
2.862454e+00 + 4.721189e+00i
3.420074e+00 + 5.446096e+00i
3.977695e+00 + 6.171004e+00i
4.535316e+00 + 6.895911e+00i
```
### Incorrect behavior at -O2

Compile: `riscv64-unknown-linux-gnu-gcc -march=rv64iv -Wall -O2 main.c -static`
Run: `qemu-riscv64 -cpu rv64,vext_spec=v1.0,v=true,vlen=128 -B 0x100000 ./a.out`
```
Output: v[k] <- v[k] / (1-1.3I)
-4.832714e-01 + 3.717472e-01i
5.576208e-01 + 8.550186e-01i   // wrong starting from here
6.319702e-01 + 1.821561e+00i
1.672863e+00 + 8.550186e-01i
1.747212e+00 + 3.271375e+00i
2.788104e+00 + 8.550186e-01i
2.862454e+00 + 4.721189e+00i
3.903346e+00 + 8.550186e-01i
3.977695e+00 + 6.171004e+00i
5.018588e+00 + 8.550186e-01i
```
The very first entry of the output vector is correct, all other entries are wrong. 

### Further observations

- When `vlen` is set to another value in the qemu run, the output vector changes for `-O2`.
- Replacing all occurrences of `float` with `double` in the program yields a correct computation regardless of the compiler optimization level. In other words, a computation in double complex works.

### Version & configuration

```
Target: riscv64-unknown-linux-gnu
Configured with: [...]/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-linux-gnu --prefix=[...]/riscv-gnu-toolchain/buildrv64gv --with-sysroot=[...]/riscv-gnu-toolchain/buildrv64gv/sysroot --with-pkgversion=g42df3464463 --with-system-zlib --enable-shared --enable-tls --enable-languages=c,c++,fortran --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libsanitizer --disable-nls --disable-bootstrap --src=.././riscv-gcc --disable-multilib --with-abi=lp64d --with-arch=rv64gv --with-tune=rocket --with-isa-spec=2.2 'CFLAGS_FOR_TARGET=-O2   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-O2   -mcmodel=medlow'
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 12.0.1 20220505 (prerelease) (g42df3464463) 
```
```
qemu-riscv64 version 7.1.50 (v7.1.0-1123-g0529245488)
```

zhongjuzhe, 2023-01-15:
You are not using the latest rvv-gcc. Can you make sure you are using latest RVV GCC:
https://github.com/riscv-collab/riscv-gcc/commit/eb96155f49deefcde0b054f6c02aa67c607ac482
?

TommyMurphyTM1234, 2023-01-15:
Hi @zhongjuzhe - would you be able to outline or point to instructions for building the latest bleeding edge RVV toolchain as this seems to repeatedly cause confusion for those who want to experiment with vector extension enabled tools? Thanks.

angsch, 2023-01-15:
I would be grateful for a quick explanation on the relationship between the branches. rvv-next is stale now (the latest commit there is dated after the one you pointed at) and riscv-rvv-next-gcc is the one to take?

Checking out your code now and testing...

angsch, 2023-01-15:
I second Tommy.  `./configure --target=riscv64-linux-gnu --prefix=[...]`, followed by `make` fails when multilib support is built. I do not see a configure option to only build lp64d.

TommyMurphyTM1234, 2023-01-15:
> I second Tommy. `./configure --target=riscv64-linux-gnu --prefix=[...]`, followed by `make` fails when multilib support is built.

This may be unrelated to V extension support. Possibly multilib build issues relating to canonicalised architecture strings. E.g.:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1126

> I do not see a configure option to only build lp64d.

What do you mean by this? `lp64d` is an ABI. What corresponding architecture do you need? E.g. if you want to build for, say, `rv64gcv/lp64d` then try this:
```
./configure --prefix=`pwd`/installed-tools --with-arch=rv64gcv --with-abi=lp64d
```

angsch, 2023-01-17:
Thanks, Tommy, for your help. I managed to build the toolchain.

```
$CC -v
Using built-in specs.
COLLECT_GCC=[...]/rv64gcv/bin/riscv64-unknown-linux-gnu-gcc
COLLECT_LTO_WRAPPER=[..]/rv64gcv/libexec/gcc/riscv64-unknown-linux-gnu/12.2.0/lto-wrapper
Target: riscv64-unknown-linux-gnu
Configured with: [...]/configure --target=riscv64-unknown-linux-gnu --prefix=[...] --with-sysroot=[...]/rv64gcv/sysroot --with-pkgversion=geb96155f49d --with-system-zlib --enable-shared --enable-tls --enable-languages=c,c++,fortran --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libsanitizer --disable-nls --disable-bootstrap --src=.././riscv-gcc --disable-multilib --with-abi=lp64d --with-arch=rv64gcv --with-tune=rocket --with-isa-spec=2.2 'CFLAGS_FOR_TARGET=-O2   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-O2   -mcmodel=medlow'
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 12.2.0 (geb96155f49d) 
```

@zhongjuzhe: I compiled the above code with the new compiler build. The problem remains.

```
$CC -march=rv64gcv -Wall -O1 main.c -static
qemu-riscv64 -cpu rv64,vext_spec=v1.0,v=true,vlen=128 -B 0x100000 ./a.out
[...]
4.535316e+00 + 6.895911e+00i // last line
```

```
$CC -march=rv64gcv -Wall -O2 main.c -static
qemu-riscv64 -cpu rv64,vext_spec=v1.0,v=true,vlen=128 -B 0x100000 ./a.out
[...]
5.018588e+00 + 8.550186e-01i  // wrong
```
Could you please try to reproduce the problem on your side?

zhongjuzhe, 2023-01-17:
Sure. Thanks for reporting this.

***@***.***
 
From: angsch
Date: 2023-01-18 02:30
To: riscv-collab/riscv-gnu-toolchain
CC: Ju-Zhe Zhong; Mention
Subject: Re: [riscv-collab/riscv-gnu-toolchain] rvv-next: vectorization of single complex produces incorrect result (Issue #1177)
Thanks, Tommy, for your help. I managed to build the toolchain.
$CC -v
Using built-in specs.
COLLECT_GCC=[...]/rv64gcv/bin/riscv64-unknown-linux-gnu-gcc
COLLECT_LTO_WRAPPER=[..]/rv64gcv/libexec/gcc/riscv64-unknown-linux-gnu/12.2.0/lto-wrapper
Target: riscv64-unknown-linux-gnu
Configured with: [...]/configure --target=riscv64-unknown-linux-gnu --prefix=[...] --with-sysroot=[...]/rv64gcv/sysroot --with-pkgversion=geb96155f49d --with-system-zlib --enable-shared --enable-tls --enable-languages=c,c++,fortran --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libsanitizer --disable-nls --disable-bootstrap --src=.././riscv-gcc --disable-multilib --with-abi=lp64d --with-arch=rv64gcv --with-tune=rocket --with-isa-spec=2.2 'CFLAGS_FOR_TARGET=-O2   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-O2   -mcmodel=medlow'
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 12.2.0 (geb96155f49d) 

@zhongjuzhe: I compiled the above code with the new compiler build. The problem remains.
$CC -march=rv64gcv -Wall -O1 main.c -static
qemu-riscv64 -cpu rv64,vext_spec=v1.0,v=true,vlen=128 -B 0x100000 ./a.out
[...]
4.535316e+00 + 6.895911e+00i // last line
$CC -march=rv64gcv -Wall -O2 main.c -static
qemu-riscv64 -cpu rv64,vext_spec=v1.0,v=true,vlen=128 -B 0x100000 ./a.out
[...]
5.018588e+00 + 8.550186e-01i  // wrong

Could you please try to reproduce the problem on your side?

zhongjuzhe, 2023-01-18:
> hanks, Tommy, for your help. I managed to build the toolchain.

Hi, we are on vacation (Chinese Spring Festive) which is about a week. We will definetely fix it for you. Thanks for report this issue.

TommyMurphyTM1234, 2023-02-05:
Hi @zhongjuzhe - just wondering if there's any update on this issue? Thanks.

zhongjuzhe, 2023-03-16:
https://github.com/riscv-collab/riscv-gcc/pull/370
Issue is fixed.

zhongjuzhe, 2023-03-17:
You should enable “M” extension with -march=rv64imv for example. otherwise, compilerCan not use multiplification. This is riscv isa issue. Not compiler issue---- Replied Message ***@***.***>Date03/17/2023 22:44 ***@***.***> CcJu-Zhe ***@***.***>***@***.***>SubjectRe: [riscv-collab/riscv-gnu-toolchain] rvv-next: vectorization of single complex produces incorrect result (Issue #1177)
@zhongjuzhe  Thank you! I confirm that the bug is fixed using the linked patch.
The reproducer originates from BLIS (https://github.com/flame/blis). When I compile BLIS using march=rv64iv with the latest version of riscv-gcc-rvv-next, I hit an internal compiler error. The compiler error disappears when I use rv64imv. With that flag, all tests pass at -O3 - this is great work!
Perhaps the trace gives you a clue what is wrong. I have not been able to reduce the code to a standalone reproducer.
$CC -v
Using built-in specs.
COLLECT_GCC=/home/angelika/Documents/toolchain2/rv64gcv/bin/riscv64-unknown-linux-gnu-gcc
COLLECT_LTO_WRAPPER=/home/angelika/Documents/toolchain2/rv64gcv/libexec/gcc/riscv64-unknown-linux-gnu/12.2.0/lto-wrapper
Target: riscv64-unknown-linux-gnu
Configured with: [...]/configure --target=riscv64-unknown-linux-gnu --prefix=[...]/rv64gcv --with-sysroot=[...]/rv64gcv/sysroot --with-pkgversion=ga1861b78ada --with-system-zlib --enable-shared --enable-tls --enable-languages=c,c++,fortran --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libsanitizer --disable-nls --disable-bootstrap --src=.././riscv-gcc --disable-multilib --with-abi=lp64d --with-arch=rv64gcv --with-tune=rocket --with-isa-spec=2.2 'CFLAGS_FOR_TARGET=-O2   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-O2   -mcmodel=medlow'
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 12.2.0 (ga1861b78ada)

Compiling obj/rv64iv/ref_kernels/rv64iv/3/bli_gemm_rv64iv_ref.o ('rv64iv' CFLAGS for ref. kernels)
during RTL pass: expand
In file included from ref_kernels/3/bli_gemm_ref.c:35:
ref_kernels/3/bli_gemm_ref.c: In function 'bli_sgemm_rv64iv_ref':
./frame/include//bli_macro_defs.h:58:36: internal compiler error: in riscv_expand_quotient, at config/riscv/riscv-vector.cc:1123
   58 | #define PASTEMAC3(ch1,ch2,ch3,op)  PASTEMAC3_(ch1,ch2,ch3,op)
      |                                    ^~~~
./frame/include//bli_macro_defs.h:59:36: note: in expansion of macro 'PASTEMAC3_'
   59 |  
      |                                    ^          
ref_kernels/3/bli_gemm_ref.c:152:6: note: in expansion of macro 'PASTEMAC3'
  152 | void PASTEMAC3(ch,opname,arch,suf) \
      |      ^~~~~~~~~
./frame/include//bli_gentfunc_macro_defs.h:171:1: note: in expansion of macro 'GENTFUNC'
  171 | GENTFUNC( double,   d, tfuncname, varname1, varname2 ) \
      | ^~~~~~~~
ref_kernels/3/bli_gemm_ref.c:300:1: note: in expansion of macro 'INSERT_GENTFUNC_BASIC2'
  300 | INSERT_GENTFUNC_BASIC2( gemm, BLIS_CNAME_INFIX, BLIS_REF_SUFFIX )
      | ^~~~~~~~~~~~~~~~~~~~~~
0x7f6066 riscv_expand_quotient
	../.././riscv-gcc/gcc/config/riscv/riscv-vector.cc:1123
0x1161216 riscv_expand_quotient
	../.././riscv-gcc/gcc/config/riscv/riscv-vector.cc:1019
0x1161216 riscv_vector_expand_poly_move(machine_mode, rtx_def*, rtx_def*, rtx_def*)
	../.././riscv-gcc/gcc/config/riscv/riscv-vector.cc:1187
0x10babfe riscv_legitimize_move(machine_mode, rtx_def*, rtx_def*)
	../.././riscv-gcc/gcc/config/riscv/riscv.cc:1887
0x1741772 gen_movdi(rtx_def*, rtx_def*)
	../.././riscv-gcc/gcc/config/riscv/riscv.md:1835
0xa68068 rtx_insn* insn_gen_fn::operator()<rtx_def*, rtx_def*>(rtx_def*, rtx_def*) const
	../.././riscv-gcc/gcc/recog.h:407
0xa68068 emit_move_insn_1(rtx_def*, rtx_def*)
	../.././riscv-gcc/gcc/expr.cc:3955
0xa684a5 emit_move_insn(rtx_def*, rtx_def*)
	../.././riscv-gcc/gcc/expr.cc:4125
0xa42fc2 copy_to_mode_reg(machine_mode, rtx_def*)
	../.././riscv-gcc/gcc/explow.cc:654
0xcb029c maybe_legitimize_operand
	../.././riscv-gcc/gcc/optabs.cc:7803
0xcb029c maybe_legitimize_operands(insn_code, unsigned int, unsigned int, expand_operand*)
	../.././riscv-gcc/gcc/optabs.cc:7935
0xcac7dd maybe_gen_insn(insn_code, unsigned int, expand_operand*)
	../.././riscv-gcc/gcc/optabs.cc:7954
0xcb361d expand_binop_directly
	../.././riscv-gcc/gcc/optabs.cc:1442
0xcb1838 expand_binop(machine_mode, optab_tag, rtx_def*, rtx_def*, rtx_def*, int, optab_methods)
	../.././riscv-gcc/gcc/optabs.cc:1529
0xa5ea91 expand_expr_real_2(separate_ops*, rtx_def*, machine_mode, expand_modifier)
	../.././riscv-gcc/gcc/expr.cc:10303
0x958305 expand_gimple_stmt_1
	../.././riscv-gcc/gcc/cfgexpand.cc:3972
0x958305 expand_gimple_stmt
	../.././riscv-gcc/gcc/cfgexpand.cc:4033
0x95cc87 expand_gimple_basic_block
	../.././riscv-gcc/gcc/cfgexpand.cc:6080
0x95ecf6 execute
	../.././riscv-gcc/gcc/cfgexpand.cc:6806
Please submit a full bug report, with preprocessed source (by using -freport-bug).
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.
make: *** [Makefile:691: obj/rv64iv/ref_kernels/rv64iv/3/bli_gemm_rv64iv_ref.o] Error 1

[
{
"target": "https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1177#issuecomment-1473952694",
"url": "https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1177#issuecomment-1473952694",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

closed 2023-03-18 by angsch