riscv-collab/riscv-gnu-toolchain #1178: attribute AT

yy172, 2023-01-18:
I think __attribute__ (AT xxx) is used to define the address of variables or functions, which is very practical in embedded development. Is there any plan to add this attribute?

kito-cheng, 2023-01-18:
Do you mean the something like DW_AT_type which defined in DWARF spec? if so, RISC-V using standard dwarf spec, and define only few RISC-V specific stuffs, see more: https://github.com/riscv-non-isa/riscv-elf-psabi-doc/blob/master/riscv-dwarf.adoc

TommyMurphyTM1234, 2023-01-18:
Or does @yy172 mean the likes of the arm `at` attribute?

* https://developer.arm.com/documentation/dui0375/g/Compiler-specific-Features/--attribute----at-address----variable-attribute

TommyMurphyTM1234, 2023-01-19:
Hi @yy172 - can you clarify if you're referring to the arm `at` attribute? You refer to variables or functions. The arm `at` attribute only deals with variables. As far as I know, the GCC equivalent is to use `.section` attributes and linker script section references to do something similar:

* https://stackoverflow.com/questions/4067811/how-to-place-a-variable-at-a-given-absolute-address-in-memory-with-gcc

TommyMurphyTM1234, 2023-01-19:
This blog post might also be of use in this context...

* https://mcuoneclipse.com/2012/11/01/defining-variables-at-absolute-addresses-with-gcc/

TommyMurphyTM1234, 2023-01-21:
Any update @yy172?
I think that this issue can be closed as invalid.

yy172, 2023-01-30:
I'm sorry that it took so long to reply. What I said is similar to the “ at” attribute in armcc. This effect can be achieved by using section and modifying the LD file. However, when modifying the LD, you need to manually readjust the FLASH or RAM area where this section is placed to avoid wasting or address conflict. It is not particularly convenient

TommyMurphyTM1234, 2023-01-30:
> What I said is similar to the “ at” attribute in armcc. 

In that case you'd need to log an enhancement request upstream at the GCC project since this is not really a RISC-V specific issue. 

> However, when modifying the LD, you need to manually readjust the FLASH or RAM area where this section is placed to avoid wasting or address conflict. It is not particularly convenient

Can you give an example please?

Modifying the linker script for an embedded target is normally necessary in order to use the embedded target efficiently.

closed 2023-01-30 by TommyMurphyTM1234