riscv-collab/riscv-gnu-toolchain #1179: make reports "make: ...../riscv-gnu-toolchain/build/../scripts/march-to-cpu-opt: Command not found"

TallAllie, 2023-01-20:
I have commit `65056bdb149c87db4e7223c4e8b5466cf326ff86`.

Submodule status is:
 ```
b51c2fec1da205ea3e7354cbb3e253018d64873c binutils (binutils-2_39)
 d51e3f9f8188a735a64e7b2348e8fb7084f1d666 dejagnu (dejagnu-1.6.3-release)
 2ee5e4300186a92ad73f1a1a64cb918dc76c8d67 gcc (releases/gcc-12.2.0)
 e53a8e8685685c97588f8319d993ea6cd5635e47 gdb (gdb-12.1-release)
 9826b03b747b841f5fc6de2054bf1ef3f5c4bdf3 glibc (glibc-2.33)
 85e0e3519655220688e757b9d5bfd314923548bd musl (v1.2.2)
 415fdd4279b85eeec9d54775ce13c5c412451e08 newlib (newlib-4.1.0)
 2efabd3e6604b8a9e8f70baf52f57696680c7855 pk (v1.0.0-89-g2efabd3)
 621da7789083b80d6f1ff1c0fb499334007b4f51 qemu (v7.1.0)
 a0298a33e7b2091ba8d9f3a20838d96dc1164cac spike (v1.1.0-212-ga0298a33)
```

I want baremetal, so configure like this:
```
%> cd <repos>/riscv-gnu-toolchain
%> git pull
%> mkdir build
%> cd build
%> ../configure \
  --prefix=/tools/xtools/riscv-gnu-toolchain \
  --disable-linux \
  --enable-gdb \
  --enable-multilib \
  --enable-qemu-system \
  --with-arch=rv64imafdqc_zicsr_zifencei \
  --with-cmodel=medany \
  --with-isa-spec=20191213 \
  --with-multilib-generator \
  --without-system-zlib \
  ;
%> make
```

What am I missing here...?
Thank you!

TommyMurphyTM1234, 2023-01-20:
> This fails becuase there is no file below `scripts` with name `march-to-cpu-opt`. 

There is:

* https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/scripts/march-to-cpu-opt

Perhaps your checked out commit predates this?

TallAllie, 2023-01-20:
Ah, I missed the `README.md` instructions for out-of-tree builds...

closed 2023-01-20 by TallAllie

TallAllie, 2023-01-20:
The problem is actually that the first line of "march-to-cpu-opt" uses `#!/usr/bin/python3`.

For me, I replace that with `#!/usr/bin/env python3`.

cmuellner, 2023-01-20:
I think that would be a reasonable change.
Feel free to file a PR.

TommyMurphyTM1234, 2023-01-21:
I think that @TallAllie mentioned previously that you're doing an out of tree build using some sources from upstream (although that comment seems to have disappeared). I guess that the script in question is simply one of the RISC-V specific changes that hasn't yet been upstreamed so it's only local to this repo for now.