riscv-collab/riscv-gnu-toolchain #1180: feature request: Support inline assembly with vector instructions

angsch, 2023-01-22:
Hello,

We would be grateful if the next stable riscv-gcc release could support inline assembly with vector instructions. Currently, the latest stable release in the master branch does not allow clobbering vector registers. 

```
#include <stdio.h>

int main(int argc, char **argv) {
    double a_ptr[10];
    for (int i = 0; i < 10; i++)
        a_ptr[i] = (double)(i + 1);

    // Zero-out the first vlenb/sizeof(double) entries
    asm volatile (
       " vsetvli zero, zero, e64, m1, ta, ma   \n\t"
       " fmv.d.x fa3, zero                     \n\t"
       " vfmv.v.f v0, fa3                      \n\t"
       " vse64.v v0, %[a_ptr]                  \n\t"
      :
      : [a_ptr] "m" (a_ptr)
      : "v0", "fa3"
    );

    // Expect the first vlenb/sizeof(double) to be zero
    for (int i = 0; i < 10; i++)
        printf("%.6e\n", a_ptr[i]);
}
```
```
$CC -v
[...]
gcc version 12.2.0 (g2ee5e430018) 
$CC -march=rv64iv -O1 main.c -static -Wall
main.c: In function 'main':
main.c:9:5: error: unknown register name 'v0' in 'asm'
```

The above code works with the `riscv-gcc-rvv-next` branch. Would it be possible to cherry-pick the commits that enable clobbering and include them in the next stable release?

cmuellner, 2023-01-30:
There is no such riscv-gcc release.
Instead, all code that is ready and targeting ratified (or at least frozen) extensions is going to the upstream GCC repository.

You might refer to the riscv-gcc repository, which is used as a shared and public git repository to develop the RISC-V vector extension support. However, note that this branch is not considered completed but represents a development snapshot under constant improvement and review. Parts considered ready are carried to the GCC mailing list, where they are reviewed again and merged upstream (if reviews are completed).

All of this is independent of this repository, which provides a build framework for RISC-V toolchains.

Regarding the issue itself:
My assumption is that the "v0" in the clobber list is causing the error.
If so, then you could probably simply remove "v0" from the clobber list, because if GCC does not know of this register, it won't care about its content getting overwritten by the inline assembly sequence.

angsch, 2023-01-31:
> If so, then you could probably simply remove "v0" from the clobber list, because if GCC does not know of this register, it won't care about its content getting overwritten by the inline assembly sequence.

The clobbering issue is a bit inconvenient because it prevents compilation of existing inline assembly code (for example the development going on at OpenBLAS).

Close due to moving the issue to the riscv-gcc repository

closed 2023-01-31 by angsch