riscv-collab/riscv-gnu-toolchain #1181: Atomics not working with Newlib toolchain

Plociennik, 2023-01-25:
Dear all,

I am trying to use the Newlib toolchain for the `ima/ilp32` architecture and ABI to compile source code making use of atomics, e.g., the following one (from the LLVM code base):

```
#include <atomic>

std::atomic<int> x;
std::atomic<short> y;
std::atomic<char> z;

int main() {
  ++z;
  ++y;
  return ++x;
}
```

When trying to build this using `riscv32-unknown-elf-g++` I get the following error:

```
test.cpp:(.text._ZNSt13__atomic_baseIcEppEv[_ZNSt13__atomic_baseIcEppEv]+0x24): undefined reference to `__atomic_fetch_add_1'
.../riscv32-unknown-elf/bin/ld: test.cpp:(.text._ZNSt13__atomic_baseIsEppEv[_ZNSt13__atomic_baseIsEppEv]+0x24): undefined reference to `__atomic_fetch_add_2'
```

I tried to supply `-latomic` but it seems this library is not included in the Newlib build.

I am having trouble to understand what to do here. I can find the C++ `atomic` header in my toolchain build, and the compiler also seems to generate atomic builtins, but I cannot find any definition for the required functions in any of the libraries.

As mentioned, I am using the `ima` ISA, so my target machine has atomics support, but I cannot see how I could use it.

Am I doing something wrong here and is there a way to make this work with the Newlib toolchain? I noticed that `libatomic` is built for the Linux toolchain, but I cannot use that one since I want to run my code on a bare metal target.

Any help would be greatly appreciated.

TommyMurphyTM1234, 2023-01-25:
Is this of any relevance?

* https://www.spinics.net/lists/gcchelp/msg52874.html

Plociennik, 2023-01-25:
Thanks a lot!

Yes, this clarifies the situation and I think the idea of using the necessary code from the Linux `libatomic` is something I will try.

As a note, I tried to compile the code without the `short` and `char` atomics, as you suggested, and this compiles. However, I have different kinds of code I need to compile and cannot yet see which exact types of atomics are included, or whether I can change it, so having a general approach such as providing the necessary atomics code using things from the Linux implementation is good.

TommyMurphyTM1234, 2023-01-25:
Do you have to use the bare metal/newlib toolchain or could you use the Linux one? Maybe with `-ffreestanding`? Or would the resulting executables still depend on Linux OS features that you don't have?

Plociennik, 2023-01-25:
Hmm, that is a good question. The problem is that I have to integrate with lots of existing code, some of which provides low level calls to make Newlib run on our specific bare metal hardware. Furthermore, we need to have code that is really optimized for size, which makes the Linux/glibc toolchain maybe not so appropriate.

So I think in principle one could use the Linux toolchain but due to the above requirements I think I will go for the above-described solution.

Thanks again for your help!

TommyMurphyTM1234, 2023-01-25:
One other thing - presumably you are targeting a multi hart/CPU environment that actually does need atomics? Just that, if not, then they are presumably redundant?

Plociennik, 2023-01-26:
Thanks for the hint! However, I will be running the code on a cluster with several RISCV32 cores that need to be synchronized at certain points, so I actually do need the `a` instructions.

TommyMurphyTM1234, 2023-01-26:
Fair enough. Just asking in case... :⁠-⁠)

hyperion009, 2024-08-26:
@Plociennik  Did you have this done successfully? I want to know how many extra efforts I need put in. thanks in advance.

Plociennik, 2024-08-26:
No, I was not able to build libatomic in the Newlib build.