riscv-collab/riscv-gnu-toolchain #1184: compile gnu-linux-gcc error with glibc configure failed

gf91597, 2023-02-03:
cd build-glibc-linux-rv64imafdc-lp64d && \
        CC="riscv64-unknown-linux-gnu-gcc " \
        CXX="this-is-not-the-compiler-youre-looking-for" \
        CFLAGS="  -mcmodel=medlow -g -O2 " \
        CXXFLAGS="  -mcmodel=medlow -g -O2 " \
        ASFLAGS=" -mcmodel=medlow " \
        /home/gaofeng2/mySrc/RISCV/riscv-gnu-toolchain1/build/../glibc/configure \
        --host=riscv64-unknown-linux-gnu \
        --prefix=/usr \
        --disable-werror \
        --enable-shared \
        --enable-obsolete-rpc \
        --with-headers=/home/gaofeng2/mySrc/RISCV/riscv-gnu-toolchain1/build/../linux-headers/include \
        --disable-multilib \
        --enable-kernel=3.0.0 \
         \
        --libdir=/usr/lib libc_cv_slibdir=/lib libc_cv_rtlddir=/lib
cd build-glibc-linux-headers && CC="riscv64-unknown-linux-gnu-gcc" /home/gaofeng2/mySrc/RISCV/riscv-gnu-toolchain1/build/../glibc/configure \
        --host=riscv64-unknown-linux-gnu \
        --prefix=/home/gaofeng2/riscv-gnu/sysroot/usr \
        --enable-shared \
        --with-headers=/home/gaofeng2/mySrc/RISCV/riscv-gnu-toolchain1/build/../linux-headers/include \
        --disable-multilib \
        --enable-kernel=3.0.0
/bin/sh: line 1: /home/gaofeng2/mySrc/RISCV/riscv-gnu-toolchain1/build/../glibc/configure: No such file or directory
Makefile:355: recipe for target 'stamps/build-glibc-linux-rv64imafdc-lp64d' failed
/bin/sh: /home/gaofeng2/mySrc/RISCV/riscv-gnu-toolchain1/build/../glibc/configure: No such file or directory
make: *** [stamps/build-glibc-linux-rv64imafdc-lp64d] Error 127
make: *** Waiting for unfinished jobs....
Makefile:338: recipe for target 'stamps/build-glibc-linux-headers' failed
make: *** [stamps/build-glibc-linux-headers] Error 127

TommyMurphyTM1234, 2023-02-03:
Please post the full steps required to reproduce your issue and also attach (not embed) your full `git clone`, `configure` and `make`  logs.

gf91597, 2023-02-03:
The gcc gdb and other codes  I used the  gitee.mirror.  
My steps:
git clone --recursive https://github.com/riscv-collab/riscv-gnu-toolchain.git
git clone --recursive https://gitee.com/mirrors/riscv-gcc.git dejagnu
 git clone --recursive https://gitee.com/mirrors/riscv-gcc.git gcc
git clone --recursive https://gitee.com/mirrors/riscv-glibc.git glibc
git clone --recursive https://gitee.com/mirrors/riscv-newlib.git newlib
git clone --recursive -b fsf-gdb-10.1-with-sim https://gitee.com/mirrors/riscv-binutils-gdb.git gdb
git clone --recursive -b riscv-binutils-2.38 https://gitee.com/mirrors/riscv-binutils-gdb.git binutils

cd riscv-gnu-toolchain
mkdir build
cd build
../configure --prefix=/home/gaofeng2/riscv-gnu/
make linux -j8

But  I compiled the riscv64-unknown-elf is OK.
And my steps:
cd riscv-gnu-toolchain
mkdir build
cd build
../configure --prefix=/home/gaofeng2/riscv-tool/ --enable-multilib
make 
And riscv64-unknown-elf is OK.

TommyMurphyTM1234, 2023-02-03:
Sorry, you'll need to be clearer on the steps.
The additional `git clone` commands after the first one above don't seem to have any relevance once you `cd riscv-gnu-toolchain` and work in that directory.
Why are they there?
And what do you mean by specific things being "ok" thereafter?

Please clarify the precise steps taken to reproduce the problem and attach the logs that I asked for previously.

Without the necessary information to reproduce a problem nobody can really investigate it.

TommyMurphyTM1234, 2023-02-04:
Any update @gf91597?

TommyMurphyTM1234, 2023-02-05:
> The gcc gdb and other codes I used the gitee.mirror. My steps: git clone --recursive https://github.com/riscv-collab/riscv-gnu-toolchain.git git clone --recursive [...]

I built both toolchains and they both built fine.
I even did it after omitting all but the first `git clone` and both builds worked fine.
As I said before I don't know why you're cloning all the other stuff when it's not used.
Anyway I cannot reproduce any problem here so am closing the issue.
If you still have problems then please reopen the issue but make sure to post full detailed instructions on how to reproduce your issue.

closed 2023-02-05 by TommyMurphyTM1234