riscv-collab/riscv-gnu-toolchain #1186: riscv32-unknown-elf-run exit issue

bharishb, 2023-02-07:
Hi,

I am trying to compile a simple code with modified linker script(for memory map, start point), crt0 to different locations.  ELF run is not able to identify the exit point.
Compile command : riscv32-unknown-elf-gcc -g -ffreestanding -O0 -Wl,--gc-sections -nostartfiles -nostdlib -nodefaultlibs -Wl,-T,../riscv32.ld ../crt0.s add.c -o wiscv.o
ELF Run  command : riscv32-unknown-elf-run -t --memory-region 0x80000000,0x8000000 wiscv.o
Error : Trace went properly but it is not able to exit out of the code.
insn:     0x80000000                                       -auipc gp, 8192;  // gp = pc + 8192
insn:     0x80000004                                       -addi gp, gp, 0xfffff8b8;  // gp = gp + 0xfffff8b8
insn:     0x80000008                                       -auipc sp, 134217728;  // sp = pc + 134217728
insn:     0x8000000c                                       -addi sp, sp, 0xfffffff8;  // sp = sp + 0xfffffff8
insn:     0x80000010                                       -add s0, sp, zero;  // s0 = sp + zero
insn:     0x80000014                                       -jal ra, 8;
insn:     0x8000001c #2    main                            -addi sp, sp, 0xffffffe0;  // sp = sp + 0xffffffe0
insn:     0x80000020 #2    main                            -sw s0, 28(sp);
insn:     0x80000024 #2    main                            -addi s0, sp, 0x20;  // s0 = sp + 0x20
insn:     0x80000028 #4    main                            -addi a5, zero, 0x5;  // a5 = zero + 0x5
insn:     0x8000002c #4    main                            -sw a5, -20(s0);
insn:     0x80000030 #4    main                            -addi a5, zero, 0xa;  // a5 = zero + 0xa
insn:     0x80000034 #4    main                            -sw a5, -24(s0);
insn:     0x80000038 #7    main                            -lw a4, -20(s0);
insn:     0x8000003c #7    main                            -lw a5, -24(s0);
insn:     0x80000040 #7    main                            -add a5, a4, a5;  // a5 = a4 + a5
insn:     0x80000044 #7    main                            -sw a5, -28(s0);
insn:     0x80000048 #9    main                            -addi zero, zero, 0;  // zero = zero + 0
insn:     0x8000004c #9    main                            -lw s0, 28(sp);
insn:     0x80000050 #9    main                            -addi sp, sp, 0x20;  // sp = sp + 0x20
insn:     0x80000054 #9    main                            -jalr zero, ra, 0;
insn:     0x80000018                                       -jal zero, 64;
insn:     0x80000058                                       -ecall;
**sim: bad insn len 0x2 @ 0x8000005c: 0x1
program stopped with signal 4 (Illegal instruction).**

// CRT0
.section .init, "ax"
  2 .global _start
  3 _start:
  4     .cfi_startproc
  5     .cfi_undefined ra
  6     .option push
  7     .option norelax
  8     la gp, __global_pointer$
  9     .option pop
 10     la sp, __stack_top
 11     add s0, sp, zero
 12     jal ra, main
 13     jal zero, _exit
 14     .cfi_endproc
 15 
 16 .section .fini, "ax"
 17 .global _exit
 18 _exit:
 19     .cfi_startproc
 20     ecall
 21     .dword 1
 22     .cfi_endproc
 23 .end

//add.c
void main()
   {
   int a=5, b=10;
   int c;
   c = a + b;}

TommyMurphyTM1234, 2023-02-08:
Sounds like the same issue as here...

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1176

... namely that the `riscv<XLEN>-unknown-elf-run` simulator doesn't support the C (compressed) extension and the compressed instructions that it provides. I presume that your toolchain is configured for `rv32gc/ilp32d` by default? You should probably consider using a different simulator if you need compressed instruction support.

TommyMurphyTM1234, 2023-02-08:
BTW `-ffreestanding` doesn't really make sense when using the bare metal rather than the Linux toolchain.

bharishb, 2023-02-08:
Hi,

My issue is not with instructions.  elf-run has given the trace correctly  till the instruction. It didnt exit out after that. It assumed some instruction after the last instruction and tried to execute it.

 **(OBJDUMP)** riscv32-unknown-elf-objdump -d wiscv.o
Disassembly of section .fini:

80000058 <_exit>:
80000058:       00000073                ecall
(**ELF RUN**)[batchu@snares-10] (398)$ riscv32-unknown-elf-run -t --memory-region 0x80000000,0x8000000 wiscv.o 
insn:     0x80000054 #9    main                            -jalr zero, ra, 0;
insn:     0x80000018                                       -jal zero, 64;
insn:     0x80000058                                       -ecall;
sim: bad insn len 0x2 @ 0x8000005c: 0x14
program stopped with signal 4 (Illegal instruction).

How do I make elf run exit the program ? Thanks

bharishb, 2023-02-08:
Hi, I have found the solution. I wasnt passing the correct ABI register values at the exit section. 
The idea of my experiment is to find a way to compile raw assembly riscv code and verify the execution of elf-run trace. 
I am using elf-gcc to compile and for the object file, which I am giving input to elf-run. 

Let me know if there is any other way. Thanks for helping out.

TommyMurphyTM1234, 2023-02-08:
I don't really understand your previous post, but if you've found a solution to your problem then I guess that this issue can be closed.

closed 2023-02-08 by TommyMurphyTM1234