riscv-collab/riscv-gnu-toolchain #1190: Compiling P-ext instructions

infini8-13, 2023-02-13:
How do I compile P-extension instructions(v0.9.11) using the toolchain? I get the error 

`Error: -march=rv64imafd_zpn: unknown prefixed ISA extension 'zpn' 
psimd.S:18: Error: unrecognized opcode 'radd16 t3,t2,t4' `

when I run

`riscv64-unknown-elf-gcc -march=rv64imafd_zpn psimd.S -o psimd.o`

How do I proceed further? riscv-opcodes has zpn in unratified. How do I build the toolchain so that it supports Zpn, Zbpbo, Zpsfoperand?

pz9115, 2023-02-13:
Since the P-ext is not ratified and plan to refactor, currently we only have downstream supports, you can checkou the gcc&binutils repo into:

gcc: https://github.com/plctlab/riscv-gcc/tree/riscv-gcc-p-ext

binutils: https://github.com/plctlab/riscv-binutils-gdb/tree/riscv-binutils-p-ext

then rebuild the toolchain. Here only support intrinsic to call instructions.

infini8-13, 2023-02-14:
Great, thanks! I will do as suggested and update you.

infini8-13, 2023-02-17:
I changed the submodule link in .gitmodules, along with appropriate branches of the repositories given by @pz9115 . I ran the following commands:
`git submodule sync`
`git submodule update --init --recursive --remote`
`./configure --prefix=/opt/riscv`
`make linux`
The build was successful, but the error is still the same. 
Please let me know what the ideal configure options might be before the build, and what commands I should run to compile. 

The assembly code:

_start:
  andi t0, t0, 0 # Clear register t0
  andi t1, t1, 0 # Clear register t1
  andi t2, t2, 0 # Clear register t2
  andi t3, t3, 0 # Clear register t3
  andi t4, t4, 0 # Clear register t3
  li t0, 394 # Load register t0 
  li t1, 19 # Load register t1 
#Muldiv
  mul t2, t0, t1 
#FP single and Double precision
  fcvt.s.w     ft0, t2
  fcvt.d.w     ft1, t1
  fdiv.d  ft2, ft1, ft0 
  fadd.s  ft3, ft0, ft2
  fcvt.w.s     t4, ft3
#Psimd
  radd16 t3, t2, t4

TommyMurphyTM1234, 2023-02-17:
When I try to build the toolchain as follows:
```
cd ~/Downloads
git clone --recursive https://github.com/riscv-collab/riscv-gnu-toolchain
git clone --recursive https://github.com/plctlab/riscv-binutils-gdb -b riscv-gcc-p-ext
git clone --recursive https://github.com/plctlab/riscv-binutils-gdb -b riscv-binutils-p-ext
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools --with-arch=rv64imafd_zpn --with-abi=lp64d --with-gcc-src=$HOME/Downloads/riscv-gcc --with-binutils-src=$HOME/Downloads/riscv-gnu-toolchain
make
````
It gets this far and just seems to stall...
```
make -C build-binutils-newlib
make[1]: Entering directory '/home/tommy/Downloads/riscv-gnu-toolchain/build-binutils-newlib'
mkdir -p /home/tommy/Downloads/riscv-gnu-toolchain/installed-tools/.test || \
	(echo "Sorry, you don't have permission to write to" \
	 "'/home/tommy/Downloads/riscv-gnu-toolchain/installed-tools'. Please make sure that the location is " \
	 "writable or use --prefix to specify another destination.'" \
	 && exit 1)
rm -r /home/tommy/Downloads/riscv-gnu-toolchain/installed-tools/.test
mkdir -p stamps/ && touch stamps/check-write-permission
rm -rf stamps/build-binutils-newlib build-binutils-newlib
mkdir build-binutils-newlib
cd build-binutils-newlib && CC_FOR_TARGET=riscv64-unknown-elf-gcc /home/tommy/Downloads/riscv-gnu-toolchain/binutils/configure \
	--target=riscv64-unknown-elf \
	 \
	--prefix=/home/tommy/Downloads/riscv-gnu-toolchain/installed-tools \
	 \
	--disable-werror \
	--with-expat=yes  \
	--disable-gdb \
	--disable-sim \
	--disable-libdecnumber \
	--disable-readline \
	--with-isa-spec=2.2
```

TommyMurphyTM1234, 2023-02-17:
And when I used this modified `.gitmodules.pext` file:
```
[submodule "binutils"]
	path = binutils
	url = https://github.com/plctlab/riscv-binutils-gdb
	branch = riscv-binutils-p-ext
[submodule "gcc"]
	path = gcc
	url = https://github.com/plctlab/riscv-gcc
	branch = riscv-gcc-p-ext
[submodule "glibc"]
	path = glibc
	url = https://sourceware.org/git/glibc.git
[submodule "dejagnu"]
	path = dejagnu
	url = https://git.savannah.gnu.org/git/dejagnu.git
	branch = dejagnu-1.6.3
[submodule "newlib"]
	path = newlib
	url = https://sourceware.org/git/newlib-cygwin.git
	branch = master
[submodule "gdb"]
	path = gdb
	url = https://sourceware.org/git/binutils-gdb.git
	branch = gdb-12-branch
[submodule "qemu"]
	path = qemu
	url = https://gitlab.com/qemu-project/qemu.git
[submodule "musl"]
	path = musl
	url = https://git.musl-libc.org/git/musl
	branch = master
[submodule "spike"]
	path = spike
	url = https://github.com/riscv-software-src/riscv-isa-sim.git
	branch = master
[submodule "pk"]
	path = pk
	url = https://github.com/riscv-software-src/riscv-pk.git
	branch = master
```
which is copied here:
```
git clone  https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
cp ../.gitmodules.pext .gitmodules
./configure --prefix=`pwd`/installed-tools --with-arch=rv64imafd_zpn --with-abi=lp64d
```
and tried to build, I got this
```
make 2>&1 | tee build.log
cd /home/tommy/Downloads/riscv-gnu-toolchain && \
flock `git rev-parse --git-dir`/config git submodule init /home/tommy/Downloads/riscv-gnu-toolchain/gcc/ && \
flock `git rev-parse --git-dir`/config git submodule update /home/tommy/Downloads/riscv-gnu-toolchain/gcc/
Submodule 'gcc' (https://github.com/plctlab/riscv-gcc) registered for path 'gcc'
Cloning into '/home/tommy/Downloads/riscv-gnu-toolchain/gcc'...
From https://github.com/plctlab/riscv-gcc
 * branch                    2ee5e4300186a92ad73f1a1a64cb918dc76c8d67 -> FETCH_HEAD
Submodule path 'gcc': checked out '2ee5e4300186a92ad73f1a1a64cb918dc76c8d67'
cd /home/tommy/Downloads/riscv-gnu-toolchain && \
flock `git rev-parse --git-dir`/config git submodule init /home/tommy/Downloads/riscv-gnu-toolchain/newlib/ && \
flock `git rev-parse --git-dir`/config git submodule update /home/tommy/Downloads/riscv-gnu-toolchain/newlib/
Submodule 'newlib' (https://sourceware.org/git/newlib-cygwin.git) registered for path 'newlib'
Cloning into '/home/tommy/Downloads/riscv-gnu-toolchain/newlib'...
Submodule path 'newlib': checked out '415fdd4279b85eeec9d54775ce13c5c412451e08'
cd /home/tommy/Downloads/riscv-gnu-toolchain && \
flock `git rev-parse --git-dir`/config git submodule init /home/tommy/Downloads/riscv-gnu-toolchain/binutils/ && \
flock `git rev-parse --git-dir`/config git submodule update /home/tommy/Downloads/riscv-gnu-toolchain/binutils/
Submodule 'binutils' (https://github.com/plctlab/riscv-binutils-gdb) registered for path 'binutils'
Cloning into '/home/tommy/Downloads/riscv-gnu-toolchain/binutils'...
fatal: remote error: upload-pack: not our ref b51c2fec1da205ea3e7354cbb3e253018d64873c
fatal: the remote end hung up unexpectedly
Fetched in submodule path 'binutils', but it did not contain b51c2fec1da205ea3e7354cbb3e253018d64873c. Direct fetching of that commit failed.
make: *** [Makefile:282: /home/tommy/Downloads/riscv-gnu-toolchain/binutils/.git] Error 1
```

TommyMurphyTM1234, 2023-02-17:
> I changed the submodule link in .gitmodules, along with appropriate branches of the repositories given by @pz9115 . I ran the following commands: `git submodule sync` `git submodule update --init --recursive --remote` `./configure --prefix=/opt/riscv` `make linux` The build was successful

I tried this too.
I get various errors:
```
git submodule update --init --recursive --remote
Submodule 'binutils' (https://github.com/plctlab/riscv-binutils-gdb) registered for path 'binutils'
Submodule 'dejagnu' (https://git.savannah.gnu.org/git/dejagnu.git) registered for path 'dejagnu'
Submodule 'gcc' (https://github.com/plctlab/riscv-gcc) registered for path 'gcc'
Submodule 'gdb' (https://sourceware.org/git/binutils-gdb.git) registered for path 'gdb'
Submodule 'glibc' (https://sourceware.org/git/glibc.git) registered for path 'glibc'
Submodule 'musl' (https://git.musl-libc.org/git/musl) registered for path 'musl'
Submodule 'newlib' (https://sourceware.org/git/newlib-cygwin.git) registered for path 'newlib'
Submodule 'pk' (https://github.com/riscv-software-src/riscv-pk.git) registered for path 'pk'
Submodule 'qemu' (https://gitlab.com/qemu-project/qemu.git) registered for path 'qemu'
Submodule 'spike' (https://github.com/riscv-software-src/riscv-isa-sim.git) registered for path 'spike'
Cloning into '/home/tommy/Downloads/riscv-gnu-toolchain/binutils'...
Cloning into '/home/tommy/Downloads/riscv-gnu-toolchain/dejagnu'...
Cloning into '/home/tommy/Downloads/riscv-gnu-toolchain/gcc'...
Cloning into '/home/tommy/Downloads/riscv-gnu-toolchain/gdb'...
Cloning into '/home/tommy/Downloads/riscv-gnu-toolchain/glibc'...
Cloning into '/home/tommy/Downloads/riscv-gnu-toolchain/musl'...
Cloning into '/home/tommy/Downloads/riscv-gnu-toolchain/newlib'...
Cloning into '/home/tommy/Downloads/riscv-gnu-toolchain/pk'...
Cloning into '/home/tommy/Downloads/riscv-gnu-toolchain/qemu'...
Cloning into '/home/tommy/Downloads/riscv-gnu-toolchain/spike'...
Submodule path 'binutils': checked out '9e69931fc403608176cc056b541b567e09271194'
Submodule path 'dejagnu': checked out 'd51e3f9f8188a735a64e7b2348e8fb7084f1d666'
Submodule path 'gcc': checked out '7eeadf67b7b46ef0c814e4769d5dd301ad8401b8'
Submodule path 'gdb': checked out 'a4418a9c6f99fd31c51698b1f6a6f8dbc1b81b6f'
remote: Enumerating objects: 11, done.
remote: Counting objects: 100% (11/11), done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 6 (delta 5), reused 0 (delta 0), pack-reused 0
Unpacking objects: 100% (6/6), 892 bytes | 63.00 KiB/s, done.
From https://sourceware.org/git/glibc
   26c1769509..609054152f  master     -> origin/master
Submodule path 'glibc': checked out '609054152fd77c9b572bb04c4af2f8da1ed0c86e'
Submodule path 'musl': checked out 'f47a8cdd250d9163fcfb39bf4e9d813957c0b187'
Submodule path 'newlib': checked out 'e4cc9e48462b538253d62109012b90befaaf7bc5'
Submodule path 'pk': checked out '573c858d9071a2216537f71de651a814f76ee76d'
Submodule path 'qemu': checked out '6dffbe36af79e26a4d23f94a9a1c1201de99c261'
Submodule 'dtc' (https://gitlab.com/qemu-project/dtc.git) registered for path 'qemu/dtc'
Submodule 'meson' (https://gitlab.com/qemu-project/meson.git) registered for path 'qemu/meson'
Submodule 'roms/QemuMacDrivers' (https://gitlab.com/qemu-project/QemuMacDrivers.git) registered for path 'qemu/roms/QemuMacDrivers'
Submodule 'roms/SLOF' (https://gitlab.com/qemu-project/SLOF.git) registered for path 'qemu/roms/SLOF'
Submodule 'roms/edk2' (https://gitlab.com/qemu-project/edk2.git) registered for path 'qemu/roms/edk2'
Submodule 'roms/ipxe' (https://gitlab.com/qemu-project/ipxe.git) registered for path 'qemu/roms/ipxe'
Submodule 'roms/openbios' (https://gitlab.com/qemu-project/openbios.git) registered for path 'qemu/roms/openbios'
Submodule 'roms/opensbi' (https://gitlab.com/qemu-project/opensbi.git) registered for path 'qemu/roms/opensbi'
Submodule 'roms/qboot' (https://gitlab.com/qemu-project/qboot.git) registered for path 'qemu/roms/qboot'
Submodule 'roms/qemu-palcode' (https://gitlab.com/qemu-project/qemu-palcode.git) registered for path 'qemu/roms/qemu-palcode'
Submodule 'roms/seabios' (https://gitlab.com/qemu-project/seabios.git/) registered for path 'qemu/roms/seabios'
Submodule 'roms/seabios-hppa' (https://gitlab.com/qemu-project/seabios-hppa.git) registered for path 'qemu/roms/seabios-hppa'
Submodule 'roms/skiboot' (https://gitlab.com/qemu-project/skiboot.git) registered for path 'qemu/roms/skiboot'
Submodule 'roms/u-boot' (https://gitlab.com/qemu-project/u-boot.git) registered for path 'qemu/roms/u-boot'
Submodule 'roms/u-boot-sam460ex' (https://gitlab.com/qemu-project/u-boot-sam460ex.git) registered for path 'qemu/roms/u-boot-sam460ex'
Submodule 'roms/vbootrom' (https://gitlab.com/qemu-project/vbootrom.git) registered for path 'qemu/roms/vbootrom'
Submodule 'subprojects/libvfio-user' (https://gitlab.com/qemu-project/libvfio-user.git) registered for path 'qemu/subprojects/libvfio-user'
Submodule 'tests/fp/berkeley-softfloat-3' (https://gitlab.com/qemu-project/berkeley-softfloat-3.git) registered for path 'qemu/tests/fp/berkeley-softfloat-3'
Submodule 'tests/fp/berkeley-testfloat-3' (https://gitlab.com/qemu-project/berkeley-testfloat-3.git) registered for path 'qemu/tests/fp/berkeley-testfloat-3'
Submodule 'tests/lcitool/libvirt-ci' (https://gitlab.com/libvirt/libvirt-ci.git) registered for path 'qemu/tests/lcitool/libvirt-ci'
[... cut ...]
Submodule path 'qemu/tests/fp/berkeley-softfloat-3': checked out 'b64af41c3276f97f0e181920400ee056b9c88037'
Submodule path 'qemu/tests/fp/berkeley-testfloat-3': checked out '5a59dcec19327396a011a17fd924aed4fec416b3'
Submodule path 'qemu/tests/lcitool/libvirt-ci': checked out '1c3e16cae38407d0782dc94080d1104106456fa4'
Submodule path 'qemu/ui/keycodemapdb': checked out 'c2c09b3f4c1d7688711e6501338ac88017d4341c'
Failed to recurse into submodule path 'qemu/roms/edk2'
Submodule path 'spike': checked out '6077cf88807029e2bbe3aa4c935c7e72c3d8d7ce'
Failed to recurse into submodule path 'qemu'
```

TommyMurphyTM1234, 2023-02-17:
In spite of the previous errors that I mentioned, the toolchain build succeeds.
When put your example code:
```
_start:
andi t0, t0, 0 # Clear register t0
andi t1, t1, 0 # Clear register t1
andi t2, t2, 0 # Clear register t2
andi t3, t3, 0 # Clear register t3
andi t4, t4, 0 # Clear register t3
li t0, 394 # Load register t0
li t1, 19 # Load register t1
#Muldiv
mul t2, t0, t1
#FP single and Double precision
fcvt.s.w ft0, t2
fcvt.d.w ft1, t1
fdiv.d ft2, ft1, ft0
fadd.s ft3, ft0, ft2
fcvt.w.s t4, ft3
#Psimd
radd16 t3, t2, t4
```
into the file `test.S` and then compile it to object file, it works fine and then disassembles back OK:
```
riscv64-unknown-linux-gnu-gcc -march=rv64imafd_zpn -c test.S
riscv64-unknown-linux-gnu-objdump -d test.o

test.o:     file format elf64-littleriscv

Disassembly of section .text:

0000000000000000 <start>:
   0:	0002f293          	andi	t0,t0,0
   4:	00037313          	andi	t1,t1,0
   8:	0003f393          	andi	t2,t2,0
   c:	000e7e13          	andi	t3,t3,0
  10:	000efe93          	andi	t4,t4,0
  14:	18a00293          	li	t0,394
  18:	01300313          	li	t1,19
  1c:	026283b3          	mul	t2,t0,t1
  20:	d003f053          	fcvt.s.w	ft0,t2
  24:	d20300d3          	fcvt.d.w	ft1,t1
  28:	1a00f153          	fdiv.d	ft2,ft1,ft0
  2c:	002071d3          	fadd.s	ft3,ft0,ft2
  30:	c001fed3          	fcvt.w.s	t4,ft3
  34:	01d38e77          	radd16	t3,t2,t4
```
Did you definitely pass `-march=rv64imafd_zpn`?

infini8-13, 2023-02-18:
Hi @TommyMurphyTM1234, that's wonderful! Yes I did pass `-march=rv64imafd_zpn`, but I had configured it wrong. I'll build now and update here about the same.

TommyMurphyTM1234, 2023-02-18:
Ok, I'll close this issue now so.

closed 2023-02-18 by TommyMurphyTM1234

infini8-13, 2023-02-18:
Hi @TommyMurphyTM1234  and  @pz9115, it works! Thanks for your brilliant help.