riscv-collab/riscv-gnu-toolchain #1199: ELF RUN is giving empty trace for slt, slti, sltu, sltiu instructions

bharishb, 2023-02-28:
**// Assembly code to be compiled**
.global _start
_start:
      addi x1,x0,0x0101
      addi x3,x0,0xfffff800
      **slt x2,x1,x3**
      addi x4,x0,0xfffff801
      addi x6,x0,0x0303
      slt x5,x4,x6
      j exit

exit: 
    li a7,93
    ecall

**//ELF RUN TRACE**

insn:     0x000000 ---   _start                            -addi ra, zero, 0x101;  // ra = zero + 0x101
insn:     0x000004 ---   _start                            -addi gp, zero, 0xfffff800;  // gp = zero + 0xfffff800
**insn:     0x000008 ---   _start                            -slt**
insn:     0x00000c ---   _start                            -addi tp, zero, 0xfffff801;  // tp = zero + 0xfffff801
insn:     0x000010 ---   _start                            -addi t1, zero, 0x303;  // t1 = zero + 0x303
**insn:     0x000014 ---   _start                            -slt**
insn:     0x000018 ---   _start                            -jal zero, 4;
insn:     0x00001c ---   exit                              -addi a7, zero, 0x5d;  // a7 = zero + 0x5d
insn:     0x000020 ---   exit                              -ecall;

TommyMurphyTM1234, 2023-02-28:
Please provide the information also requested here:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1200#issuecomment-1447719876

TommyMurphyTM1234, 2023-02-28:
> If you feel that this is erroneous then you should log an issue in the upstream GDB bug tracker:
> 
> * https://www.sourceware.org/gdb/bugs/

closed 2023-02-28 by TommyMurphyTM1234