riscv-collab/riscv-gnu-toolchain #1200: ORI instruction ELF RUN exiting as unhandled exception when tried to write at x0

bharishb, 2023-02-28:
ori x0, x17, 1 # icount 40

//ELF RUN
insn:     0x0000a0 ---   _start                            -UNHANDLED EXTENSION: 35

I want to the limitations of ELF RUN simulator. Thanks

TommyMurphyTM1234, 2023-02-28:
Please provide full details on how to reproduce this including details of the toolchain that you are using, where you obtained it from, how you are compiling your code, the code that you are compiling etc.

bharishb, 2023-02-28:
Hi, I am using the Base Instruction set rv32I. I use the following command to generate my trace. For 32I, I dont think we need to give arch/abi since they are the default.

    riscv32-unknown-elf-run -t output.o 2> output.golden_run

//OBJDUMP
[objdump.txt](https://github.com/riscv-collab/riscv-gnu-toolchain/files/10851101/objdump.txt)

TommyMurphyTM1234, 2023-02-28:
See the link that I posted.
The trace output that you see is correct.
If you think that more info should be dumped then you'll need to log an issue against GDB upstream.

TommyMurphyTM1234, 2023-02-28:
You haven't said where you got the toolchain from, how it was built etc.

TommyMurphyTM1234, 2023-03-02:
> Hi, I am using the Base Instruction set rv32I. I use the following command to generate my trace. For 32I, I dont think we need to give arch/abi since they are the default.
>
> ``` [...]

Where did you get or how **exactly**  did you build the toolchain?
What is your source code?
How are you compiling/linking it?
You need to provide full details a reproducible test case if you want assistance.
Otherwise the issue will be closed as invalid.

bharishb, 2023-03-02:
Hi,
I think this issue should be tool chain build independent atleast any 32i build should give it. I am trying to write into hardwired register x0 with the help of ORI instruction.  I have literally nothing in my code 
 FAIL at highlighted instructiion by gdb simulator riscv32-unknown-elf-run by throwing unhandled exception.
When I replace x0 destination with x1. it passes. So do the toolchain expects to throw error or it silently ignores ? When I replace 0x101 with 0x2, it passes. **So handling X0 write is not consistent ??**
.global _start
 _start:
        **ori x0,x0,0x0101**
        j exit
  
 exit:
     li a7,93
      ecall

**PASS output** of elf-run
insn:     0x000000 ---   _start                            -ori ra, zero, 257;  // ra = zero | 0x101
  2 insn:     0x000004 ---   _start                            -jal zero, 4;
  3 insn:     0x000008 ---   exit                              -addi a7, zero, 0x5d;  // a7 = zero + 0x5d
  4 insn:     0x00000c ---   exit                              -ecall;

**FAIL output** 
 insn:     0x000000 ---   _start                            -UNHANDLED EXTENSION: 35

  2 program stopped with signal 4 (Illegal instruction).
**PASS OUTPUT when ored with 0x2**
 1 insn:     0x000000 ---   _start                            -ori zero, zero, 2;  // zero = zero | 0x2
  2 insn:     0x000004 ---   _start                            -jal zero, 4;
  3 insn:     0x000008 ---   exit                              -addi a7, zero, 0x5d;  // a7 = zero + 0x5d
  4 insn:     0x00000c ---   exit                              -ecall;

**Git Clone Command :**
git clone https://github.com/riscv/riscv-gnu-toolchain --recursive
 **Configure Command : .**
 /configure --prefix=/home/batchu/riscv_tool_chain_git --with-arch=rv32i --with-abi=ilp32
 **Build Command ::**
make
**Compile :** 
riscv32-unknown-elf-gcc  -nostartfiles -Wl,-T,./riscv32.ld $1.s -o output.o
**Simulate**
riscv32-unknown-elf-run -t output.o 2> output.golden_run
[riscv32.ld.txt](https://github.com/riscv-collab/riscv-gnu-toolchain/files/10871829/riscv32.ld.txt)

bharishb, 2023-03-02:
ohh Thanks..  I will start to migration to other simulators then ,.. Thanks for the help :)

TommyMurphyTM1234, 2023-05-27:
Logged upstream now 

* https://sourceware.org/bugzilla/show_bug.cgi?id=30494

so I am closing this issue.

closed 2023-05-27 by TommyMurphyTM1234

TommyMurphyTM1234, 2023-05-27:
> Logged upstream now
> 
> * https://sourceware.org/bugzilla/show_bug.cgi?id=30494
> 
> so I am closing this issue.

Somebody else pointed out that it should probably be logged against binutils/libopcodes and not gdb/sim so I logged it here instead:

* https://sourceware.org/bugzilla/show_bug.cgi?id=30495

Temperature-block, 2024-04-28:
I have updated patch on bugzilla https://sourceware.org/bugzilla/show_bug.cgi?id=30495