riscv-collab/riscv-gnu-toolchain #1201: multilib toolchain dejagnu testing broken on tip 2023-02-24 f0b00941ec64

vineetgarc, 2023-02-28:
Running dejagnu tests on tip of toolchain repo is failing

1. checkout today's latest

2023-02-24 f0b00941ec64 Merge pull request #1182 from xmzzz/update-readme-os-x 
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain toolchain-upstream
cd toolchain-upstream
git submodule init
git submodule update
```

3. Build multilib glibc toolchain

```
./configure --with-arch=rv64imafdc_zba_zbb_zbc_zbs --with-abi=lp64d --enable-multilib --enable-linux --prefix=<path-to-install>
```

4. Invoke dejagnu

```
make report-linux SIM=qemu
...
...
rm -rf testsuite/gcc-parallel
make[3]: Entering directory '/scratch/vineetg/gnu/toolchain-upstream/build-gcc-linux-stage2/gcc'
(rootme=`${PWDCMD-pwd}`; export rootme; \
srcdir=`cd /scratch/vineetg/gnu/toolchain-upstream/gcc/gcc; ${PWDCMD-pwd}` ; export srcdir ; \
if [ -n "" ] \
   && [ -n "$GCC_RUNTEST_PARALLELIZE_DIR" ] \
   && [ -f testsuite/gcc-parallel/finished ]; then \
  rm -rf testsuite/gcc; \
else \
  cd testsuite/gcc; \
  rm -f tmp-site.exp; \
  sed '/set tmpdir/ s|testsuite$|testsuite/gcc|' \
	< ../../site.exp > tmp-site.exp; \
  /bin/bash ${srcdir}/../move-if-change tmp-site.exp site.exp; \
  EXPECT=`if [ -f ${rootme}/../expect/expect ] ; then echo ${rootme}/../expect/expect ; else echo expect ; fi` ; export EXPECT ; \
  if [ -f ${rootme}/../expect/expect ] ; then  \
    TCL_LIBRARY=`cd .. ; cd ${srcdir}/../tcl/library ; ${PWDCMD-pwd}` ; \
    export TCL_LIBRARY ; \
  fi ; \
  `if [ -f ${srcdir}/../dejagnu/runtest ] ; then echo ${srcdir}/../dejagnu/runtest ; else echo runtest; fi` --tool gcc --target_board='riscv-sim/-march=rv32imac/-mabi=ilp32/-mcmodel=medlow riscv-sim/-march=rv32imafdc/-mabi=ilp32d/-mcmodel=medlow riscv-sim/-march=rv64imac/-mabi=lp64/-mcmodel=medlow riscv-sim/-march=rv64imafdc/-mabi=lp64d/-mcmodel=medlow'; \
  if [ -n "$GCC_RUNTEST_PARALLELIZE_DIR" ] ; then \
    touch ${rootme}/testsuite/gcc-parallel/finished; \
  fi ; \
fi )

wrong # args: should be "set varName ?newValue?"
    while executing
"set target_list riscv-sim/-march=rv32imac/-mabi=ilp32/-mcmodel=medlow riscv-sim/-march=rv32imafdc/-mabi=ilp32d/-mcmodel=medlow riscv-sim/-march=rv64im..."
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 set $name $value"
    (procedure "::dejagnu::command_line::restore_cmd_vars" line 5)
    invoked from within
"::dejagnu::command_line::restore_cmd_vars"
    (file "/scratch/vineetg/gnu/INSTALL/tc-up-230228-afe6cea44898/share/dejagnu/runtest.exp" line 694)
```

vineetgarc, 2023-02-28:
Issue seems to be happening with 
`--target_board='riscv-sim/-march=rv32imac/-mabi=ilp32/-mcmodel=medlow riscv-sim/-march=rv32imafdc/-mabi=ilp32d/-mcmodel=medlow riscv-sim/-march=rv64imac/-mabi=lp64/-mcmodel=medlow riscv-sim/-march=rv64imafdc/-mabi=lp64d/-mcmodel=medlow';`

vineetgarc, 2023-02-28:
@cmuellner in case you have any insights.

vineetgarc, 2023-03-01:
**8dd29548de6e9a34b8d0ee8e44c4f32a32c90f55 is the first bad commit**

commit 8dd29548de6e9a34b8d0ee8e44c4f32a32c90f55
Author: Christoph Müllner <christoph.muellner@vrull.eu>
Date:   Wed Sep 28 01:34:45 2022 +0200

    dejagnu: Bump version to 1.6.3
    
    dejagnu still defaults to 1.6 from the (already archived)
    riscv-dejagnu repo. Let's bump the version to 1.6.3
    and use the upstream git repo instead.
    
    Signed-off-by: Christoph Müllner <christoph.muellner@vrull.eu>

vineetgarc, 2023-03-01:
Apparently multilib build testing has been broken for last 6 months

vineetgarc, 2023-03-01:
It seems` process_multilib_options` in baseboards/riscv*exp needs some adjustment.

cmuellner, 2023-03-01:
Thanks for reporting this!
I was aware that there are a few multilib issues, but I was not aware that multilib-testing also has an issue.

Are you planning to fix this in DejaGnu upstream or should I do that?

vineetgarc, 2023-03-01:
Hi Christoph, I'm a total dejagnu/tcl noob. I spent a few hours looking around but couldn't figure this out. It would be great if you can fix it. Thx !

TommyMurphyTM1234, 2023-03-01:
Is this another problem caused by the earlier switch to canonicalised architecture strings - e.g. `rv64gc` versus `rv64imafdc_zicsr` or maybe `rv64imafdc_zicsr_zifencei`?

vineetgarc, 2023-03-01:
I doubt. The toolchain/libs build fine. The issue is how dejagnu test harness expects multilib params.

TommyMurphyTM1234, 2023-03-01:
> I doubt. The toolchain/libs build fine. The issue is how dejagnu test harness expects multilib params.

The canonicalisation issue doesn't affect the building, but rather the use of, multilibs. Sounds like this could be another issue caused by that change and the timeline also matches. E.g. some part of the toolchain or testing framework but matching non canonicalised names to canonicalised names or using canonicalised names in the first place. Unfortunately that change caused many problems and confusion.

vineetgarc, 2023-03-01:
> > I doubt. The toolchain/libs build fine. The issue is how dejagnu test harness expects multilib params.
> 
> The canonicalisation issue doesn't affect the building, but rather the use of, multilibs. Sounds like this could be another issue caused by that change and the timeline also matches. E.g. some part of the toolchain or testing framework not matching non-canonicalised names to canonicalised names or not using canonicalised names in the first place. Unfortunately, that change caused many problems and confusion and there are several open issues arising from that change that don't have an obvious/trivial fix.

I presume you are referring to https://github.com/ClangBuiltLinux/linux/issues/1808

This issue seems to happen with switch to upstream dejagnu 1.6.3 (vs. github fork 1.6 here on riscv-collab). We would have to wait for Christoph to confirm if it is same or there's some adaptation missing when transitioning to upstream dejagnu 1.6.3

TommyMurphyTM1234, 2023-03-01:
> > > I doubt. The toolchain/libs build fine. The issue is how dejagnu test harness expects multilib params.
> > 
> > 
> > The canonicalisation issue doesn't affect the building, but rather the use of, multilibs. Sounds like this could be another issue caused by that change and the timeline also matches. E.g. some part of the toolchain or testing framework not matching non-canonicalised names to canonicalised names or not using canonicalised names in the first place. Unfortunately, that change caused many problems and confusion and there are several open issues arising from that change that don't have an obvious/trivial fix.
> 
> I presume you are referring to [ClangBuiltLinux/linux#1808](https://github.com/ClangBuiltLinux/linux/issues/1808)
> 
> This issue seems to happen with switch to upstream dejagnu 1.6.3 (vs. github fork 1.6 here on riscv-collab). We would have to wait for Christoph to confirm if it is same or there's some adaptation missing when transitioning to upstream dejagnu 1.6.3

No - these issues for example:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1126#issuecomment-1250679690
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1038#issuecomment-1310155191

kito-cheng, 2023-03-02:
Maybe we just rollback to older version of dejagnu as workaround?

vineetgarc, 2023-03-02:
I'd let @cmuellner make that call, given on the complexity of actual 1.6. work and amount of time he can spend ?

cmuellner, 2023-04-16:
I had a look at this, and here is an update.

Initially, I tested the following commits of DejaGnu:
* BAD: upstream latest: `865dc67cd4e4d24ac722b4a865bfe4597b765f54`
* BAD: upstream new-riscv: `8c367b5ca4a79e8ab732ef7f237ab0153ce196bb` (when riscv-sim.exp was merged upstream)
* OK : upstream/rvi-archive merge base: `59c31dba8e52bf68ad766d43576c714c929979a6`
That means to get the old (working) behavior, we can use the mainline repo and use this commit.

Next, I bisected upstream to identify the commit that broke the behavior:
`5fafcd43b2d22b2227e62f7278584418c6449824 is the first bad commit`.
So the latest commit that still works is `55765f0cea891c7282e3fc7a7a4bb70a4cbe9349`.

I am still trying to come up with a proper fix.
Meanwhile, here is a hack that should get you going (essentially avoiding the issue but causing a problem under some circumstances that the first bad commit attempts to solve).

```
diff --git a/runtest.exp b/runtest.exp
index ba49e73..03e9ecb 100644
--- a/runtest.exp
+++ b/runtest.exp
@@ -684,7 +684,7 @@ proc load_lib { file } {
 load_file [file join $base_dir $local_init_file]
 
 # Ensure that command line parameters override testsuite init files.
-::dejagnu::command_line::restore_cmd_vars
+#::dejagnu::command_line::restore_cmd_vars
 
 #
 # If objdir didn't get set in $base_dir/$local_init_file, set it to
@@ -700,7 +700,7 @@ if { $objdir eq "." || $objdir eq $srcdir } {
 }
 
 # Ensure that command line parameters override testsuite init files.
-::dejagnu::command_line::restore_cmd_vars
+#::dejagnu::command_line::restore_cmd_vars
 
 #
 # Find the testsuite.
```

This hack is tested with the first bad commit and upstream/master.
You can apply this change in a local branch and use `--with-dejagnu-src=...` to use it.

cmuellner, 2023-04-17:
Proposed fix sent to the DejaGnu mailing list:
https://lists.gnu.org/archive/html/dejagnu/2023-04/msg00000.html

cmuellner, 2023-04-17:
And by the way, this is what a recent upstream/master from Apr 11 shows:
```
               ========= Summary of gcc testsuite =========
                            | # of unexpected case / # of unique unexpected case
                            |          gcc |          g++ |     gfortran |
   rv32imac/  ilp32/ medlow |   94 /     8 |    2 /     2 |    0 /     0 |
 rv32imafdc/ ilp32d/ medlow |   62 /     7 |    2 /     2 |    0 /     0 |
   rv64imac/   lp64/ medlow |    1 /     1 |    0 /     0 |    0 /     0 |
 rv64imafdc/  lp64d/ medlow |    0 /     0 |    0 /     0 |    0 /     0 |
```

vineetgarc, 2023-04-17:
> And by the way, this is what a recent upstream/master from Apr 11 shows:
>
> ```
>                ========= Summary of gcc testsuite =========
>                             | # of unexpected case / # of [...]

What was you exact cmdline to test this ? And I presume this is gcc 12.2
For this test do you have some workaround for #1206 ? Otherwise all rv32 tests seem to fail as described there.

cmuellner, 2023-04-17:
> What was you exact cmdline to test this ? And I presume this is gcc 12.2 For this test do you have some workaround for #1206 ? Otherwise all rv32 tests seem to fail as described there.

Commands are:
```
./configure --prefix=/opt/riscv-mainline --enable-multilib --with-gcc-src=/home/cm/src/gcc/riscv-mainline --with-binutils-src=/home/cm/src/binutils-gdb/riscv-mainline --with-gdb-src=/home/cm/src/binutils-gdb/riscv-mainline --with-glibc-src=/home/cm/src/glibc/riscv-mainline --with-qemu-src=/home/cm/src/qemu/riscv-mainline --with-dejagnu-src=/home/cm/src/dejagnu/riscv-multilib --enable-linux
make -j$(nproc)
make -j$(nproc) build-sim SIM=qemu
make -j$(nproc) report SIM=qemu
```

All repos relatively recent (updated within the last 2 weeks) and the DejaGnu branch has the proposed fix on top.
As mentioned, I did not run releases but upstream/master.
I have already updated the GCC branch, so I can't say the commit for sure, but I guess it was `2d7ad38707`.

I have not done anything besides this (especially not anything to address #1206).

cmuellner, 2023-04-17:
FWIW, here is the result for GCC `6d4ad4cca5` from today:
```
               ========= Summary of gcc testsuite =========
                            | # of unexpected case / # of unique unexpected case
                            |          gcc |          g++ |     gfortran |
   rv32imac/  ilp32/ medlow |   53 /     6 |    2 /     2 |    0 /     0 |
 rv32imafdc/ ilp32d/ medlow |   53 /     6 |    2 /     2 |    0 /     0 |
   rv64imac/   lp64/ medlow |    1 /     1 |    0 /     0 |    0 /     0 |
 rv64imafdc/  lp64d/ medlow |    0 /     0 |    0 /     0 |    0 /     0 |
```

vineetgarc, 2023-04-18:
Tested-by: Vineet Gupta <vineetg@rivosinc.com>

I can confirm that the fix works for me (modulo the rv32 bulk failures for #1206)

Thx Christoph. Do you want to merge this to existing dejagnu submodule, or wait for the different fix which upstream dejagnu folks will work on.

PR #1247 (2023-05-13): dejagnu: Bump dejagnu to include multilib fix
This changes Dejagnu to the most recent version that includes @cmuellner's fixes.

The Dejagnu commit with @cmuellner's fixes has not been tagged for release.

If we only want to use released versions of Dejagnu, I propose switching back to Dejagnu 1.6.3 to resolve #1243 since the rollback did not resolve #1201.

Resolves: #1201 and #1243.
closed 2023-05-13 by kito-cheng