riscv-collab/riscv-gnu-toolchain #1204: riscv32-unknown-elf-gdb: error while loading shared libraries: libpython3.8.so.1.0: cannot open shared object file: No such file or directory

WajahatRiaz, 2023-03-01:
I'm trying to use the **riscv32-unknown-elf-gdb** for debugging purposes. Whenever I enter the following command

`iscv32-unknown-elf-gdb` or `iscv32-unknown-elf-gdb ./a.out` I get the following error:

`riscv32-unknown-elf-gdb: error while loading shared libraries: libpython3.8.so.1.0: cannot open shared object file: No such file or directory`

[image]

Everything else like like gcc, readelf, objdump is working.

TommyMurphyTM1234, 2023-03-01:
Where did you get the toolchain or how did you build it?

WajahatRiaz, 2023-03-01:
> Where did you get the toolchain or how did you build it?
> 

From the repo and built it as mentioned. Everything else works apart from gdb.

TommyMurphyTM1234, 2023-03-01:
I build the tools recently (see here:) and gdb works fine:
```
tommy@kingfisher:~/Downloads/riscv-gnu-toolchain/installed-tools/bin$ ./riscv32-unknown-elf-gdb
GNU gdb (GDB) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=riscv32-unknown-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) quit
tommy@kingfisher:~/Downloads/riscv-gnu-toolchain/installed-tools/bin$ ldd riscv32-unknown-elf-gdb
	linux-vdso.so.1 (0x00007ffeb2d9b000)
	libncursesw.so.6 => /lib/x86_64-linux-gnu/libncursesw.so.6 (0x00007f566034a000)
	libtinfo.so.6 => /lib/x86_64-linux-gnu/libtinfo.so.6 (0x00007f566031a000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f5660314000)
	libpython2.7.so.1.0 => /lib/x86_64-linux-gnu/libpython2.7.so.1.0 (0x00007f565ffa4000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f565ff81000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f565fe32000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f565fe02000)
	libmpfr.so.6 => /lib/x86_64-linux-gnu/libmpfr.so.6 (0x00007f565fd81000)
	libgmp.so.10 => /lib/x86_64-linux-gnu/libgmp.so.10 (0x00007f565fcfd000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f565fb1b000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f565fb00000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f565f90e000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f5660c3e000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f565f8f0000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007f565f8eb000)
```
The only Python development package that I have installed seems to be `python2.7-dev`:
```
tommy@kingfisher:~/Downloads/riscv-gnu-toolchain/installed-tools/bin$ apt list python*-dev
Listing... Done
python-all-dev/focal 2.7.17-2ubuntu4 amd64
python-all-dev/focal 2.7.17-2ubuntu4 i386
python-apt-dev/focal,focal 2.0.0ubuntu0.20.04.8+zorin1 all
python-cairo-dev/focal,focal 1.16.2-2ubuntu2 all
python-dbus-dev/focal,focal 1.2.16-1build1 all
python-gi-dev/focal 3.36.0-1 amd64
python-gi-dev/focal 3.36.0-1 i386
python-gobject-2-dev/focal,focal 2.28.6-14ubuntu1 all
python-greenlet-dev/focal 0.4.15-4.1 amd64
python-sip-dev/focal 4.19.21+dfsg-1build1 amd64
python-sip-dev/focal 4.19.21+dfsg-1build1 i386
python-wxgtk3.0-dev/focal,focal 3.0.2.0+dfsg-8build1 all
python2-dev/focal,now 2.7.17-2ubuntu4 amd64 [installed,automatic]
python2-dev/focal 2.7.17-2ubuntu4 i386
python2.7-dev/focal-updates,focal-security,now 2.7.18-1~20.04.3 amd64 [installed,automatic]
python2.7-dev/focal-updates,focal-security 2.7.18-1~20.04.3 i386
python3-all-dev/focal 3.8.2-0ubuntu2 amd64
python3-all-dev/focal 3.8.2-0ubuntu2 i386
python3-cairo-dev/focal,focal 1.16.2-2ubuntu2 all
python3-cxx-dev/focal,focal 7.1.3-0.1 all
python3-dev/focal 3.8.2-0ubuntu2 amd64
python3-dev/focal 3.8.2-0ubuntu2 i386
python3-ldb-dev/focal-updates,focal-security 2:2.2.3-0ubuntu0.20.04.3 amd64
python3-ldb-dev/focal-updates,focal-security 2:2.2.3-0ubuntu0.20.04.3 i386
python3-sip-dev/focal 4.19.21+dfsg-1build1 amd64
python3-sip-dev/focal 4.19.21+dfsg-1build1 i386
python3-talloc-dev/focal-updates,focal-security 2.3.1-0ubuntu0.20.04.1 amd64
python3-talloc-dev/focal-updates,focal-security 2.3.1-0ubuntu0.20.04.1 i386
python3.8-dev/focal-updates,focal-security 3.8.10-0ubuntu1~20.04.6 amd64
python3.8-dev/focal-updates,focal-security 3.8.10-0ubuntu1~20.04.6 i386
python3.9-dev/focal-updates,focal-security 3.9.5-3ubuntu0~20.04.1 amd64
python3.9-dev/focal-updates,focal-security 3.9.5-3ubuntu0~20.04.1 i386
```

TommyMurphyTM1234, 2023-03-02:
I installed the following:
```
apt install python3.9 python3.9-dev python-is-python3
```
and when I do a clean rebuild of the tools I get this:
```
ommy@kingfisher:~/Downloads/riscv-gnu-toolchain-issue1204/installed-tools/bin$ ldd riscv32-unknown-elf-gdb
	linux-vdso.so.1 (0x00007ffc40bd9000)
	libncursesw.so.6 => /lib/x86_64-linux-gnu/libncursesw.so.6 (0x00007f4ef8084000)
	libtinfo.so.6 => /lib/x86_64-linux-gnu/libtinfo.so.6 (0x00007f4ef8054000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f4ef804e000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f4ef8020000)
	libmpfr.so.6 => /lib/x86_64-linux-gnu/libmpfr.so.6 (0x00007f4ef7f9f000)
	libgmp.so.10 => /lib/x86_64-linux-gnu/libgmp.so.10 (0x00007f4ef7f1b000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f4ef7d37000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f4ef7be8000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f4ef7bcd000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f4ef7baa000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f4ef79b8000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f4ef890b000)
```
I.e. - no gdb python support.

TommyMurphyTM1234, 2023-03-02:
As far as I can see, this is an issue with your build and/or run time Python packages/setup/environment and not an issue with the toolchain itself and, consequently, I am closing it.

closed 2023-03-02 by TommyMurphyTM1234