riscv-collab/riscv-gnu-toolchain #1205: Unable to build multiple libraries correctly

dreamcmi, 2023-03-01:
My requirement is to build some newlib libraries that I need, they correspond to different arch and abi, but when I use `--with-multilib-generator` to build, he does not seem to build the required lib correctly, Instead, it exited normally.

The abnormal behavior is as follows
```shell
darren@darren-virtual-machine:~/riscv-gnu-toolchain$ ./configure --prefix=/home/darren/riscv-gnu-toolchain/build --enable-multilib --with-arch=rv32imac --with-abi=ilp32 --with-multilib-generator="rv32imac-ilp32--,rv32imafc-ilp32f--"
darren@darren-virtual-machine:~/riscv-gnu-toolchain$ make -j8
...
...
mkdir -p -- /home/darren/riscv-gnu-toolchain/build/share/gdb/python/gdb/command
mkdir -p -- /home/darren/riscv-gnu-toolchain/build/share/gdb/python/gdb/function
mkdir -p -- /home/darren/riscv-gnu-toolchain/build/share/gdb/python/gdb/printer
make[7]: Leaving directory '/home/darren/riscv-gnu-toolchain/build-gdb-newlib/gdb/data-directory'
make[6]: Leaving directory '/home/darren/riscv-gnu-toolchain/build-gdb-newlib/gdb/data-directory'
make[5]: Leaving directory '/home/darren/riscv-gnu-toolchain/build-gdb-newlib/gdb'
make[4]: Leaving directory '/home/darren/riscv-gnu-toolchain/build-gdb-newlib/gdb'
make[3]: Leaving directory '/home/darren/riscv-gnu-toolchain/build-gdb-newlib/gdb'
make[2]: Leaving directory '/home/darren/riscv-gnu-toolchain/build-gdb-newlib'
make[1]: Leaving directory '/home/darren/riscv-gnu-toolchain/build-gdb-newlib'
mkdir -p stamps/ && touch stamps/build-gdb-newlib
darren@darren-virtual-machine:~/riscv-gnu-toolchain$ ls build/riscv-none-elf/lib/
ldscripts
darren@darren-virtual-machine:~/riscv-gnu-toolchain$ 
```

When I remove `--with-multilib-generator`, it builds the lib fine, but only a few defaults, not what I need.

```shell
darren@darren-virtual-machine:~/riscv-gnu-toolchain$ ./configure --prefix=/home/darren/riscv-gnu-toolchain/build --enable-multilib --with-arch=rv32imac --with-abi=ilp32
darren@darren-virtual-machine:~/riscv-gnu-toolchain$ make -j8
...
...
make[9]: Nothing to be done for 'install-data-am'.
make[9]: Leaving directory '/home/darren/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv-none-elf/rv64imafdc/lp64d/libstdc++-v3'
make[8]: Leaving directory '/home/darren/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv-none-elf/rv64imafdc/lp64d/libstdc++-v3'
make[7]: Leaving directory '/home/darren/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv-none-elf/rv64imafdc/lp64d/libstdc++-v3'
make[6]: Leaving directory '/home/darren/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv-none-elf/libstdc++-v3'
make[5]: Leaving directory '/home/darren/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv-none-elf/libstdc++-v3'
make[4]: Leaving directory '/home/darren/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv-none-elf/libstdc++-v3'
make[3]: Leaving directory '/home/darren/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv-none-elf/libstdc++-v3'
make[2]: Leaving directory '/home/darren/riscv-gnu-toolchain/build-gcc-newlib-stage2'
make[1]: Leaving directory '/home/darren/riscv-gnu-toolchain/build-gcc-newlib-stage2'
mkdir -p stamps/ && touch stamps/build-gcc-newlib-stage2
darren@darren-virtual-machine:~/riscv-gnu-toolchain$ ls build/riscv-none-elf/lib/
crt0.o     libc_nano.a  libgloss_nano.a  libm_nano.a    libsim.a            libstdc++.la  nano.specs   rv32iac    rv64imac        sim.specs
ldscripts  libg.a       libg_nano.a      libnosys.a     libstdc++.a         libsupc++.a   nosys.specs  rv32im     rv64imafdc
libc.a     libgloss.a   libm.a           libsemihost.a  libstdc++.a-gdb.py  libsupc++.la  rv32i        rv32imafc  semihost.specs
```

TommyMurphyTM1234, 2023-03-01:
Are you sure that you're doing a clean build each time?

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1196

Try omitting `--enable-multilib` and just specifying `--with-multilib-generator="rv32imac-ilp32--,rv32imafc-ilp32f--"` and see if that makes any difference.

TommyMurphyTM1234, 2023-03-01:
`--with-multilib-generator="rv32imac-ilp32--,rv32imafc-ilp32f--"`

Actually, the problem is most likely that your multilib list is comma separated and should be semicolon separated.

* https://github.com/riscv-collab/riscv-gnu-toolchain#build-with-customized-multi-lib-configure

dreamcmi, 2023-03-01:
Oh, there is another question, if I want to compile the gcc toolchain that supports `rv32imac` `rv32imafc` `rv32ec` at the same time, what should I do?
I am now executing `./configure --prefix=/home/darren/riscv-gnu-toolchain/build --enable-multilib --with-arch=rv32imac --with-abi=ilp32 --with-multilib-generator="rv32imac-ilp32--;rv32imafc-ilp32f--;rv32ec-ilp32e--"
`
But when compiling the rv32imafc target, there will be a newlib problem. It seems that the hard floating point (abi is `ilp32f`) is not used when newlib is compiled, but the abi is set to `ilp32`.

```shell
/home/darren/riscv-gnu-toolchain/build/lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/bin/ld: failed to merge target specific data of file /home/darren/riscv-gnu-toolchain/build/lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none- elf/lib/libnosys.a(fstat.o)
/home/darren/riscv-gnu-toolchain/build/lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/bin/ld:/ home/darren/riscv-gnu-toolchain/build/lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/lib/libnosys.a( isatty.o): can't link soft-float modules with single-float modules
```

TommyMurphyTM1234, 2023-03-01:
Have you tried the following?
```
./configure --prefix=/home/darren/riscv-gnu-toolchain/build --with-multilib-generator="rv32imac-ilp32--;rv32imafc-ilp32f--;rv32ec-ilp32e--" 
```
to get a toolchain that supports `rv32gc/ilp32d` by default (i.e. when compilation is done with no explicit `-march=... -mabi=...` options) but also supports your required `arch/abi` targets through multilibs?

dreamcmi, 2023-03-01:
I set `add_compile_options(-march=rv32imafc -mabi=ilp32f)` when compiling

dreamcmi, 2023-03-01:
When I set `./configure --prefix=/home/darren/riscv-gnu-toolchain/build --with-multilib-generator="rv32imac-ilp32--;rv32imafc-ilp32f--;rv32ec-ilp32e-- " `
It throws new error `ABI is incompatible with that of the selected emulation:target emulation `elf32-littleriscv' does not match `elf64-littleriscv'`

TommyMurphyTM1234, 2023-03-01:
> I set `add_compile_options(-march=rv32imafc -mabi=ilp32f)` when compiling

I don't understand. Where are you setting this?
You need to provide more details about how to reproduce your issue if you want help.

> When I set ./configure --prefix=/home/darren/riscv-gnu-toolchain/build --with-multilib-generator="rv32imac-ilp32--;rv32imafc-ilp32f--;rv32ec-ilp32e-- " 
It throws new error ABI is incompatible with that of the selected emulation:target emulation elf32-littleriscv does not match elf64-littleriscv'

That's because, for some reason, when you're targeting `rv32imafc/ilp32f` it's falling back to the toolchain default of `rv64gc/lp64d` and generating that mismatch error at link time.

But it's difficult to know what's going on when you are not providing adequate info to reproduce your issues.

TommyMurphyTM1234, 2023-03-01:
I built the toolchain as follows:
```
cd ~/Downloads
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools --with-arch=rv32imac --with-abi=ilp32 --with-multilib-generator="rv32imafc-ilp32f--;rv32ec-ilp32e--"
make
```
Then I created a simple `test.c` program in `.../installed-tools/bin`:
```
/* test.c */
#include<stdio.h>

int main(int argc, char *argv[])
{
    printf("Hello, World!\n");

    return 0;
}
```
And then I compiled it as follows with all `arch/abi` combinations supported by the toolchain and had no problems:
```
cd installed-tools/bin
./riscv32-unknown-elf-gcc test.c -o test-rv32imac-ilp32.exe
./riscv32-unknown-elf-gcc -march=rv32imafc -mabi=ilp32f test.c -o test-rv32imafc-ilp32f.exe
./riscv32-unknown-elf-gcc -march=rv32ec -mabi=ilp32e test.c -o test-rv32ec-ilp32e.exe
```

dreamcmi, 2023-03-02:
I think the problem lies in the `ld` stage. It can be seen that your steps are for running on the local machine without explicitly linking newlib. Because I am going to use it for riscv mcu, so newlib will be statically linked when compiling. It is written like this in cmake
```cmake
set(LINKER_SCRIPT ${CMAKE_SOURCE_DIR}/Ld/Link.ld)
add_link_options(-nostartfiles
                 -Xlinker --gc-sections
                 -Wl, --print-memory-usage
                 -Wl,-Map,${PROJECT_NAME}.map
                 --specs=nano.specs
                 --specs=nosys.specs)
```

https://github.com/dreamcmi/CH32V307-CMake

When I follow your command line (`./configure --prefix=/home/darren/riscv-gnu-toolchain/build  --with-arch=rv32imac --with-abi=ilp32 --with-multilib-generator="rv32imafc-ilp32f--;rv32ec-ilp32e--"`) after compiling it still gives the error.

```shell
darren@darren-virtual-machine:~/CH32V307-CMake$ rm -rf build
darren@darren-virtual-machine:~/CH32V307-CMake$ cmake -G Ninja -B build
-- Now is UNIX-like OS!
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/darren/riscv-gnu-toolchain/build/bin/riscv-none-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/darren/riscv-gnu-toolchain/build/bin/riscv-none-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/darren/riscv-gnu-toolchain/build/bin/riscv-none-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/darren/CH32V307-CMake/build
darren@darren-virtual-machine:~/CH32V307-CMake$ cmake --build build
[2/33] Building C object CMakeFiles/ch32v307-ninja.elf.dir/APP/ch32v30x_it.c.obj
/home/darren/CH32V307-CMake/APP/ch32v30x_it.c:12:1: warning: argument to 'interrupt' attribute is not '"user"', '"supervisor"', or '"machine"' [-Wattributes]
   12 | void NMI_Handler(void) __attribute__((interrupt("WCH-Interrupt-fast")));
      | ^~~~
/home/darren/CH32V307-CMake/APP/ch32v30x_it.c:13:1: warning: argument to 'interrupt' attribute is not '"user"', '"supervisor"', or '"machine"' [-Wattributes]
   13 | void HardFault_Handler(void) __attribute__((interrupt("WCH-Interrupt-fast")));
      | ^~~~
[33/33] Linking C executable ch32v307-ninja.elf
FAILED: ch32v307-ninja.elf 
: && /home/darren/riscv-gnu-toolchain/build/bin/riscv-none-elf-gcc  -nostartfiles -Xlinker --gc-sections -Wl,--print-memory-usage -Wl,-Map,ch32v307-ninja.map --specs=nano.specs --specs=nosys.specs -T /home/darren/CH32V307-CMake/Ld/Link.ld CMakeFiles/ch32v307-ninja.elf.dir/APP/ch32v30x_it.c.obj CMakeFiles/ch32v307-ninja.elf.dir/APP/main.c.obj CMakeFiles/ch32v307-ninja.elf.dir/APP/system_ch32v30x.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Core/core_riscv.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Debug/debug.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Peripheral/src/ch32v30x_adc.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Peripheral/src/ch32v30x_bkp.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Peripheral/src/ch32v30x_can.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Peripheral/src/ch32v30x_crc.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Peripheral/src/ch32v30x_dac.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Peripheral/src/ch32v30x_dbgmcu.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Peripheral/src/ch32v30x_dma.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Peripheral/src/ch32v30x_dvp.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Peripheral/src/ch32v30x_eth.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Peripheral/src/ch32v30x_exti.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Peripheral/src/ch32v30x_flash.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Peripheral/src/ch32v30x_fsmc.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Peripheral/src/ch32v30x_gpio.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Peripheral/src/ch32v30x_i2c.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Peripheral/src/ch32v30x_iwdg.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Peripheral/src/ch32v30x_misc.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Peripheral/src/ch32v30x_opa.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Peripheral/src/ch32v30x_pwr.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Peripheral/src/ch32v30x_rcc.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Peripheral/src/ch32v30x_rng.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Peripheral/src/ch32v30x_rtc.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Peripheral/src/ch32v30x_sdio.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Peripheral/src/ch32v30x_spi.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Peripheral/src/ch32v30x_tim.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Peripheral/src/ch32v30x_usart.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Peripheral/src/ch32v30x_wwdg.c.obj CMakeFiles/ch32v307-ninja.elf.dir/Startup/startup_ch32v30x_D8C.S.obj -o ch32v307-ninja.elf   && cd /home/darren/CH32V307-CMake/build && /home/darren/riscv-gnu-toolchain/build/bin/riscv-none-elf-objcopy -Oihex /home/darren/CH32V307-CMake/build/ch32v307-ninja.elf /home/darren/CH32V307-CMake/build/ch32v307-ninja.hex && /home/darren/riscv-gnu-toolchain/build/bin/riscv-none-elf-objcopy -Obinary /home/darren/CH32V307-CMake/build/ch32v307-ninja.elf /home/darren/CH32V307-CMake/build/ch32v307-ninja.bin && /home/darren/riscv-gnu-toolchain/build/bin/riscv-none-elf-objdump --all-headers --demangle --disassemble / [... cut ...]
/home/darren/riscv-gnu-toolchain/build/lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/bin/ld: failed to merge target specific data of file /home/darren/riscv-gnu-toolchain/build/lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/lib/libnosys.a(read.o)
Memory region         Used Size  Region Size  %age Used
           FLASH:        8324 B       288 KB      2.82%
             RAM:         32 KB        32 KB    100.00%
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

TommyMurphyTM1234, 2023-03-02:
> I think the problem lies in the `ld` stage. It can be seen that your steps are for running on the local machine without explicitly linking newlib. 

Yes, but it is implicitly/automatically linking newlib. There's no other C library in the bare metal toolchain (unless you build the Picolib version of the toolchain if the patches for doing so have all been integrated - I'm not sure).

> Because I am going to use it for riscv mcu, so newlib will be statically linked when compiling. 

Same with my simple example. Obviously everything is statically linked for a bare metal target.

> It is written like this in cmake

Your issue is with your build process and not with the toolchain. I.e. it's a CMake/Ninja issue and not a toolchain issue. You should probably seek support for these tools in another more appropriate forum.

closed 2023-03-02 by TommyMurphyTM1234

dreamcmi, 2023-03-02:
Oh, you reminded me that I did not set march and mabi for ld, which caused this error. After I set it, it worked normally, thank you very much.