riscv-collab/riscv-gnu-toolchain #1206: multilib toolchain dejagnu testing broken chapter 2

vineetgarc, 2023-03-01:
Current multilib dejagnu testing is broken per #1201. However I was desperate (to thoroughly test some gcc patches in flight) so locally hacked dejagnu submodule to point back to github 1.6 version. But it seems that is broken still, hence this separate issue.

Now dejagnu **execute** tests (using riscv qemu) fail - example below.

```
Executing on host: /scratch/vineetg/gnu/toolchain-upstream/build-gcc-linux-stage2/gcc/xgcc -B/scratch/vineetg/gnu/toolchain-upstream/build-gcc-linux-stage2/gcc/ /scratch/vineetg/gnu/toolchain-upstream/gcc/gcc/testsuite/gcc.target/riscv/li.c  -march=rv32imac -mabi=ilp32 -mcmodel=medlow   -fdiagnostics-plain-output    -O0       -lm  -o ./li.exe    (timeout = 600)

spawn -ignore SIGHUP /scratch/vineetg/gnu/toolchain-upstream/build-gcc-linux-stage2/gcc/xgcc -B/scratch/vineetg/gnu/toolchain-upstream/build-gcc-linux-stage2/gcc/ /scratch/vineetg/gnu/toolchain-upstream/gcc/gcc/testsuite/gcc.target/riscv/li.c -march=rv32imac -mabi=ilp32 -mcmodel=medlow -fdiagnostics-plain-output -O0 -lm -o ./li.exe

PASS: gcc.target/riscv/li.c   -O0  (test for excess errors)

spawn riscv64-unknown-linux-gnu-run ./li.exe
+ qemu_args=()
+ [[ ./li.exe != '' ]]
+ case "$1" in
+ break
++ readelf -h ./li.exe
++ grep Class
++ cut -d: -f 2
++ xargs echo
++ sed 's/^ELF//'
+ xlen=32
+ QEMU_CPU=rv64
+ qemu-riscv32 -r 5.10 -L /scratch/vineetg/gnu/INSTALL/tc-up-230228-afe6cea44898/sysroot ./li.exe
qemu-riscv32: unable to find CPU model 'rv64'
FAIL: gcc.target/riscv/li.c   -O0  execution test
```

It seems that commit 55982428cdf8a7e4b9ba1b9d2bdaf0b471 ("Pass right extension info to qemu") had right intent to automate qemu args for adhoc extension builds. However it ended up making multilib worse as the default $QEMU_CPU corresponds to default WITH_ARCH (rv64) which doesn't apply to all multilib variants: e.g. 32-bit qemu doesn't like rv64 passed to it.

The solution would be to defer evaluation of QEMU_CPU from build time (Makefile.in) to qemu invocation time and somehow use the info from current multilib being processed in wrapper runner to generate QEMU_CPU.

I wanted to solicit some feedback on this approach or are there better ideas to go about doing this.

/CC @palmer-dabbelt @kito-cheng @cmuellner

kito-cheng, 2023-03-02:
One solution is extract the ELF attribute from ELF file and then use it as qemu option, that's the way we did internally, although it come with some performance penalty since it must read and parse the ELF file before run.

vineetgarc, 2023-03-02:
Yeah that feels excessive. Considering there 10's of thousands of dejagnu tests which execute. We need to think of a better way.

FWIW someone internally was complaining about the inefficient runner wrapper with shell pipeline: readelf | grep | cut | xargs | sed  and ensuing 2 sub-shells when one could builtin variable substitution. :-)

TommyMurphyTM1234, 2023-03-02:
Does this really need a second issue when it would be arguably better to keep all info in the existing one?

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1201

vineetgarc, 2023-03-02:
Yes it does, because even if we go back to working dejagnu (1.6) multilib still works due to other changes as described at top.

TommyMurphyTM1234, 2023-03-02:
> Yes it does, because even if we go back to working dejagnu (1.6) multilib still works due to other changes as described at top.

I don't understand. How exactly do the two issues cover different unrelated subjects? There seems to be to be enough overlap for there to be just a single issue?

vineetgarc, 2023-03-02:
Sure both are related to same high level multilib build topic, but they are 2 distinct activities (ion different areas).

#1201 requires fixing to upstream dejagnu 1.6.3+ to adapt to riscv multib invocations (or worst case revert back to 1.6. on github)
With that fixed, we run into a different issue where we need different changes to how qemu is invoked for multib.

I think it is generally better to separate things that way - but sure if you feel they belong to same issue feel free to close one and overload the other one with additional failure symptoms.

cmuellner, 2023-04-17:
@vineetgarc please retest this issue.

As shown in #1201, running the tests works fine for me once DejaGnu supports spaces in command line arguments again.

vineetgarc, 2023-04-17:
> @vineetgarc please retest this issue.
> 
> As shown in #1201, running the tests works fine for me once DejaGnu supports spaces in command line arguments again.

Unfortunately I see the same bulk failure for all of rv32 

cat ./build-gcc-linux-stage2/gcc/testsuite/gcc/gcc.log
```
...
PASS: gcc.c-torture/execute/builtins/20010124-1.c compilation,  -O0 
spawn riscv64-unknown-linux-gnu-run /scratch/vineetg/gnu/toolchain-upstream-13/build-gcc-linux-stage2/gcc/testsuite/gcc/20010124-1.x0
qemu-riscv32: unable to find CPU model 'rv64'
FAIL: gcc.c-torture/execute/builtins/20010124-1.c execution,  -O0 
```

vineetgarc, 2023-04-17:
Can you check if there's some other qemu installed in your setup. Also could you please check the gcc.log for the same cmd and see how qemu is being invoked (perhaps stick on set -x in the run script)

commit 31a8322e (2023-04-20): Set qemu cpu option from ELF attribute

This could help multi-lib testing, but the price is slightly increase
the testing time since it will need to extract ELF attribute from binary
before running qemu.

But I think the cost is acceptable compare to make build system more
complicate, and actually we already use this approach in our internal stuffs
for years.

Fix #1206.
PR #1233 (2023-05-13): Set qemu cpu option from ELF attribute
This could help multi-lib testing, but the price is slightly increase the testing time since it will need to extract ELF attribute from binary before running qemu.

But I think the cost is acceptable compare to make build system more complicate, and actually we already use this approach in our internal stuffs for years.

Fix #1206.
closed 2023-05-13 by kito-cheng