riscv-collab/riscv-gnu-toolchain #1208: Which flags are necessary for simple matrix-mult app to use vector extension, instead of removing the loop out?

NazerkeT, 2023-03-08:
Hi there,

When I was compiling my simple matrix multiplication app, i noticed that riscv vector compiler, optimizes the loop out of the execution and does not use vector extension instructions at all. Is there any hint on which flags to use to optimize enough to use vector extension, but not to remove the loop at all? 

So far I tried different combinations of the `-O1` to `-O3` with `--ftree-vectorize`, and my full command is following: `/opt/riscv-vector/bin/riscv64-unknown-elf-gcc -march=rv64gcv -ftree-vectorize building-blocks.c -lm -o vec-blocks.o` 
however so far it does not work :(

Thanks,
Best wishes,

TommyMurphyTM1234, 2023-03-08:
One question - can you clarify how you're building your toolchain?
Not sure about others, but I'm constantly confused about how one is supposed to build the latest vector enabled toolchain and it's not clearly documented anywhere that I'm aware of...

NazerkeT, 2023-03-08:
hmm, so far I think I build the tool-chain from 'rvv-next' branch of this gnu-toolchain, hope it is helpful @TommyMurphyTM1234

TommyMurphyTM1234, 2023-03-09:
> Hi there,
> 
> When I was compiling my simple matrix multiplication app, i noticed that riscv vector compiler, optimizes the loop out of the execution and does not use vector extension instructions at all. Is there any hint on which flags to use to optimize enough to use vector extension, but not to remove the loop at all?
> 
> So far I tried different combinations of the `-O1` to `-O3` with `--ftree-vectorize`, and my full command is following: `/opt/riscv-vector/bin/riscv64-unknown-elf-gcc -march=rv64gcv -ftree-vectorize -O3 building-blocks.c -lm -o vec-blocks.o` however so far it does not work :(
> 
> Thanks, Best wishes,

Can you provide some example code please?

TommyMurphyTM1234, 2023-03-11:
Any update @NazerkeT?
Without a reproducible test case there's not much that anybody can do and the issue will have to be closed.

NazerkeT, 2023-03-11:
Hi @TommyMurphyTM1234 ! Sorry for getting lost, got switched into other projects temporarily :(

So far this is the code that I am testing with command: `/opt/riscv-vector/bin/riscv64-unknown-elf-gcc -march=rv64gcv -ftree-vectorize -O3 building-blocks.c -lm -o vec-blocks.o`

```
#include <stdio.h>
#include <stdlib.h>
#include <math.h>

#define HEIGHT 64
#define LENGTH  64

int main(){

  double a [HEIGHT][LENGTH];
  double b [HEIGHT][LENGTH];

  for (int i=0; i<HEIGHT; i++){
     for (int j=0; j<LENGTH;j++){
       a[i][j] = i+j;
    }
  }

  for (int i=0; i<HEIGHT; i++){
    for (int j=0; j<LENGTH;j++){
      b[i][j] = (i+j)/LENGTH;
    }
  }

  double c [HEIGHT][LENGTH] = {0};

  for (int i=0; i<HEIGHT; i++){
     for (int j=0; j<LENGTH; j++){
       for (int k=0; k<HEIGHT; k++){
          c[i][j] += a[i][k]*b[k][j]; 
       }
     }
  }

  return 0;
}
```

TommyMurphyTM1234, 2023-03-12:
Isn't the compiler correct to optimise all of this away because, ultimately, none of the calculations are actually used for anything?
If I add `volatile` to the definition of all of the matrices then I get this:
(Note that my toolchain used `./configure --with-arch=rv64gc --with-abi=lp64d` so defaults to using that `arch/abi` when compiling),
```
riscv64-unknown-elf-gcc -g -c -ftree-vectorize -O3 test.c
riscv64-unknown-elf-objdump -dS test.o 

test.o:     file format elf64-littleriscv

Disassembly of section .text.startup:

0000000000000000 <main>:
#include <math.h>

#define HEIGHT 64
#define LENGTH  64

int main(){
   0:	7179                	add	sp,sp,-48
   2:	72a1                	lui	t0,0xfffe8

0000000000000004 <.LBB3>:
  volatile double a [HEIGHT][LENGTH];
  volatile double b [HEIGHT][LENGTH];

  for (int i=0; i<HEIGHT; i++){
     for (int j=0; j<LENGTH;j++){
       a[i][j] = i+j;
   4:	77a1                	lui	a5,0xfffe8

0000000000000006 <.LBE3>:
int main(){
   6:	f406                	sd	ra,40(sp)

0000000000000008 <.LBB4>:
       a[i][j] = i+j;
   8:	6761                	lui	a4,0x18

000000000000000a <.LBE4>:
int main(){
   a:	9116                	add	sp,sp,t0

000000000000000c <.LBB5>:
       a[i][j] = i+j;
   c:	973e                	add	a4,a4,a5
   e:	101c                	add	a5,sp,32
  10:	97ba                	add	a5,a5,a4

0000000000000012 <.LBE5>:
  for (int i=0; i<HEIGHT; i++){
  12:	4601                	li	a2,0

0000000000000014 <.LBB6>:
       a[i][j] = i+j;
  14:	e43e                	sd	a5,8(sp)
     for (int j=0; j<LENGTH;j++){
  16:	04000593          	li	a1,64

000000000000001a <.L2>:
  1a:	4701                	li	a4,0
  1c:	00661513          	sll	a0,a2,0x6

0000000000000020 <.L3>:
       a[i][j] = i+j;
  20:	00e606bb          	addw	a3,a2,a4
  24:	d20687d3          	fcvt.d.w	fa5,a3
  28:	66a2                	ld	a3,8(sp)
  2a:	00e507b3          	add	a5,a0,a4
  2e:	078e                	sll	a5,a5,0x3
  30:	97b6                	add	a5,a5,a3
[... 150 lines cut ...]

00000000000000f0 <.LVL14>:
  f0:	fcd595e3          	bne	a1,a3,ba <.L11>

00000000000000f4 <.LBE18>:
  for (int i=0; i<HEIGHT; i++){
  f4:	2885                	addw	a7,a7,1

00000000000000f6 <.LVL15>:
  f6:	fab89fe3          	bne	a7,a1,b4 <.L7>

00000000000000fa <.LBE13>:
       }
     }
  }

  return 0;
}
  fa:	62e1                	lui	t0,0x18
  fc:	9116                	add	sp,sp,t0
  fe:	70a2                	ld	ra,40(sp)
 100:	4501                	li	a0,0
 102:	6145                	add	sp,sp,48
 104:	8082                	ret
```

NazerkeT, 2023-03-13:
Hi @TommyMurphyTM1234,

Thanks for reproducing my code! So far, yeah, when I build the vector support, via `rvv-next` branch of this gnu-toolchain repo, it turns out that it fetches from `riscv-gcc-rvv-next` branch of the riscv-gcc already, as a result I was able to get the vectorized results when my matmul operations are followed by some others ops. When, i did not had any code to follow matmul though, I had around similar results to yours both with and without `volatile` keyword too :(

I guess it is indeed because it is logical for a compiler to optimize out the loop logic as it is not going to be used later. However, i thought that there can be other way to circumvent that optimization, but to optimize only to autovectorization for any code. If that is possible please let me know! if it is not, then i guess closing the issue could help :(

TommyMurphyTM1234, 2023-03-13:
> However, i thought that there can be other way to circumvent that optimization, but to optimize only to autovectorization for any code. If that is possible please let me know! 

Sorry, I don't understand your point/question. Maybe you can clarify?

NazerkeT, 2023-03-13:
Oh what I meant is coming back to my initial question. Can we optimize the code to include autovectorization instead of removing the loops even though there are no code to follow the matmul? I was just wondering whether there are some ways to do it, that I am not aware of yet, and looking at gcc manual made a bit confused.

But if you think it is not possible, we can close the issue here, and I can post my question at riscv-gcc repo :)

Thanks for your help!

TommyMurphyTM1234, 2023-03-15:
Polling for assistance...

Do any of @zhongjuzhe, @kito-cheng or @cmuellner (or anybody else) - know if/how a GCC toolchain can be built with auto-vectorization support that will use V instructions for the simple example code in this issue? I have tried building the toolchain as follows:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain -b rvv-next
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools --with-arch=rv64gcv --with-abi=lp64d
date > build.log && make 2>&1 | tee -a build.log && date >> build.log
```
but the resulting toolchain does not generate any vector instructions for the example code above (even when `volatile` is used in the definitions of the relevant matrices/arrays).

@NazerkeT - I wonder if you might be better off trying LLVM which, I believe, has had auto-vectorization support for a good while now?

NazerkeT, 2023-03-16:
@TommyMurphyTM1234, thanks for your help! i think so far I will stick with gcc, as I got used to it vs llvm and I am a bit short in time :(

Regarding the solution though, I think I am just going to blend my matmul with other complex code and get the necessary data looking at the assembly and spike simulator results, if other ways are not possible. Will be looking towards the answers of others too though!

Thanks,
Best wishes,

zhongjuzhe, 2023-03-16:
> 

I am so sorry for late reply. Actually, I already fixed it but I forgot to update the "rvv-next" since I am so busy recently.

Can you make a PR to fix it?

The fix is very simple:
https://github.com/riscv-collab/riscv-gcc/blob/eb96155f49deefcde0b054f6c02aa67c607ac482/gcc/config/riscv/riscv-vector.cc#L4370

- rtx step = CONST_VECTOR_ELT (src, 2);
+ rtx step = GEN_INT(INTVAL(CONST_VECTOR_ELT (src, 4)) - INTVAL(CONST_VECTOR_ELT (src, 2)));

pz9115, 2023-03-16:
I had open an PR to fix it as Juzhe suggestions: https://github.com/riscv-collab/riscv-gcc/pull/370

zhongjuzhe, 2023-03-16:
> I had open an PR to fix it as Juzhe suggestions: [riscv-collab/riscv-gcc#370](https://github.com/riscv-collab/riscv-gcc/pull/370)

Thanks. Issue fixed.

TommyMurphyTM1234, 2023-03-16:
Hi @zhongjuzhe and @pz9115 - thanks a lot for the updates.

However after building the toolchain today as follows:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain -b rvv-next
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools --with-arch=rv64gcv --with-abi=lp64d
date > build.log && make 2>&1 | tee -a build.log && date >> build.log
```
and then compiling this code:
```
/* test.c */

#include <stdio.h>
#include <stdlib.h>
#include <math.h>

#define HEIGHT 64
#define LENGTH  64

int main(){

  volatile int a [HEIGHT][LENGTH];
  volatile int b [HEIGHT][LENGTH];

  for (int i=0; i<HEIGHT; i++){
     for (int j=0; j<LENGTH;j++){
       a[i][j] = i+j;
    }
  }

  for (int i=0; i<HEIGHT; i++){
    for (int j=0; j<LENGTH;j++){
      b[i][j] = (i+j)/LENGTH;
    }
  }

  volatile int c [HEIGHT][LENGTH] = {0};

  for (int i=0; i<HEIGHT; i++){
     for (int j=0; j<LENGTH; j++){
       for (int k=0; k<HEIGHT; k++){
          c[i][j] += a[i][k]*b[k][j]; 
       }
     }
  }

  return 0;
}
```
as follows:
```
riscv64-unknown-elf-gcc -g -c -ftree-vectorize -O3 test.c
```
I still get no vector instructions:
```
riscv64-unknown-elf-objdump -dS test.o

test.o:     file format elf64-littleriscv

Disassembly of section .text.startup:

0000000000000000 <main>:
#include <math.h>

#define HEIGHT 64
#define LENGTH  64

int main(){
   0:	7179                	addi	sp,sp,-48
[... 201 lines cut ...]
00000000000000ec <.LVL14>:
  ec:	fcd514e3          	bne	a0,a3,b4 <.L11>

00000000000000f0 <.LBE18>:
  for (int i=0; i<HEIGHT; i++){
  f0:	2305                	addiw	t1,t1,1

00000000000000f2 <.LVL15>:
  f2:	faa31ee3          	bne	t1,a0,ae <.L7>

00000000000000f6 <.LBE13>:
       }
     }
  }

  return 0;
}
  f6:	62b1                	lui	t0,0xc
  f8:	9116                	add	sp,sp,t0
  fa:	70a2                	ld	ra,40(sp)
  fc:	4501                	li	a0,0
  fe:	6145                	addi	sp,sp,48
 100:	8082                	ret
```
Am I doing something wrong?

zhongjuzhe, 2023-03-16:
> Hi @zhongjuzhe and @pz9115 - thanks a lot for the updates. Much appreciated.
> 
> However, after building the toolchain today as follows:
> 
> ```
> git clone https://github.com/riscv-collab/riscv-gnu-toolchain -b rvv-next
> cd riscv-gnu-toolchain
> ./configure --prefix=`pwd`/installed-tools --with-arch=rv64gcv --with-abi=lp64d
> date > build.log && make 2>&1 | tee -a build.log && date >> build.log
> ```
> 
> and then compiling this code:
> 
> ```
> /* test.c */
> 
> #include <stdio.h>
> #include <stdlib.h>
> #include <math.h>
> 
> #define HEIGHT 64
> #define LENGTH  64
> 
> int main(){
> 
>   volatile int a [HEIGHT][LENGTH];
>   volatile int b [HEIGHT][LENGTH];
> 
>   for (int i=0; i<HEIGHT; i++){
>      for (int j=0; j<LENGTH;j++){
>        a[i][j] = i+j;
>     }
>   }
> 
>   for (int i=0; i<HEIGHT; i++){
>     for (int j=0; j<LENGTH;j++){
>       b[i][j] = (i+j)/LENGTH;
>     }
>   }
> 
>   volatile int c [HEIGHT][LENGTH] = {0};
> 
>   for (int i=0; i<HEIGHT; i++){
>      for (int j=0; j<LENGTH; j++){
>        for (int k=0; k<HEIGHT; k++){
>           c[i][j] += a[i][k]*b[k][j]; 
>        }
>      }
>   }
> 
>   return 0;
> }
> ```
> 
> as follows:
> 
> ```
> riscv64-unknown-elf-gcc -g -c -ftree-vectorize -O3 test.c
> ```
> 
> I still get no vector instructions:
> 
> ```
> riscv64-unknown-elf-objdump -dS test.o
> 
> test.o:     file format elf64-littleriscv
> 
> 
> Disassembly of section .text.startup:
> 
[... 216 lines cut ...]
> 00000000000000f2 <.LVL15>:
>   f2:	faa31ee3          	bne	t1,a0,ae <.L7>
> 
> 00000000000000f6 <.LBE13>:
>        }
>      }
>   }
> 
>   return 0;
> }
>   f6:	62b1                	lui	t0,0xc
>   f8:	9116                	add	sp,sp,t0
>   fa:	70a2                	ld	ra,40(sp)
>   fc:	4501                	li	a0,0
>   fe:	6145                	addi	sp,sp,48
>  100:	8082                	ret
> ```
> 
> Am I doing something wrong?

I have no idea. I think you should figure out what's going on in your codes.
You can check here:
https://godbolt.org/z/nPWz8n5P8

Both ARM-clang and ARM GCC can not vectorize your codes.

zhongjuzhe, 2023-03-16:
> Hi @zhongjuzhe and @pz9115 - thanks a lot for the updates. Much appreciated.
>
> However, after building the toolchain today as follows:
>
> ```
> git clone [...]

Oh, my bad. I am confused by the issues.

The PR is fixing this issue:
https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1177

For this issue, I have no idea. I think you should figure out what's going on in your codes.
You can check here:
https://godbolt.org/z/nPWz8n5P8

Both ARM-clang and ARM GCC can not vectorize your codes.

I think something wrong in the codes that make compiler faill to vectorize

TommyMurphyTM1234, 2023-03-16:
@NazerkeT - I've tried various different things but cannot get your code to generate V instructions even with `volatile int` instead of the original `double`.
Maybe it's simply not a candidate for the generation of V instructions?
Is there a particular reason why you expect V instructions to be generated for this code?

angsch, 2023-03-17:
Matrix-matrix multiplication cannot be vectorized if you mark all matrices as `volatile`. Remove `volatile` and add a dummy printf() at the end to prevent the entire routine to be optimized away.

```C
#include <stdio.h>
#include <stdlib.h>
#include <math.h>

#define HEIGHT 64
#define LENGTH  64

int main(){

  double a [HEIGHT][LENGTH];
  double b [HEIGHT][LENGTH];

  for (int i=0; i<HEIGHT; i++){
     for (int j=0; j<LENGTH;j++){
       a[i][j] = i+j;
    }
  }

  for (int i=0; i<HEIGHT; i++){
    for (int j=0; j<LENGTH;j++){
      b[i][j] = (i+j)/LENGTH;
    }
  }

  double c [HEIGHT][LENGTH] = {0};

  for (int i=0; i<HEIGHT; i++){
     for (int j=0; j<LENGTH; j++){
       for (int k=0; k<HEIGHT; k++){
          c[i][j] += a[i][k]*b[k][j]; 
       }
     }
  }

  printf("%.6e\n", c[0][0]);

  return 0;
}
```
`riscv64-unknown-linux-gnu-gcc -march=rv64gcv -O3 test.c -S` then yields
```asm
[...]
101 .L11:
102     add a5,a1,t3
103     mv  a4,t1
104     vsetvli a7,a3,e64,m1,ta,mu
105     vle64.v v24,(a0)
106 .L8:
107     fld fa5,0(a4)
108     vle64.v v25,(a5)
109     addi    a5,a5,512
110     addi    a4,a4,8
111     vfmacc.vf   v24,fa5,v25
112     bne a5,a1,.L8
113     vse64.v v24,(a0)
114     sub a3,a3,a7
115     add a0,a0,a6
116     add a1,a5,a6
117     bne a3,zero,.L11
118     addi    s0,s0,1
119     addi    t4,t4,512
120     bne s0,t5,.L7
[...]
```

NazerkeT, 2023-03-18:
Hey guys,

Thanks a lot for discussions and help here!

I tested the approach that @angsch suggested and it works for me. Actually, I tried prints before too, but i remember it not working, i guess i did it in rush and did not get the right results :(

I will close this issue here then!

Thanks,
Best wishes

closed 2023-03-18 by NazerkeT

adamsir32, 2023-03-23:
> 嗨，@zhongjuzhe@pz9115 - 非常感谢您的更新。非常感谢。
> 
> 但是，在今天构建工具链之后，如下所示：
> 
> ```
> git clone https://github.com/riscv-collab/riscv-gnu-toolchain -b rvv-next
> cd riscv-gnu-toolchain
> ./configure --prefix=`pwd`/installed-tools --with-arch=rv64gcv --with-abi=lp64d
> date > build.log && make 2>&1 | tee -a build.log && date >> build.log
> ```
> 
> 然后编译此代码：
> 
> ```
> /* test.c */
> 
> #include <stdio.h>
> #include <stdlib.h>
> #include <math.h>
> 
> #define HEIGHT 64
> #define LENGTH  64
> 
> int main(){
> 
>   volatile int a [HEIGHT][LENGTH];
>   volatile int b [HEIGHT][LENGTH];
> 
>   for (int i=0; i<HEIGHT; i++){
>      for (int j=0; j<LENGTH;j++){
>        a[i][j] = i+j;
>     }
>   }
> 
>   for (int i=0; i<HEIGHT; i++){
>     for (int j=0; j<LENGTH;j++){
>       b[i][j] = (i+j)/LENGTH;
>     }
>   }
> 
>   volatile int c [HEIGHT][LENGTH] = {0};
> 
>   for (int i=0; i<HEIGHT; i++){
>      for (int j=0; j<LENGTH; j++){
>        for (int k=0; k<HEIGHT; k++){
>           c[i][j] += a[i][k]*b[k][j]; 
>        }
>      }
>   }
> 
>   return 0;
> }
> ```
> 
> 如下：
> 
> ```
> riscv64-unknown-elf-gcc -g -c -ftree-vectorize -O3 test.c
> ```
> 
> 我仍然没有得到矢量指令：
> 
> ```
> riscv64-unknown-elf-objdump -dS test.o
> 
> test.o:     file format elf64-littleriscv
> 
> 
> Disassembly of section .text.startup:
> 
[... 211 lines cut ...]
> 
> 00000000000000f0 <.LBE18>:
>   for (int i=0; i<HEIGHT; i++){
>   f0:	2305                	addiw	t1,t1,1
> 
> 00000000000000f2 <.LVL15>:
>   f2:	faa31ee3          	bne	t1,a0,ae <.L7>
> 
> 00000000000000f6 <.LBE13>:
>        }
>      }
>   }
> 
>   return 0;
> }
>   f6:	62b1                	lui	t0,0xc
>   f8:	9116                	add	sp,sp,t0
>   fa:	70a2                	ld	ra,40(sp)
>   fc:	4501                	li	a0,0
>   fe:	6145                	addi	sp,sp,48
>  100:	8082                	ret
> ```
> 
> Am I doing something wrong?
I built a toolchain with V extension following your instructions, but the solution mentioned below doesn't seem to work. Can you help me identify the problem? Best regards.

kito-cheng, 2023-03-23:
Remove volatile